MiczFlor/RPi-Jukebox-RFID #1034: 🐛 | Spotify does not play some tracks

marvingrieger, 2020-07-05:
## Bug

### What I did

I installed the latest RPi-Jukebox-RFID release (v. 2.0.0.) and tried to play an album using spotify.

### What happened

Some tracks do not play. In iris, you can see that the progress bar ("seconds played") gets reset to 0 after 3-4 seconds.
However, most tracks work as expected.

I recorded the phenomenon in Iris, you can see it in the following GIF.
There, you can see that the first track of the album (here: the theme-song) is not played but gets reset after 3 seconds over and over again. Also, I don't hear any sound.
When I skip that track, the next one works as expected as you can see by the time played.
Going back to the previous track, it still does not play and gets reset again.

[image]

### I expected this to happen

I expect that every track is playable.

### Further information that might help

I assume that this problem is somehow related to caching.

When setting `allow_cache = false` in `/etc/mopidy/mopidy.conf`, some tracks that previously worked, stop working.
However, tracks that previously didn't work, still don't work.

## Software

### Base image and version

```
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

### Branch / Release

`master`

### Installscript

`scripts/installscripts/buster-install-default.sh`

## Hardware

### RaspberryPi version

`Raspberry Pi 3 Model B Rev 1.2`

### RFID Reader

`13ba:0018 PCPlay Barcode PCP-BCG4209`

### Soundcard

`Built-In`

### Other notable hardware

`/`

s-martin, 2020-07-05:
Are these local tracks or Spotify?

Are these tracks any different, e.g. length, size, encoding than the tracks which work?

Could you try a working and a not working track and afterwards post the log results from `journalctl -u mopidy`?

marvingrieger, 2020-07-05:
All of these tracks are Spotify-based tracks.
In fact, I am playing an album on Spotify which consists of multiple tracks.

Technically, I am not aware of any differences.
Of course, they have different length, but I don't see anything special.
Also, as written above, changing the `allow_cache = false` property affects the playback.
Here are the Spotify-URIs of the example shown above:

Album No. 1: `spotify:album:0VfB018ZTsA9Ai8OWAyUzS`
Not working (Track 01 - "Theme song"): `spotify:track:7AFFqxBwDNjfSGg7wls0kf`
Working (Track 02): `spotify:track:5Bc98Aa2ORHrMfwmTG75oC`

So far, I observed the problem only on the first track of an album.
I.e., tracks after the first one were not affected (at least, I didn't observe this so far).

My primary use case for the Phoniebox are audiobooks.
That means, I have multiple Spotify-based albums on the Phoniebox.
The first track usually is the "theme song".
I assumed that this track might be "shared" between the audiobooks, therefore I suspected a caching problem (see above).
However. the Spotify-URIs for the theme song are different.
For example, the the URI of the theme song of Album No. 2 is `spotify:track:4SkritXhMzYwjwrGS0VG2T`

The log does not give much information.
Here is the output of `journalctl -u mopidy` after starting an album.
As described, the first track failed whereafter I skipped it.

```
-- Logs begin at Sun 2020-07-05 14:07:17 BST, end at Sun 2020-07-05 14:09:31 BST. --
Jul 05 14:07:29 raspberrypi systemd[1]: Starting Mopidy music server...
Jul 05 14:07:29 raspberrypi systemd[1]: Started Mopidy music server.
Jul 05 14:07:33 raspberrypi mopidy[570]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Jul 05 14:07:33 raspberrypi mopidy[570]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Jul 05 14:07:33 raspberrypi mopidy[570]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Jul 05 14:07:33 raspberrypi mopidy[570]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Jul 05 14:07:33 raspberrypi mopidy[570]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Jul 05 14:07:33 raspberrypi mopidy[570]: INFO     [MainThread] mopidy.config Loading config from command line options
Jul 05 14:07:33 raspberrypi mopidy[570]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Jul 05 14:07:34 raspberrypi mopidy[570]: INFO     [MainThread] mopidy.__main__ Enabled extensions: file, http, local, mpd, stream, iris, softwaremixer, spotify, m3u
Jul 05 14:07:34 raspberrypi mopidy[570]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Jul 05 14:07:35 raspberrypi mopidy[570]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Jul 05 14:07:35 raspberrypi mopidy[570]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Jul 05 14:07:35 raspberrypi mopidy[570]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Jul 05 14:07:35 raspberrypi mopidy[570]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBackend, LocalBackend
Jul 05 14:07:35 raspberrypi mopidy[570]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Jul 05 14:07:36 raspberrypi mopidy[570]: INFO     [SpotifyEventLoop] mopidy_spotify.backend Logged in to Spotify in online mode
Jul 05 14:07:36 raspberrypi mopidy[570]: INFO     [SpotifyBackend-6] mopidy_spotify.web Logged into Spotify Web API as XXX
Jul 05 14:08:09 raspberrypi mopidy[570]: INFO     [SpotifyBackend-6] mopidy_spotify.playlists Refreshed 58 Spotify playlists
Jul 05 14:08:09 raspberrypi mopidy[570]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Jul 05 14:08:09 raspberrypi mopidy[570]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: IrisFrontend, HttpFrontend, MpdFrontend
Jul 05 14:08:09 raspberrypi mopidy[570]: INFO     [IrisFrontend-11] mopidy_iris.core Starting Iris 3.45.0
Jul 05 14:08:09 raspberrypi mopidy[570]: INFO     [HttpFrontend-13] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Jul 05 14:08:09 raspberrypi mopidy[570]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Jul 05 14:08:09 raspberrypi mopidy[570]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
Jul 05 14:08:10 raspberrypi mopidy[570]: INFO     [MpdSession-15] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:58206
Jul 05 14:08:11 raspberrypi mopidy[570]: INFO     [MpdSession-16] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:58210
Jul 05 14:08:16 raspberrypi mopidy[570]: INFO     [MpdSession-17] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:58218
Jul 05 14:08:21 raspberrypi mopidy[570]: INFO     [MpdSession-18] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:58222
Jul 05 14:08:26 raspberrypi mopidy[570]: INFO     [MpdSession-19] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:58226
Jul 05 14:08:31 raspberrypi mopidy[570]: INFO     [MpdSession-20] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:58230
Jul 05 14:08:36 raspberrypi mopidy[570]: INFO     [MpdSession-21] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:58234
[... cut ...]
Jul 05 14:09:23 raspberrypi mopidy[570]: INFO     [MpdSession-35] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:58286
Jul 05 14:09:23 raspberrypi mopidy[570]: INFO     [MpdSession-36] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:58290
Jul 05 14:09:23 raspberrypi mopidy[570]: INFO     [MpdSession-37] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:58294
Jul 05 14:09:24 raspberrypi mopidy[570]: INFO     [MpdSession-39] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:58298
```

marvingrieger, 2020-07-05:
Ok, I enabled the mopidy-logging by creating a `/etc/mopidy/logging.conf` and setting the log-level to debug.

When the track is not played and gets reset all the time, this is what is logged:

```
2020-07-05 14:27:42,990 DEBUG [571:MpdSession-19] mopidy_mpd.session: Request from [::ffff:127.0.0.1]:45870: command_list_begin
2020-07-05 14:27:42,991 DEBUG [571:MpdSession-19] mopidy_mpd.session: Request from [::ffff:127.0.0.1]:45870: load "Conni - Folge 01"
2020-07-05 14:27:42,992 DEBUG [571:MpdSession-19] mopidy_mpd.session: Request from [::ffff:127.0.0.1]:45870: command_list_end
2020-07-05 14:27:43,047 DEBUG [571:HttpServer] mopidy.http.handlers: Received WebSocket message from 192.168.142.60: '{"method":"core.tracklist.get_tl_tracks","params":{},"jsonrpc":"2.0","id":27}'
2020-07-05 14:27:43,128 DEBUG [571:SpotifyBackend-6] urllib3.connectionpool: https://api.spotify.com:443 "GET /v1/me/playlists?limit=50 HTTP/1.1" 200 None
2020-07-05 14:27:43,148 DEBUG [571:SpotifyBackend-6] mopidy_spotify.utils: playlists.as_list() took 137ms
2020-07-05 14:27:43,154 DEBUG [571:HttpServer] mopidy.http.handlers: Sent WebSocket message to 192.168.142.60: '{"jsonrpc": "2.0", "id": 27, "result": []}'
2020-07-05 14:27:43,180 DEBUG [571:Dummy-20] spotify.session: Notify main thread
2020-07-05 14:27:43,181 DEBUG [571:SpotifyEventLoop] spotify.eventloop: Notification received; processing events
2020-07-05 14:27:43,186 DEBUG [571:SpotifyBackend-6] spotify.session: Notify main thread
2020-07-05 14:27:43,188 DEBUG [571:SpotifyEventLoop] spotify.session: Notify main thread
2020-07-05 14:27:43,188 DEBUG [571:SpotifyEventLoop] spotify.album: albumbrowse_complete_callback called
2020-07-05 14:27:43,189 DEBUG [571:SpotifyEventLoop] spotify.session: Metadata updated
2020-07-05 14:27:43,189 DEBUG [571:SpotifyEventLoop] spotify.eventloop: Waiting 2.997s for new events
2020-07-05 14:27:43,191 DEBUG [571:SpotifyEventLoop] spotify.eventloop: Notification received; processing events
2020-07-05 14:27:43,192 DEBUG [571:SpotifyEventLoop] spotify.eventloop: Waiting 2.993s for new events
2020-07-05 14:27:43,192 DEBUG [571:SpotifyEventLoop] spotify.eventloop: Notification received; processing events
2020-07-05 14:27:43,193 DEBUG [571:SpotifyEventLoop] spotify.eventloop: Waiting 2.992s for new events
2020-07-05 14:27:43,245 DEBUG [571:Core-10] mopidy.core.tracklist: Triggering event: tracklist_changed()
2020-07-05 14:27:43,246 DEBUG [571:Core-10] mopidy.listener: Sending tracklist_changed to CoreListener: {}
2020-07-05 14:27:43,249 DEBUG [571:MpdSession-19] mopidy_mpd.session: Response to [::ffff:127.0.0.1]:45870: OK
2020-07-05 14:27:43,251 DEBUG [571:MpdFrontend-14] mopidy.listener: Sending playlist to MpdSession: {}
2020-07-05 14:27:43,256 DEBUG [571:MpdSession-19] mopidy_mpd.network: Client most likely disconnected.
2020-07-05 14:27:43,262 DEBUG [571:MpdSession-19] pykka: Unregistered MpdSession (urn:uuid:e7293f21-3558-49e2-92ac-28b16c2ce0f0)
2020-07-05 14:27:43,264 DEBUG [571:MpdSession-19] pykka: Stopped MpdSession (urn:uuid:e7293f21-3558-49e2-92ac-28b16c2ce0f0)
2020-07-05 14:27:43,266 DEBUG [571:HttpServer] mopidy.http.handlers: Received WebSocket message from 192.168.142.60: '{"method":"core.tracklist.get_tl_tracks","params":{},"jsonrpc":"2.0","id":28}'
2020-07-05 14:27:43,266 DEBUG [571:MpdSession-19] mopidy_mpd.network: Already stopping: Actor is shutting down.
2020-07-05 14:27:43,292 DEBUG [571:HttpServer] mopidy.http.handlers: Sent WebSocket message to 192.168.142.60: '{"jsonrpc": "2.0", "id": 28, "result": [{"__model__": "TlTrack", "tlid": 1, "track": {"__model__": "Track", "uri": "spotify:track:7AFFqxBwDNjfSGg7wls0kf", "name": "Intro Theme Song \\"Meine Freundin Conni\\"", "artists": [{"__model__": "Artist", "uri": "spotify:artist:0HerMOOFZXxIF3Yg9BVl0W", "name": "Meine Freundin Conni"}], "album": {"__model__": "Album", "uri": "spotify:album:0VfB018ZTsA9Ai8OWAyUzS", "name": "01: Conni zieht um / Conni bekommt eine Katze / Conni macht das Seepferdchen / Conni geht zelten (Vier H\\u00f6rspiele zur TV-Serie)", "artists": [{"__model__": "Artist", "uri": "spotify:artist:0HerMOOFZXxIF3Yg9BVl0W", "name": "Meine Freundin Conni"}], "date": "2012"}, "track_no": 1, "disc_no": 1, "date": "2012", "length": 38000, "bitrate": 160}}, {"__model__": "TlTrack", "tlid": 2, "track": {"__model__": "Track", "uri": "spotify:track:5Bc98Aa2ORHrMfwmTG75oC", "name": "Conni zieht um - Teil 01", "artists": [{"__model__": "Artist", "uri": "spotify:artist:0HerMOOFZXxIF3Yg9BVl0W", "name": "Meine Freundin Conni"}], "album": {"__model__": "Album", "uri": "spotify:album:0VfB018ZTsA9Ai8OWAyUzS", "name": "01: Conni zieht um / Conni bekommt eine Katze / Conni macht das Seepferdchen / Conni geht zelten (Vier H\\u00f6rspiele zur TV-Serie)", "artists": [{"__model__": "Artist", "uri": "spotify:artist:0HerMOOFZXxIF3Yg9BVl0W", "name": "Meine Freundin Conni"}], "date": "2012"}, "track_no": 2, "disc_no": 1, "date": "2012", "length": 92000, "bitrate": 160}}, {"__model__": "TlTrack", "tlid": 3, "track": {"__model__": "Track", "uri": "spotify:track:0tWls6T6mjGkwcM7Rg8RBo", "name": "Conni zieht um - Teil 02", "artists": [{"__model__": "Artist", "uri": "spotify:artist:0HerMOOFZXxIF3Yg9BVl0W", "name": "Meine Freundin Conni"}], "album": {"__model__": "Album", "uri": "spotify:album:0VfB018ZTsA9Ai8OWAyUzS", "name": "01: Conni zieht um / Conni bekommt eine Katze / Conni macht das Seepferdchen / Conni geht zelten (Vier H\\u00f6rspiele zur TV-Serie)", "artists": [{"__model__": "Artist", " [... cut ...]
2020-07-05 14:28:06,133 DEBUG [571:HttpServer] mopidy.http.handlers: Sent WebSocket message to 192.168.142.60: '{"jsonrpc": "2.0", "id": 69, "result": 0}'
2020-07-05 14:28:07,470 DEBUG [571:HttpServer] mopidy.http.handlers: Received WebSocket message from 192.168.142.60: '{"method":"core.playback.next","params":{},"jsonrpc":"2.0","id":70}'
2020-07-05 14:28:07,474 DEBUG [571:Audio-2] mopidy.audio.actor: Position query failed
2020-07-05 14:28:07,487 DEBUG [571:Audio-2] mopidy.audio.gst: Changing state to GST_STATE_READY: result=GST_STATE_CHANGE_SUCCESS
```

You can see, that the `elapsed` time is alwyays `0.000` as the track is not played.

marvingrieger, 2020-07-05:
After skipping the "broken" track, the next track immediately starts.
This is what gets logged:

```
2020-07-05 14:28:07,489 DEBUG [571:SpotifyBackend-6] mopidy_spotify.playback: Audio requested change of track; loading and starting Spotify player
2020-07-05 14:28:07,491 DEBUG [571:SpotifyBackend-6] spotify.session: Notify main thread
2020-07-05 14:28:07,493 DEBUG [571:SpotifyEventLoop] spotify.eventloop: Notification received; processing events
2020-07-05 14:28:07,493 DEBUG [571:SpotifyBackend-6] spotify.session: Audio buffer stats requested, but no listener
2020-07-05 14:28:07,496 DEBUG [571:SpotifyEventLoop] spotify.eventloop: Waiting 18.732s for new events
2020-07-05 14:28:07,508 DEBUG [571:Audio-2] mopidy.audio.gst: Sending TAG event for track 'spotify:track:5Bc98Aa2ORHrMfwmTG75oC': 'taglist, artist=(string)"Meine\\ Freundin\\ Conni", title=(string)"Conni\\ zieht\\ um\\ -\\ Teil\\ 01", album=(string)"01:\\ Conni\\ zieht\\ um\\ /\\ Conni\\ bekommt\\ eine\\ Katze\\ /\\ Conni\\ macht\\ das\\ Seepferdchen\\ /\\ Conni\\ geht\\ zelten\\ \\(Vier\\ H\\303\\266rspiele\\ zur\\ TV-Serie\\)";'
2020-07-05 14:28:07,513 DEBUG [571:Dummy-9] spotify.session: Audio buffer stats requested, but no listener
2020-07-05 14:28:07,515 DEBUG [571:MainThread] mopidy.audio.gst: Got STATE_CHANGED bus message: old=GST_STATE_READY new=GST_STATE_READY pending=GST_STATE_VOID_PENDING
2020-07-05 14:28:07,517 DEBUG [571:Dummy-9] spotify.session: Audio buffer stats requested, but no listener
2020-07-05 14:28:07,519 DEBUG [571:Audio-2] mopidy.audio.gst: Got source-setup signal: element=GstAppSrc
2020-07-05 14:28:07,521 DEBUG [571:Dummy-9] spotify.session: Audio buffer stats requested, but no listener
2020-07-05 14:28:07,526 DEBUG [571:Dummy-9] spotify.session: Audio buffer stats requested, but no listener
2020-07-05 14:28:07,528 DEBUG [571:Audio-2] mopidy.audio.gst: Changing state to GST_STATE_PLAYING: result=GST_STATE_CHANGE_ASYNC
2020-07-05 14:28:07,532 DEBUG [571:Dummy-9] spotify.session: Audio buffer stats requested, but no listener
2020-07-05 14:28:07,533 DEBUG [571:SpotifyBackend-6] mopidy_spotify.playback: Audio requested seek to 0
2020-07-05 14:28:07,538 DEBUG [571:SpotifyBackend-6] mopidy_spotify.playback: Skipping seek due to issue mopidy/mopidy#300
2020-07-05 14:28:07,537 DEBUG [571:HttpServer] mopidy.http.handlers: Sent WebSocket message to 192.168.142.60: '{"jsonrpc": "2.0", "id": 70, "result": null}'
2020-07-05 14:28:07,544 DEBUG [571:Dummy-9] spotify.session: Audio buffer stats requested, but no listener
2020-07-05 14:28:07,554 DEBUG [571:Dummy-9] spotify.session: Audio buffer stats requested, but no listener
2020-07-05 14:28:07,555 DEBUG [571:HttpServer] mopidy_iris.handlers: Iris websocket message received: {"jsonrpc":"2.0","id":159395568755145,"method":"broadcast","params":{"method":"notification","params":{"notification":{"content":"159395565550064 skipped \"Intro Theme Song \"Meine Freundin Conni\"\"","icon":null}}}}
2020-07-05 14:28:07,560 DEBUG [571:HttpServer] mopidy_iris.core: {'method': 'notification', 'params': {'notification': {'content': '159395565550064 skipped "Intro Theme Song "Meine Freundin Conni""', 'icon': None}}, 'connection_id': 'STO7YP6BTLFY'}
2020-07-05 14:28:07,561 DEBUG [571:Dummy-9] spotify.session: Audio buffer stats requested, but no listener
2020-07-05 14:28:07,562 DEBUG [571:HttpServer] mopidy_iris.core: {'id': 159395568755145, 'jsonrpc': '2.0', 'method': 'broadcast', 'result': {'message': 'Broadcast to 1 connections'}, 'recipient': 'STO7YP6BTLFY'}
2020-07-05 14:28:07,567 DEBUG [571:Dummy-9] spotify.session: Audio buffer stats requested, but no listener
2020-07-05 14:28:07,573 DEBUG [571:MainThread] mopidy.audio.gst: Got STREAM_START bus message
2020-07-05 14:28:07,574 DEBUG [571:Dummy-9] spotify.session: Audio buffer stats requested, but no listener
2020-07-05 14:28:07,575 DEBUG [571:MainThread] mopidy.audio.actor: Audio event: stream_changed(uri='appsrc://')
2020-07-05 14:28:07,578 DEBUG [571:MainThread] mopidy.listener: Sending stream_changed to AudioListener: {'uri': 'appsrc://'}
2020-07-05 14:28:07,580 DEBUG [571:Core-10] mopidy.core.playback: Triggering track playback ended event
2020-07-05 14:28:07,581 DEBUG [571:Dummy-9] spotify.session: Audio buffer stats requested, but no listener
2020-07-05 14:28:07,582 DEBUG [571:Core-10] mopidy.listener: Sending track_playback_ended to CoreListener: {'tl_track': TlTrack(tlid=1, track=Track(album=Album(artists=[Artist(name='Meine Freundin Conni', uri='spotify:artist:0HerMOOFZXxIF3Yg9BVl0W')], date='2012', name='01: Conni zieht um / Conni bekommt eine Katze / Conni macht das Seepferdchen / Conni geht zelten (Vier Hörspiele zur TV-Serie)', uri='spotify:album:0VfB018ZTsA9Ai8OWAyUzS'), artists=[Artist(name='Meine Freundin Conni', uri='spotify:artist:0HerMOOFZXxIF3Yg9BVl0W')], bitrate=160, date='2012', disc_no=1, length=38000, name='Intro Theme Song "Meine Freundin Conni"', track_no=1, uri='spotify:track:7AFFqxBwDNjfSGg7wls0kf')), 'time_position': 0}
2020-07-05 14:28:07,586 DEBUG [571:Core-10] mopidy.core.playback: Changing state: playing -> playing
2020-07-05 14:28:07,589 DEBUG [571:Dummy-9] spotify.session: Audio buffer stats requested, but no listener
2020-07-05 14:28:07,590 DEBUG [571:Core-10] mopidy.core.playback: Triggering playback state change event
2020-07-05 14:28:07,593 DEBUG [571:Core-10] mopidy.listener: Sending playback_state_changed to CoreListener: {'old_state': 'playing', 'new_state': 'playing'}
[... cut ...]
2020-07-05 14:28:11,258 DEBUG [571:MpdSession-32] mopidy_mpd.session: Request from [::ffff:127.0.0.1]:45914: close
2020-07-05 14:28:11,260 DEBUG [571:MpdSession-32] mopidy_mpd.session: Response to [::ffff:127.0.0.1]:45914: OK
2020-07-05 14:28:11,262 DEBUG [571:MpdSession-32] pykka: Unregistered MpdSession (urn:uuid:7b9f119b-e700-4594-bea2-2ca4441b5c28)
2020-07-05 14:28:11,263 DEBUG [571:MpdSession-32] pykka: Stopped MpdSession (urn:uuid:7b9f119b-e700-4594-bea2-2ca4441b5c28)
2020-07-05 14:28:11,263 DEBUG [571:MpdSession-32] mopidy_mpd.network: Actor is shutting down.
```

sonicdee, 2020-07-06:
for me it's kind of the same issue, I've added some new kids audio drama, this are inside a playlist, each, I just swap the tracks, to keed the card, - and there had been a lot of issues with the first track (only) not playing, deleting this first track in my playlist works - with all not playable cards ... strange

s-martin, 2020-07-09:
@kingosticks, do you have any idea?

s-martin, 2020-07-09:
Could you check, if the issue #1039 may be related?

Simse90, 2020-07-09:
It is the same problem. This occurs often in the first track because the intros are often under 60s. Choose a playlist with more shorter tracks (<60 seconds) and you will see that all shorter tracks will not be played. Has nothing to do with track #1.

s-martin, 2020-07-09:
So as mentioned in #1039 the issue may be tracks shorter than 60secs won't be played.

I still don't know, what's the issue. This may be a configuration issue or a bug in mopidy-spotify.

closed 2024-11-10 by s-martin