MiczFlor/RPi-Jukebox-RFID #1036: Available audio - lost my music suddenly

oscarmeier-git, 2020-07-06:
Hi,
I installed everything as written in the installation guide and it worked fine since around 3month. I did not change anything but suddenly a can not play music anymore. I checked the web view and can not find my available audio. It seems that I also can not create a folder in the web view. So I checked my sd Card and found the playlists and audio files, thats not deleted. 
Is there any automatically update that changed the folder structure? And can you give me a tip how to fix it? Should I use my backup or is it only a simple setting? 
Best regards!

s-martin, 2020-07-09:
Are you using Spotify or Classic edition?

oscarmeier-git, 2020-07-10:
I am using the Classic Edition.

s-martin, 2020-07-10:
Please try `mpc update`

oscarmeier-git, 2020-07-11:
I tried mpc update, but there is still no available audio in the web view.

pi@phoniebox:~$ sudo mpc update
Updating DB (#1) ...
volume: 80%   repeat: off   random: off   single: off   consume: off

s-martin, 2020-07-11:
Could you run the script https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki#reporting-bugs and post the results?

Also what’s the result of `tail –n 500 /var/log/syslog`?

oscarmeier-git, 2020-07-15:
pi@phoniebox:~$ /home/pi/RPi-Jukebox-RFID/scripts/helperscripts/Analytics_AfterInstallScript.sh
************************************
*** PHONIEBOX INFO
*** version: 2.0 rc6
*** edition: classic
*** Audio_iFace_Name: Master
*** Audio_Folders_Path: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** Audio_Volume_Change_Step: 10
*** Max_Volume_Limit: 80
*** Idle_Time_Before_Shutdown: 15
*** Second_Swipe: RESTART
*** Playlists_Folders_Path: /home/pi/RPi-Jukebox-RFID/playlists
*** ShowCover: OFF
************************************
*** CONF FILES DEFAULT

*** /etc/samba/smb.conf
-rw-r--r-- 1 root root 9.5K Dec 15  2019 /etc/samba/smb.conf
   path=/home/pi/RPi-Jukebox-RFID/shared
   path=/home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** /etc/lighttpd/lighttpd.conf
-rw-r--r-- 1 root root 1.1K Dec 15  2019 /etc/lighttpd/lighttpd.conf
*** /etc/lighttpd/conf-available/15-fastcgi-php.conf
-rw-r--r-- 1 root root 398 Dec 15  2019 /etc/lighttpd/conf-available/15-fastcgi-php.conf
*** /etc/php/7.0/fpm/php.ini
-rw-r--r-- 1 root root 70K Dec 15  2019 /etc/php/7.0/fpm/php.ini
*** /etc/sudoers
-r--r----- 1 root root 784 Dec 15  2019 /etc/sudoers
*** /etc/systemd/system/phoniebox*
-rw-r--r-- 1 root root 296 Dec 15  2019 /etc/systemd/system/phoniebox-rfid-reader.service
-rw-r--r-- 1 root root 285 Dec 15  2019 /etc/systemd/system/phoniebox-startup-sound.service
-rw-r--r-- 1 root root 289 Dec 18  2019 /etc/systemd/system/phoniebox-gpio-buttons.service
-rw-r--r-- 1 root root 291 Dec 15  2019 /etc/systemd/system/phoniebox-idle-watchdog.service
*** /etc/mpd.conf
-rw-r----- 1 mpd audio 14K Dec 15  2019 /etc/mpd.conf
music_directory         "/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
# music_directory are changed.
# symbolic links outside of the configured music_directory.
# symbolic links inside of the configured music_directory.
        mixer_control   "Master"                # optional
#       mixer_control   "Master"                # optional
*** /etc/dhcpcd.conf
-rw-rw-r-- 1 root netdev 1.8K Jan 13  2020 /etc/dhcpcd.conf
#static ip_address=192.168.0.10/24
#static ip_address=192.168.1.23/24
static ip_address=192.168.178.6
#static routers=192.168.0.1
#static routers=192.168.1.1
static routers=192.168.178.1
option domain_name_servers, domain_name, domain_search, host_name
#static domain_name_servers=192.168.0.1 8.8.8.8 fd51:42f8:caae:d92e::1
#static domain_name_servers=192.168.1.1
static domain_name_servers=192.168.178.1
*** /etc/wpa_supplicant/wpa_supplicant.conf
-rw-rw-r-- 1 root netdev 320 May 12 13:52 /etc/wpa_supplicant/wpa_supplicant.conf
country=DE
************************************
*** +Spotify Edition

*** /etc/locale.gen
-rw-r--r-- 1 root root 8.9K Mar 13  2018 /etc/locale.gen
*** /etc/locale.gen
-rw-r--r-- 1 root root 8.9K Mar 13  2018 /etc/locale.gen
*** /etc/mopidy/mopidy.conf
ls: cannot access '/etc/mopidy/mopidy.conf': No such file or directory
cat: /etc/mopidy/mopidy.conf: No such file or directory
*** ~/.config/mopidy/mopidy.conf
ls: cannot access '/home/pi/.config/mopidy/mopidy.conf': No such file or directory
cat: /home/pi/.config/mopidy/mopidy.conf: No such file or directory

pi@phoniebox:~$

oscarmeier-git, 2020-07-15:
pi@phoniebox:~$ tail –n 500 /var/log/syslog
tail: cannot open '–n' for reading: No such file or directory
tail: cannot open '500' for reading: No such file or directory
==> /var/log/syslog <==
Jul 15 10:32:37 phoniebox systemd[1]: Started Update UTMP about System Runlevel Changes.
Jul 15 10:32:37 phoniebox mpg123[957]: Directory: /home/pi/RPi-Jukebox-RFID/shared/
Jul 15 10:32:37 phoniebox mpg123[957]: Playing MPEG stream 1 of 1: startupsound.mp3 ...
Jul 15 10:32:37 phoniebox mpg123[957]: MPEG 1.0 L III vbr 44100 j-s
Jul 15 10:32:40 phoniebox mpg123[957]: [0:02] Decoding of startupsound.mp3 finished.
Jul 15 10:32:40 phoniebox systemd[1]: Started Phoniebox Startup Sound.
Jul 15 10:32:40 phoniebox systemd[1]: Startup finished in 1.542s (kernel) + 16.275s (userspace) = 17.818s.
Jul 15 10:32:42 phoniebox dhcpcd[351]: wlan0: no IPv6 Routers available
Jul 15 10:32:46 phoniebox systemd[1]: Started Session c2 of user pi.
Jul 15 10:32:49 phoniebox systemd[1]: Started Session c3 of user pi.
pi@phoniebox:~$

MiczFlor, 2020-07-16:
Hi @oscarmeier-git 

you said it worked three months ago and then suddenly not anymore. The Phoniebox does not change the code itself or update the codebase from git. Did you do anything manually, like a `git pull` or something like this on the codebase?

There is the off chance that your SD card is broken. But I would not want you to go through a reinstall until we can rule everything else out.

oscarmeier-git, 2020-07-16:
Yes it worked fine, but there is no working available music anymore. Instead there is the bin and boot directory linked at the main web view. Suddenly it was not possible to create a folder and reupload the music. So I thougt it was a automatic update that crashed something.

s-martin, 2020-12-18:
Are you still experiencing this issue?

s-martin, 2023-05-09:
Closing this issue, feel free to reopen if this issue still occurs.

closed 2023-05-09 by s-martin