MiczFlor/RPi-Jukebox-RFID #1037: How to delete folders/Playlists in web Gui #502

SirY4, 2020-07-08:
this issue was marked closed, however it does not seem to be solved yet, I have the same problem.
How can I remove playlists or folders, if I created them by mistake or don't need them anymore?
I tried to remove the m3u from the playlist folder, still no change in the Webinterface.
any help?

s-martin, 2020-07-09:
Do you use Spotify or Classic edition?

SirY4, 2020-07-09:
> Do you use Spotify or Classic edition?

spotify edition

s-martin, 2020-07-10:
Please try `mopidy local scan` to update the database after you deleted stuff on the file system.

SirY4, 2020-07-10:
Thank you for the advice. I already tried that. Infact the playlists and files disappear from the modipy(iris) interface, but not from the Phoniebox "Player" overview (with the RFID assignments).

s-martin, 2020-07-11:
Try `mpc update`

SirY4, 2020-07-12:
the terminal gives out:  volume,  repeat,   random,   single,  consume values, and current track.

SirY4, 2020-07-14:
Possible idea for future versions:
The interface offers a "track management" option and a "Card editing" menu. A "folder" menu would very useful as well. A simple button with"delete folder" or "delete playlist entry" would be fine as well. 
Thank you for your contributions to the project everyone!

gazza7, 2020-07-23:
The way I clear the folders and playlist is to ssh into the IP address of your Raspi and then cd to the RPi-Jukebox-RFID folder.
In there you will need to go to the playlist folder and clear the playlist you no longer need.
Then go back one folder so you are back to the RPi-Jukebox-RFID, the go to the shared folder and in there you need to go to the audiofolders and delete the folders you no longer require.
Make sure you only delete the folders and playlist not needed, otherwise, you will need to set them up again