MiczFlor/RPi-Jukebox-RFID #1038: Spotify Phonie Box does not work

Cali227, 2020-07-09:
Hey, I hope you can help me. I am totaly new here and i have no idea of prgramming. I just wanted to creat a music box for my daughter =)

So i have this problem: When i install the Phoniebox without Spotify, everything works very well. I changed "PCM" to "Headphone" in "Sudo nano etc/mpd.conf" ( I have no soundcard btw).

But i want to use the Version with Spotify and when i chose [y] in sthe One Line installer, the Phonebox does not work.
I can upload a song from my computer but there is no sound and the track is not playing.

It would be great if you can write exactly what i have to type/change in the console.

I hope you can help me =)

Tahnk you!!!

s-martin, 2020-07-09:
Could you type `journalctl -u mopidy` in the console and post the results here?

Cali227, 2020-07-09:
Hey, i am not sure, how to copy corecct. I am just tasting another script from: 
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/docs/SPOTIFY-INTEGRATION.md

But it does not work. 
Are you german?

login as: pi
pi@192.168.178.58's password:
Linux raspberrypi 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Thu Jul  9 13:41:13 2020 from 192.168.178.20
pi@raspberrypi:~ $ journalctl -u mopidy
-- Logs begin at Thu 2020-07-09 13:39:22 CEST, end at Thu 2020-07-09 13:42:03 CE
Jul 09 13:39:27 raspberrypi systemd[1]: Starting Mopidy music server...
Jul 09 13:39:27 raspberrypi systemd[1]: Started Mopidy music server.
Jul 09 13:39:34 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.__main__ S
Jul 09 13:39:34 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.config Loa
Jul 09 13:39:34 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.config Loa
Jul 09 13:39:34 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.config Loa
Jul 09 13:39:34 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.config Loa
Jul 09 13:39:34 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.config Loa
Jul 09 13:39:34 raspberrypi mopidy[510]: ERROR    [MainThread] mopidy.internal.l
Jul 09 13:39:36 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.__main__ E
Jul 09 13:39:36 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.__main__ D
Jul 09 13:39:38 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.commands S
Jul 09 13:39:38 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.commands M
Jul 09 13:39:38 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.commands S
Jul 09 13:39:38 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.commands S
Jul 09 13:39:38 raspberrypi mopidy[510]: ERROR    [Audio-2] mopidy.audio.actor F
Jul 09 13:39:38 raspberrypi mopidy[510]: ERROR    [Audio-2] pykka Unhandled exce
Jul 09 13:39:38 raspberrypi mopidy[510]: Traceback (most recent call last):
Jul 09 13:39:38 raspberrypi mopidy[510]:   File "/usr/lib/python3/dist-packages/
Jul 09 13:39:38 raspberrypi mopidy[510]:     description, ghost_unlinked_pads=Tr
Jul 09 13:39:38 raspberrypi mopidy[510]: gi.repository.GLib.GError: gst_parse_er
Jul 09 13:39:38 raspberrypi mopidy[510]: During handling of the above exception,
lines 1-23

s-martin, 2020-07-09:
That log is the right direction, but at the bottom something was cut off which is probably the most relevant.
Could you try to post that also?

Cali227, 2020-07-09:
I try IT at home. How dobi copy correct? Right click ist paste 😂

Cali227, 2020-07-09:
I Think this look better?

login as: pi
pi@192.168.178.58's password:
Linux raspberrypi 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Thu Jul  9 15:20:02 2020 from 192.168.178.20
pi@raspberrypi:~ $ journalctl -u mopidy
-- Logs begin at Thu 2020-07-09 13:46:40 CEST, end at Thu 2020-07-09 15:20:54 CEST. --
Jul 09 13:46:45 raspberrypi systemd[1]: Starting Mopidy music server...
Jul 09 13:46:45 raspberrypi systemd[1]: Started Mopidy music server.
Jul 09 13:46:52 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Jul 09 13:46:52 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Jul 09 13:46:52 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Jul 09 13:46:52 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Jul 09 13:46:52 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Jul 09 13:46:52 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.config Loading config from command line options
Jul 09 13:46:52 raspberrypi mopidy[510]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Jul 09 13:46:54 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.__main__ Enabled extensions: iris, spotify, local, mpd, m3u, softwaremixer, http, strea
Jul 09 13:46:54 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Jul 09 13:46:56 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Jul 09 13:46:56 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.commands Mixer volume set to 90
Jul 09 13:46:56 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Jul 09 13:46:56 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBacke
Jul 09 13:46:56 raspberrypi mopidy[510]: ERROR    [Audio-2] mopidy.audio.actor Failed to create audio output "Headphone": gst_parse_error: no element "Headph
Jul 09 13:46:56 raspberrypi mopidy[510]: ERROR    [Audio-2] pykka Unhandled exception in Audio (urn:uuid:40bdf253-6264-45b3-bfb1-41f03423c3c1):
Jul 09 13:46:56 raspberrypi mopidy[510]: Traceback (most recent call last):
Jul 09 13:46:56 raspberrypi mopidy[510]:   File "/usr/lib/python3/dist-packages/mopidy/audio/actor.py", line 125, in add_output
Jul 09 13:46:56 raspberrypi mopidy[510]:     description, ghost_unlinked_pads=True
Jul 09 13:46:56 raspberrypi mopidy[510]: gi.repository.GLib.GError: gst_parse_error: no element "Headphone" (1)
Jul 09 13:46:56 raspberrypi mopidy[510]: During handling of the above exception, another exception occurred:
Jul 09 13:46:56 raspberrypi mopidy[510]: Traceback (most recent call last):
Jul 09 13:46:56 raspberrypi mopidy[510]:   File "/usr/lib/python3/dist-packages/pykka/_actor.py", line 186, in _actor_loop
Jul 09 13:46:56 raspberrypi mopidy[510]:     self.on_start()
Jul 09 13:46:56 raspberrypi mopidy[510]:   File "/usr/lib/python3/dist-packages/mopidy/audio/actor.py", line 458, in on_start
Jul 09 13:46:56 raspberrypi mopidy[510]:     self._setup_outputs()
Jul 09 13:46:56 raspberrypi mopidy[510]:   File "/usr/lib/python3/dist-packages/mopidy/audio/actor.py", line 507, in _setup_outputs
Jul 09 13:46:56 raspberrypi mopidy[510]:     self._outputs.add_output(self._config["audio"]["output"])
Jul 09 13:46:56 raspberrypi mopidy[510]:   File "/usr/lib/python3/dist-packages/mopidy/audio/actor.py", line 131, in add_output
Jul 09 13:46:56 raspberrypi mopidy[510]:     raise exceptions.AudioException(bytes(ex))
Jul 09 13:46:56 raspberrypi mopidy[510]: TypeError: cannot convert 'Error' object to bytes
Jul 09 13:46:57 raspberrypi mopidy[510]: INFO     [SpotifyEventLoop] mopidy_spotify.backend Logged in to Spotify in online mode
Jul 09 13:46:58 raspberrypi mopidy[510]: INFO     [SpotifyBackend-6] mopidy_spotify.web Logged into Spotify Web API as 11149771401
Jul 09 13:47:00 raspberrypi mopidy[510]: INFO     [SpotifyBackend-6] mopidy_spotify.playlists Refreshed 15 Spotify playlists
Jul 09 13:47:00 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Jul 09 13:47:00 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: IrisFrontend, HttpFrontend, MpdFrontend
Jul 09 13:47:00 raspberrypi mopidy[510]: INFO     [IrisFrontend-11] mopidy_iris.core Starting Iris 3.45.0
Jul 09 13:47:00 raspberrypi mopidy[510]: INFO     [HttpFrontend-13] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Jul 09 13:47:00 raspberrypi mopidy[510]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Jul 09 13:47:00 raspberrypi mopidy[510]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
lines 1-42/42 (END)

s-martin, 2020-07-09:
Headphone doesn't look right.

What's the output of `aplay -l`?

Cali227, 2020-07-09:
In the normal Version i had to change IT to Headphone, than IT worked. I did it here as well. What do you mean with the second Part? Is IT correct that IT is not possible at this time to install phoniebox Spotify via the onelineinstaller?

s-martin, 2020-07-09:
Please type `aplay -l` in the console and post the output here.

Spotify should work with the online installer, too.

Cali227, 2020-07-10:
Ok, I take my time and try to say exactly what i did. First for my first try, then second try and the actual try:

**Hardware**;

- Raspberry Pi 3
- 64 GB SD Card
- Euftech USB Reader
- **No** Sound Card (now i have and USB Sound Card, but i didnt try it yet)
- No GPIO Buttons (I wanted to have a functional software first without it)

**First try:**
1. Install RasberianBuster Desktop Version on SD Card
2.  Put it in the Rasberry and let him boot
3.  Using the One-Line Installer ( `cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh `
4.  I choose iFace [y] and MDP also [y]
5.  I Chose Spotify [n]
**Result**
First there was no Sound. I changend with `sudo nano /etc/mdp.conf` in Audio output to "Headphone" an everythning worked well.

**Second try:**
1, Steps 1. - 4. as above
2. I chose Spotify [y]
Result
Nothing worked =( I can see the interface on my desktop PC by typing the IP of the Phoniebox, but when i Push play, they didnt start and there was no Sound. I also changed audio output as before to "Headphone" but there was no different.

**Actual try:**
1. Install RasberianBuster Desktop Version on SD Card2. Creating SSH file and used putty for installation)
2. I used [this](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/docs/SPOTIFY-INTEGRATION.md) for the installation. (Installing mopidy f.e.)
3.  I choose iFace [y] and MDP [n] (like in the discription, corecct?)
**Result:**
As the second try, nothing worked. I posted you the stuf u asked me for. Now, in addition, here is the second part u wanted.

`aplay -l`

login as: pi
pi@192.168.178.58's password:
Linux raspberrypi 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Thu Jul  9 15:22:34 2020
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
pi@raspberrypi:~ $

s-martin, 2020-07-10:
I think you need to write "Headphone**s**" with the**s** at the end.

Cali227, 2020-07-10:
Nothing changed =(
"Headphone" without s worked in my first try btw.
I think there is a bigger problem because its not possible to start any song on the phoniebox

s-martin, 2020-07-10:
Did you reboot after the change to Headphones?

Cali227, 2020-07-10:
You mean in sudo nano /etc/mopidy/mopidy.conf?
yes

Cali227, 2020-07-10:
I dont understand, why the version without spotify works and with spotify not?...

s-martin, 2020-07-10:
Spotify uses mopidy for audio control, Classic edition uses MPD.

That's two  different tools.

Cali227, 2020-07-10:
I understand. But you dont see the problem i have?

s-martin, 2020-07-10:
Yes, it's a configuration issue and step by step we need to find the correct settings.

I don't use Spotify, so sorry, if it's a little searching around.

Cali227, 2020-07-10:
OK :D What i have to do?

Cali227, 2020-07-11:
What can I do for this? Any ideas? =)

s-martin, 2020-07-11:
I tried to sum up changes in our FAQ https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#raspberry-pi-update-may-2020-broke-my-audio

Could you try that?

Cali227, 2020-07-12:
For sure, so lets go:

**1)** Typing`alsa mixer`
**Result:**
`lqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqq AlsaMixer v1.1.8 qqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqk
x Card: bcm2835 Headphones                                                                                                          F1:  Help               x
x Chip: Broadcom Mixer                                                                                                              F2:  System information x
x View: F3:[Playback] F4: Capture  F5: All                                                                                          F6:  Select sound card  x
x Item: Headphone [dB gain: -20.00]                                                                                                 Esc: Exit               x
x                                                                                                                                                           x
x                                                                                                                                                           x
x                                                                                                                                                           x
x                                                                                                                                                           x
x                                                                                                                                                           x
x                                                                                                                                                           x
x                                                                                                                                                           x
x                                                                                                                                                           x
x                                                                                                                                                           x
x                                                                                                                                                           x
x                                                                           lqqk                                                                            x
x                                                                           x  x                                                                            x
x                                                                           x  x                                                                            x
x                                                                           x  x                                                                            x
x                                                                           x  x                                                                            x
x                                                                           x  x                                                                            x
x                                                                           x  x                                                                            x
x                                                                           x  x                                                                            x
x                                                                           x  x                                                                            x
x                                                                           x  x                                                                            x
x                                                                           x  x                                                                            x
x                                                                           x  x                                                                            x
x                                                                           x  x                                                                            x
x                                                                           x  x                                                                            x
x                                                                           x  x                                                                            x
x                                                                           x  x                                                                            x
x                                                                           x  x                                                                            x
x                                                                           x  x                                                                            x
x                                                                           xaax                                                                            x
[... cut ...]
`card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3`

----
I have to say, that i realy do not understand all of this, but I think you do. I Try my best =)

s-martin, 2020-07-12:
Ok, so it’s Headphone

Please post the outputs of

`cat /etc/mpd.conf`

`cat /home/pi/RPi-Jukebox-RFID/settings/global.conf` 

`cat /home/pi/RPi-Jukebox-RFID/settings/Audio_iFace_Name`

`cat /etc/asound.conf`

`cat /etc/mopidy/mopidy.conf`

s-martin, 2020-07-12:
Oh, and please post 

`cat /home/pi/RPi-Jukebox-RFID/settings/version-number`

Cali227, 2020-07-12:
All right;
1) `sudo cat /etc/mpd.conf`
` # An example configuration file for MPD.
# Read the user manual for documentation: http://www.musicpd.org/doc/user/
# or /usr/share/doc/mpd/html/user.html

# Files and directories #######################################################
#
# This setting controls the top directory which MPD will search to discover the
# available audio files and add them to the daemon's online database. This
# setting defaults to the XDG directory, otherwise the music directory will be
# be disabled and audio files will only be accepted over ipc socket (using
# file:// protocol) or streaming files over an accepted protocol.
#
music_directory         "/var/lib/mpd/music"
#
# This setting sets the MPD internal playlist directory. The purpose of this
# directory is storage for playlists created by MPD. The server will use
# playlist files not created by the server but only if they are in the MPD
# format. This setting defaults to playlist saving being disabled.
#
playlist_directory              "/var/lib/mpd/playlists"
#
# This setting sets the location of the MPD database. This file is used to
# load the database at server start up and store the database while the
# server is not up. This setting defaults to disabled which will allow
# MPD to accept files over ipc socket (using file:// protocol) or streaming
# files over an accepted protocol.
#
db_file                 "/var/lib/mpd/tag_cache"
#
# These settings are the locations for the daemon log files for the daemon.
# These logs are great for troubleshooting, depending on your log_level
# settings.
#
# The special value "syslog" makes MPD use the local syslog daemon. This
# setting defaults to logging to syslog, or to journal if mpd was started as
# a systemd service.
#
log_file                        "/var/log/mpd/mpd.log"
#
# This setting sets the location of the file which stores the process ID
# for use of mpd --kill and some init scripts. This setting is disabled by
# default and the pid file will not be stored.
#
pid_file                        "/run/mpd/pid"
#
# This setting sets the location of the file which contains information about
# most variables to get MPD back into the same general shape it was in before
# it was brought down. This setting is disabled by default and the server
# state will be reset on server start up.
#
state_file                      "/var/lib/mpd/state"
#
# The location of the sticker database.  This is a database which
# manages dynamic information attached to songs.
#
sticker_file                   "/var/lib/mpd/sticker.sql"
#
###############################################################################

# General music daemon options ################################################
#
# This setting specifies the user that MPD will run as. MPD should never run as
# root and you may use this setting to make MPD change its user ID after
# initialization. This setting is disabled by default and MPD is run as the
# current user.
#
user                            "mpd"
#
[... 398 lines cut ...]
hostname = 0.0.0.0

[iris]
country = DE
locale = de_DE

[spotify]
enabled = true
username = jonasleisse@gmx.de
password = gjogjo11
client_id = 1f23b4ae-6e71-4616-baff-dbdab36f9b7c
client_secret = ywFmKYGdrV-h2InNxWRWKqVQAixyK8E9Vv1yr_COaSc=
#bitrate = 160
#volume_normalization = true
#private_session = false
#timeout = 10
#allow_cache = true
#allow_network = true
#allow_playlists = true
#search_album_count = 20
#search_artist_count = 10
#search_track_count = 50
#toplist_countries =`

6) Its `2.0`

s-martin, 2020-07-12:
use `sudo nano /home/pi/RPi-Jukebox-RFID/settings/global.conf`
and set `AUDIOIFACENAME` to `Headphone`.
Save and exit the editor nano with `CTRL + O` and `CTRL + X`.

use `sudo nano /home/pi/RPi-Jukebox-RFID/settings/Audio_iFace_Name`
and replace `PCM` with `Headphone`.

use `sudo nano /etc/mopidy/mopidy.conf`
and set |output` to `alsasink`.

Reboot.

s-martin, 2020-07-12:
Ah and to use Spotify you need to add your account data under the `[spotify]` section, i.e. user, password, client_id, client_secret.

Cali227, 2020-07-12:
Ok, i changed everythink, but stil no sound =(

This are my steps:
1) Going into Phoniebox on PI by entering the IP adress (Starting volume is 0% btw, i change it manual to 100%)
2) click on "Spotify"
3)Click on a playlist
4) Cop URL Link
5) Switch back to Phniebox and click on "Card ID"
6) Scan a new card
7) Edit the URL Link under "Stream URL" and select "Spotify" as Type
8) Submit
9) Go to "Player" (and change Volume again to 100%)
10) Scaning ID Card
11) The picture appears but thats all. I can click on Play, but nothing happens.

s-martin, 2020-07-12:
@Cali227 I've deleted your comment, because the Spotify data was visible in the history. You should change the Spotify password also!
Sorry didn't realize that.

s-martin, 2020-07-12:
Please post the output of `journalctl -u mopidy`

Cali227, 2020-07-12:
Yeah i saw it =) I thought i deleted it myselve, thank you.

So:
`journalctl -u mopidy`

```
-- Logs begin at Sun 2020-07-12 10:39:13 CEST, end at Sun 2020-07-12 10:55:00 CEST. --
Jul 12 10:39:13 raspberrypi mopidy[5415]:   File "/usr/lib/python3/dist-packages/pykka/_threading.py", line 45, in get
Jul 12 10:39:13 raspberrypi mopidy[5415]:     _compat.reraise(*self._data['exc_info'])
Jul 12 10:39:13 raspberrypi mopidy[5415]:   File "/usr/lib/python3/dist-packages/pykka/_compat/__init__.py", line 29, in reraise
Jul 12 10:39:13 raspberrypi mopidy[5415]:     raise value
Jul 12 10:39:13 raspberrypi mopidy[5415]:   File "/usr/lib/python3/dist-packages/pykka/_ref.py", line 107, in ask
Jul 12 10:39:13 raspberrypi mopidy[5415]:     raise ActorDeadError('{} not found'.format(self))
Jul 12 10:39:13 raspberrypi mopidy[5415]: pykka._exceptions.ActorDeadError: Audio (urn:uuid:72c62911-1ce0-4da9-8a5e-22ea6f761d93) not found
Jul 12 10:39:13 raspberrypi mopidy[5415]: From cffi callback <function _SessionCallbacks.music_delivery at 0x74278930>:
Jul 12 10:39:13 raspberrypi mopidy[5415]: Traceback (most recent call last):
Jul 12 10:39:13 raspberrypi mopidy[5415]:   File "/usr/lib/python3/dist-packages/spotify/session.py", line 1101, in music_delivery
Jul 12 10:39:13 raspberrypi mopidy[5415]:     num_frames,
Jul 12 10:39:13 raspberrypi mopidy[5415]:   File "/usr/lib/python3/dist-packages/spotify/utils.py", line 98, in call
Jul 12 10:39:13 raspberrypi mopidy[5415]:     return listener.callback(*args)
Jul 12 10:39:13 raspberrypi mopidy[5415]:   File "/usr/lib/python3/dist-packages/mopidy_spotify/playback.py", line 191, in music_delivery_callback
Jul 12 10:39:13 raspberrypi mopidy[5415]:     consumed = audio_actor.emit_data(buffer_).get()
Jul 12 10:39:13 raspberrypi mopidy[5415]:   File "/usr/lib/python3/dist-packages/pykka/_threading.py", line 45, in get
Jul 12 10:39:13 raspberrypi mopidy[5415]:     _compat.reraise(*self._data['exc_info'])
Jul 12 10:39:13 raspberrypi mopidy[5415]:   File "/usr/lib/python3/dist-packages/pykka/_compat/__init__.py", line 29, in reraise
Jul 12 10:39:13 raspberrypi mopidy[5415]:     raise value
Jul 12 10:39:13 raspberrypi mopidy[5415]:   File "/usr/lib/python3/dist-packages/pykka/_ref.py", line 107, in ask
Jul 12 10:39:13 raspberrypi mopidy[5415]:     raise ActorDeadError('{} not found'.format(self))
Jul 12 10:39:13 raspberrypi mopidy[5415]: pykka._exceptions.ActorDeadError: Audio (urn:uuid:72c62911-1ce0-4da9-8a5e-22ea6f761d93) not found
Jul 12 10:39:13 raspberrypi mopidy[5415]: From cffi callback <function _SessionCallbacks.music_delivery at 0x74278930>:
Jul 12 10:39:13 raspberrypi mopidy[5415]: Traceback (most recent call last):
Jul 12 10:39:13 raspberrypi mopidy[5415]:   File "/usr/lib/python3/dist-packages/spotify/session.py", line 1101, in music_delivery
Jul 12 10:39:13 raspberrypi mopidy[5415]:     num_frames,
Jul 12 10:39:13 raspberrypi mopidy[5415]:   File "/usr/lib/python3/dist-packages/spotify/utils.py", line 98, in call
Jul 12 10:39:13 raspberrypi mopidy[5415]:     return listener.callback(*args)
Jul 12 10:39:13 raspberrypi mopidy[5415]:   File "/usr/lib/python3/dist-packages/mopidy_spotify/playback.py", line 191, in music_delivery_callback
Jul 12 10:39:13 raspberrypi mopidy[5415]:     consumed = audio_actor.emit_data(buffer_).get()
Jul 12 10:39:13 raspberrypi mopidy[5415]:   File "/usr/lib/python3/dist-packages/pykka/_threading.py", line 45, in get
Jul 12 10:39:13 raspberrypi mopidy[5415]:     _compat.reraise(*self._data['exc_info'])
Jul 12 10:39:13 raspberrypi mopidy[5415]:   File "/usr/lib/python3/dist-packages/pykka/_compat/__init__.py", line 29, in reraise
Jul 12 10:39:13 raspberrypi mopidy[5415]:     raise value
Jul 12 10:39:13 raspberrypi mopidy[5415]:   File "/usr/lib/python3/dist-packages/pykka/_ref.py", line 107, in ask
Jul 12 10:39:13 raspberrypi mopidy[5415]:     raise ActorDeadError('{} not found'.format(self))
Jul 12 10:39:13 raspberrypi mopidy[5415]: pykka._exceptions.ActorDeadError: Audio (urn:uuid:72c62911-1ce0-4da9-8a5e-22ea6f761d93) not found
Jul 12 10:39:13 raspberrypi mopidy[5415]: From cffi callback <function _SessionCallbacks.music_delivery at 0x74278930>:
Jul 12 10:39:13 raspberrypi mopidy[5415]: Traceback (most recent call last):
Jul 12 10:39:13 raspberrypi mopidy[5415]:   File "/usr/lib/python3/dist-packages/spotify/session.py", line 1101, in music_delivery
Jul 12 10:39:13 raspberrypi mopidy[5415]:     num_frames,
Jul 12 10:39:13 raspberrypi mopidy[5415]:   File "/usr/lib/python3/dist-packages/spotify/utils.py", line 98, in call
Jul 12 10:39:13 raspberrypi mopidy[5415]:     return listener.callback(*args)
Jul 12 10:39:13 raspberrypi mopidy[5415]:   File "/usr/lib/python3/dist-packages/mopidy_spotify/playback.py", line 191, in music_delivery_callback
Jul 12 10:39:13 raspberrypi mopidy[5415]:     consumed = audio_actor.emit_data(buffer_).get()
Jul 12 10:39:13 raspberrypi mopidy[5415]:   File "/usr/lib/python3/dist-packages/pykka/_threading.py", line 45, in get
Jul 12 10:39:13 raspberrypi mopidy[5415]:     _compat.reraise(*self._data['exc_info'])
Jul 12 10:39:13 raspberrypi mopidy[5415]:   File "/usr/lib/python3/dist-packages/pykka/_compat/__init__.py", line 29, in reraise
Jul 12 10:39:13 raspberrypi mopidy[5415]:     raise value
Jul 12 10:39:13 raspberrypi mopidy[5415]:   File "/usr/lib/python3/dist-packages/pykka/_ref.py", line 107, in ask
Jul 12 10:39:13 raspberrypi mopidy[5415]:     raise ActorDeadError('{} not found'.format(self))
[... cut ...]
Jul 12 10:39:13 raspberrypi mopidy[5415]: From cffi callback <function _SessionCallbacks.music_delivery at 0x74278930>:
Jul 12 10:39:13 raspberrypi mopidy[5415]: Traceback (most recent call last):
Jul 12 10:39:13 raspberrypi mopidy[5415]:   File "/usr/lib/python3/dist-packages/spotify/session.py", line 1101, in music_delivery
Jul 12 10:39:13 raspberrypi mopidy[5415]:     num_frames,
Jul 12 10:39:13 raspberrypi mopidy[5415]:   File "/usr/lib/python3/dist-packages/spotify/utils.py", line 98, in call
Jul 12 10:39:13 raspberrypi mopidy[5415]:     return listener.callback(*args)
```

s-martin, 2020-07-12:
Looks like there's a typo or something like that somewhere.

Could you post again the current mopidy.conf file **without the Spotify credentials**?

Cali227, 2020-07-12:
Sure.

```
[core]
cache_dir = /var/cache/mopidy
config_dir = /etc/mopidy
data_dir = /var/lib/mopidy

[logging]
config_file = /etc/mopidy/logging.conf
debug_file = /var/log/mopidy/mopidy-debug.log

[local]
enabled = true
media_dir = /home/pi/RPi-Jukebox-RFID/shared/audiofolders
excluded_file_extensions =
  .conf
  .jpg
  .txt
  placeholder

[file]
#enabled = true
metadata_timeout = 1

[m3u]
playlists_dir = /home/pi/RPi-Jukebox-RFID/playlists
default_encoding = UTF-8
default_extension = .m3u

[audio]
output = alsaink
mixer_volume = 90

[mpd]
hostname = 0.0.0.0

[http]
hostname = 0.0.0.0

[iris]
country = DE
locale = de_DE

[spotify]
enabled = true
username = asd
password = asd
client_id = asd
client_secret = asd
#bitrate = 160
#volume_normalization = true
#private_session = false
#timeout = 10
#allow_cache = true
#allow_network = true
#allow_playlists = true
#search_album_count = 20
```

s-martin, 2020-07-12:
Sorry, I edited my previous comment

Cali227, 2020-07-12:
Ah i see the first typo. ala**s**ink

i try it again correct

Cali227, 2020-07-12:
> his `pykka._exceptions.ActorDeadError: Audio (urn:uuid:72c62911-1ce0-4da9-8a5e-22ea6f761d93) not found` looks like some Spotify address is wrong.

You mean ID or client?

s-martin, 2020-07-12:
> Ah i see the first typo. ala**s**ink
> 
> i try it again correct

Yeah, please try that first.

After the edit you need to restart mopidy with ` sudo systemctl restart mopidy`.

Cali227, 2020-07-12:
Done, nothing changed.

Cali227, 2020-07-12:
The track is runnig now, but without sound

s-martin, 2020-07-12:
What do you use as audio output?

In a previous comment you mentioned the volume was 0 after reboot, is this still the case?

Cali227, 2020-07-12:
The volume is 90% now!
My audio output are this [https://www.amazon.de/Trust-Lautsprecher-USB-Stromversorgung-Smartphone-schwarz/dp/B00JRW0M32/ref=as_li_ss_tl?s=computers&ie=UTF8&qid=1542459671&sr=1-1&keywords=trust+leto&linkCode=sl1&tag=welovetech-21&linkId=81f87a80899f024bb73a434adae5673f&language=de_DE](url)

Cali227, 2020-07-14:
Any ideas? :-)

Cali227, 2020-07-17:
I dont want to annoy you, but do you stil have more ideas? =))

s-martin, 2020-07-19:
Is probably an issue in mopidy (see https://github.com/mopidy/mopidy/issues/1924) and according to that ticket it will be fixed in the next mopidy version 3.0.3.

So at the moment you probably have to wait until that is released, unfortunately I'm not aware of any workaround.

Cali227, 2020-07-20:
Ok, then i will wait for it. Thank you for your help =))

MiczFlor, 2020-07-24:
Hi @Cali227 

looks like to you have been digging very deep into programming for somebody with no programming background :)

I am always a bit lost what to do with problems like yours. Many people seem to have their Spotify Phoniebox running, but yours does not seem to work. And it's very difficult to narrow down the problem from a distance.

Do you think you can work with the Classic edition for the time being? (save the planet for your daughter, too, by not streaming all the time, which is consuming a lot of electricity...). And/or try a totally fresh install? If it really is the `mopidy` issue as @s-martin suggests, we will find out once they fixed that at their end. 

Until then, I would really like to close this issue, because it is very unique and does not help others.

Cali227, 2020-07-24:
Hey Miczflor,
Yes you are absolute right, i have no experience in programming 🙃

I realy would like to get the Spotify Version running. There are So many Songs and playlists my daughter Listen on Spotify every Day. 

Sure i can make a reinstall, as i did so many Times before, but i think that would not change something. 

If there is any oppertunity that you can help me, i would thank you so much!!!

ndd, 2020-08-05:
I've got exactly the same issues as described by @Cali227.
Tracks are running, no sound at all. It's a fresh Raspberry Pi OS installation as well as a fresh Phoniebox installation (One line script installation; Master branch; latest version).

Sound is working with YT videos, start-up sound etc.

s-martin, 2020-08-05:
> I've got exactly the same issues as described by @Cali227.
> Tracks are running, no sound at all. It's a fresh Raspberry Pi OS installation as well as a fresh Phoniebox installation (One line script installation; Master branch; latest version).
> 
> Sound is working with YT videos, start-up sound etc.

Please post the result of `journalctl -u mopidy`

fbneu, 2020-12-09:
Had the same issue, but only when I used the URI of private playlists. 
If you run on public playlists, albums etc. it works fine. Don't know whats the background of this, but I think this is a spotify-thing, not RPi-Jukebox.

closed 2020-12-15 by s-martin