MiczFlor/RPi-Jukebox-RFID #1039: Short Spotify tracks doesn‘t work

Simse90, 2020-07-09:
Spotify tracks which are smaller than 60 seconds doesn‘t work. I tried it with several tracks and playlists and have on all the same behaviour. 
Offline tracks are working pretty good. 
It doesn’t matter if I play them from a linked RFID card or via the Webinterface Iris. 

I tried it with my Phoniebox and a second raspberry with a second SD card and another Spotify premium account (complete new system) and it has the same behaviour. 

I think this issue is related to changes from Spotify. 
Another person have exactly geb same problem like me. 

Would be great if this problem could be fixed. 

Thanks

s-martin, 2020-07-09:
Maybe related to #1034

s-martin, 2020-07-09:
Please close this, if you think this is a duplicate of #1034.

closed 2020-07-09 by Simse90