MiczFlor/RPi-Jukebox-RFID #1040: 🐛 | MPD won't play mp3 file archive (in a renamed .zip)

theswoosh, 2020-07-11:
## Bug (or feature request?)
MPD (mpc) wont' play archived mp3-files
mpd -V shows archive plugins loaded:

Archive plugins:
 [bz2] bz2
 [zzip] zip
 [iso] iso

### What I did
I've used 7z to merge a bundle of MP3 into one zip-file and changed the file format to mp3.
On Windows the usual player (vlc, wmp) are fine and play the archive without issues
The mpd however just outputs a short noise ans skips the track, but is able to get it's length. 

### What happened

7z a target.zip <file1.mp3> <file2.mp3>
ren target.zip target.mp3

### I expected this to happen
Archived (stored) mp3 should be playable

## Software

### Base image and version
BetriebssystemRaspbian
BeschreibungRaspbian GNU/Linux 10 (buster)
Release10
Codenamebuster

### Branch / Release
master

### RaspberryPi version
3b+ rev 1.3

s-martin, 2020-07-11:
Until now this was not used, so I would say it's a feature request.

I don't know, if MOD would support this, maybe it's just a configuration task.

Anyone with experience is welcome to help.

MiczFlor, 2020-07-27:
Hi @theswoosh 
what you are doing, what's the benefit of handling multiple MP3 files that way?

theswoosh, 2020-07-27:
Hi @MiczFlor 
overall I try to reduce the number of RFID-Cards floating around :)
For some CDs the whole story is splitted into separate files (subtracks), my kids however want to skip certain files without having to press the skip track button for each subtrack. By zipping them I have access to the original files and would be able to provide that feature.
For now I have a python script that extracts all files and binary copies them to one big file like the archive is (DOS CMD copy / ...)

MiczFlor, 2020-07-27:
The RFID cards are an issue :) I have been working with a barcode scanner recently. Not so neat, but easier to expand and adjust and cheaper... And you can make little posters with illustrations for stories and albums.

I will look into this archiving issue with `mpd`. I never heard of this before. Thanks for a new thought :)

MiczFlor, 2020-07-27:
I found this:
https://github.com/KaOSx/main/issues/26

*After installation of mpd-0.19.16-2 and updating the library, Cantata displays (and plays) the content of zip archives in my music folder in the play list and the folder view. Thanks a lot.*

Does that help?