MiczFlor/RPi-Jukebox-RFID #1042: RFID card scans cards continuous

gazza7, 2020-07-24:
When I scan a card I get multiple outputs, so for example if I scan to move onto next track it will move on two or three tracks.
Is there a way of stopping this happening.  See below from syslog as you can see all the triggers have the same time of execution.

Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: 2020-07-24 20:24:01,146 - Reader - INFO - Card detected.
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: 2020-07-24 20:24:01,152 - Reader - INFO - 25130181179
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: 2020-07-24 20:24:01,153 - root - INFO - Trigger Play Cardid=25130181179
Jul 24 20:24:01 raspberrypi mopidy[634]: INFO     [MpdSession-715] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36558
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: Bruce Springsteen - Born to Run
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: [paused]  #4/37   0:00/4:48 (0%)
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: volume:100%   repeat: off   random: off   single: off   consume: off
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: /home/pi/RPi-Jukebox-RFID/scripts/rfid_trigger_play.sh: line 319: [: too many arguments
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: 2020-07-24 20:24:01,342 - Reader - INFO - Card detected.
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: 2020-07-24 20:24:01,346 - Reader - INFO - 25130181179
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: 2020-07-24 20:24:01,346 - root - INFO - Trigger Play Cardid=25130181179
Jul 24 20:24:01 raspberrypi mopidy[634]: INFO     [MpdSession-716] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36562
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: Bruce Springsteen - Tenth Avenue Freeze-Out
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: [playing] #5/37   0:00/4:30 (0%)
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: volume:100%   repeat: off   random: off   single: off   consume: off
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: /home/pi/RPi-Jukebox-RFID/scripts/rfid_trigger_play.sh: line 319: [: too many arguments
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: 2020-07-24 20:24:01,517 - Reader - INFO - Card detected.
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: 2020-07-24 20:24:01,522 - Reader - INFO - 25130181179
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: 2020-07-24 20:24:01,522 - root - INFO - Trigger Play Cardid=25130181179
Jul 24 20:24:01 raspberrypi mopidy[634]: INFO     [MpdSession-717] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36566
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: Bruce Springsteen - Tenth Avenue Freeze-Out
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: [playing] #6/37   0:00/3:10 (0%)
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: volume:100%   repeat: off   random: off   single: off   consume: off
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: /home/pi/RPi-Jukebox-RFID/scripts/rfid_trigger_play.sh: line 319: [: too many arguments
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: 2020-07-24 20:24:01,640 - Reader - INFO - Card detected.
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: 2020-07-24 20:24:01,643 - Reader - INFO - 25130181179
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: 2020-07-24 20:24:01,644 - root - INFO - Trigger Play Cardid=25130181179
Jul 24 20:24:01 raspberrypi mopidy[634]: INFO     [MpdSession-718] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36570
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: Bruce Springsteen - The Promised Land
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: [playing] #7/37   0:00/4:02 (0%)
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: volume:100%   repeat: off   random: off   single: off   consume: off
Jul 24 20:24:01 raspberrypi daemon_rfid_reader.py[618]: /home/pi/RPi-Jukebox-RFID/scripts/rfid_trigger_play.sh: line 319: [: too man

Gazza

gazza7, 2020-07-24:
Hi,
To stop this happening I've put time.sleep(1) in daemon_rfid_reader.py to make it wait before scanning the next card, seems to work a lot better and now skipping to one track at a time. See below.

``
if cardid is not None:
            **time.sleep(1)**
            if cardid != previous_id or (time.time() - previous_time) >= same_id_delay:
                logger.info('Trigger Play Cardid={cardid}'.format(cardid=cardid))
                subprocess.call([dir_path + '/rfid_trigger_play.sh --cardid=' + cardid], shell=True)
                previous_id = cardid
                previous_time = time.time()
            else:
                logger.debug('Ignoring Card id {cardid} due to same-card-delay, delay: {same_id_delay}'.format(
                    cardid=cardid,
                    same_id_delay=same_id_delay
                ))

s-martin, 2020-07-31:
Could you check, if this is fixed, if you use `develop` branch, because PR #1030 is already merged there?

For me it looks like, it's the same behavior.

gazza7, 2020-07-31:
Hi thanks for the reply, I've used the same_id_delay  in daemon_rfid_reader.py and set to 2 seconds and now seems to work.
So I will close this issue.

closed 2020-07-31 by gazza7