MiczFlor/RPi-Jukebox-RFID #1043: Next/Prev skips one track

jlarsch, 2020-07-24:
I started noticing a funny behavior of my phoniebox. Each press on my next/prev buttons advances the playlist by two songs instead of just one. I might have messed up my gpio settings somewhere. Any ideas what might cause this behavior?

You can see in the logs that the command really gets called twice with each press of the button:

########### SCRIPT playout_controls.sh (2020-07-24.20:45:02) ##
########### SCRIPT playout_controls.sh (2020-07-24.20:45:02) ##
VAR COMMAND: playerprev
VAR COMMAND: playerprev
VAR VALUE: 
VAR VALUE:

jlarsch, 2020-07-24:
hmm, i found out how to stop this behavior but I am puzzled by the solution: In the webApp, under Settings > External Devices & Interfaces , I have to DEACTIVATE GPIO Buttons to make the GPIO Buttons work properly.
When `GPIO Buttons` is toggeled ON, I get the skipping issue.
When `GPIO Buttons` is deactiveated, everything is fine.

This seems counter intuitive.

MiczFlor, 2020-07-27:
Hi @jlarsch 
this is not only counter intuitive, it's not how it should be :) did you do a fresh install or upgraded your system with `git pull`?

s-martin, 2020-07-31:
What's the result of `sudo tail -n 300 /var/log/lighttpd/error.log` after you experienced that behavior?

MiczFlor, 2020-08-20:
Hi @jlarsch is this still happening? If not, can you share what had been happening? And if not, please close the ticket.
Thanks for your input (I keep my fingers crossed that the issue is resolved :) nobody else reported it, so I wouldn't even know where to start looking...)

mkiefel, 2020-09-14:
Hi @jlarsch. Were you able to fix this? I just setup a new box with the interactive install script and face the same issues. It almost seems that the service that checks for button presses is running twice. But I haven't dug any deeper. Maybe you also have a hunch of what could go wrong?

mkiefel, 2020-09-14:
It seems that this is related to the debouncing of the buttons in my case. What helped was to start a thread for every press to trigger the event. https://www.raspberrypi.org/forums/viewtopic.php?t=143478 pointed me in that direction.

I still find it surprising that are restarted service with the original code does not exhibit that behavior.

s-martin, 2022-05-29:
Closing. Please feel free to reopen this issue, if the problem persists.

closed 2022-05-29 by s-martin