MiczFlor/RPi-Jukebox-RFID #1045: Volume Randomly Jumps On Its Own with Rotary Encoder (without Breakout)

BzowK, 2020-07-31:
Hey Guys - 

90% done building this for my 2 year old daughter - just have one issue I cannot resolve.  My installation is fairly simple as it's a RasPi3 with the following components:
- MFC522 RFID Reader
- USB powered Speakers
- C-Media USB Audio Card
- [Rotary Encoder without Breakout](https://www.amazon.com/gp/product/B07DM2YMT4/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) (CYT1100)

After config, everything works correctly except the rotary encoder.  When its enabled in Settings, the volume dynamically jumps up and down without me touching it.  I followed all steps in the Wiki [plus these steps](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Audio-RotaryKnobVolume) to install the service and such.  I even tried [commenting out all lines in gpio-buttons.py dealing with volume](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/325) to no avail.  Same thing occurs if i swap the two Out pins.  When it's disabled in settings, the behavior stops, so know it's related to the encoder, itself.

My best guess is that it is because I am using a rotary encoder without breakout therefore has slightly different pins having two grounds, a switch, and 2 outs.  Here's a quick sketch I made of how its attached where the switch & outs are on the default GPIOs.  
[image]

I've looked all through issues and cannot find anything similar.  Below are additional build specs.  Any suggestions would be appreciated.  Thanks!

- OS Distribution: Raspbian
- Description: Raspbian GNU/Linux 10 (buster)
- Release:10
- Codename: Buster
- Version: 2.1rc1 - 6e63024 - develop
- Edition: Plus edition (feat. Spotify integration)

```
pi@lkmusicbox:~ $ service rotary-encoder status
● rotary-encoder.service - Rotary Encoder Service
   Loaded: loaded (/etc/systemd/system/rotary-encoder.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2020-07-31 15:45:43 BST; 1min 51s ago
 Main PID: 532 (rotary-encoder.)
    Tasks: 2 (limit: 2200)
   Memory: 6.6M
   CGroup: /system.slice/rotary-encoder.service
           └─532 /usr/bin/python3 /home/pi/RPi-Jukebox-RFID/scripts/rotary-encoder.py

Jul 31 15:47:34 lkmusicbox rotary-encoder.py[532]: OK
Jul 31 15:47:35 lkmusicbox rotary-encoder.py[532]: OK MPD 0.19.0
Jul 31 15:47:35 lkmusicbox rotary-encoder.py[532]: OK
Jul 31 15:47:35 lkmusicbox rotary-encoder.py[532]: OK
Jul 31 15:47:35 lkmusicbox rotary-encoder.py[532]: OK MPD 0.19.0
Jul 31 15:47:35 lkmusicbox rotary-encoder.py[532]: OK
Jul 31 15:47:35 lkmusicbox rotary-encoder.py[532]: OK
Jul 31 15:47:35 lkmusicbox rotary-encoder.py[532]: OK MPD 0.19.0
Jul 31 15:47:35 lkmusicbox rotary-encoder.py[532]: OK
Jul 31 15:47:35 lkmusicbox rotary-encoder.py[532]: OK
```

s-martin, 2020-07-31:
What's the result of `journalctl -u mopidy`?

BzowK, 2020-07-31:
Thanks for the reply - output is below.  To note, the rotary encoder is disconnected and I removed the service temporarially.  Can hook back up and run again if needed.

```
pi@lkmusicbox:~ $ journalctl -u mopidy
-- Logs begin at Fri 2020-07-31 20:17:01 BST, end at Fri 2020-07-31 20:17:18 BST. --
Jul 31 20:17:05 lkmusicbox systemd[1]: Starting Mopidy music server...
Jul 31 20:17:06 lkmusicbox systemd[1]: Started Mopidy music server.
Jul 31 20:17:11 lkmusicbox mopidy[552]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Jul 31 20:17:11 lkmusicbox mopidy[552]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Jul 31 20:17:11 lkmusicbox mopidy[552]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Jul 31 20:17:11 lkmusicbox mopidy[552]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Jul 31 20:17:11 lkmusicbox mopidy[552]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Jul 31 20:17:11 lkmusicbox mopidy[552]: INFO     [MainThread] mopidy.config Loading config from command line options
Jul 31 20:17:11 lkmusicbox mopidy[552]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Jul 31 20:17:12 lkmusicbox mopidy[552]: INFO     [MainThread] mopidy.__main__ Enabled extensions: stream, iris, softwaremixer, spotify, m3u, http, file, mpd, loca
Jul 31 20:17:12 lkmusicbox mopidy[552]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Jul 31 20:17:14 lkmusicbox mopidy[552]: ERROR    [MainThread] mopidy.__main__ Extension spotify failed during setup. This might have left the registry in a bad st
Jul 31 20:17:14 lkmusicbox mopidy[552]: Traceback (most recent call last):
Jul 31 20:17:14 lkmusicbox mopidy[552]:   File "/usr/local/lib/python3.7/dist-packages/mopidy/__main__.py", line 129, in main
Jul 31 20:17:14 lkmusicbox mopidy[552]:     extension.setup(registry)
Jul 31 20:17:14 lkmusicbox mopidy[552]:   File "/usr/lib/python3/dist-packages/mopidy_spotify/__init__.py", line 49, in setup
Jul 31 20:17:14 lkmusicbox mopidy[552]:     from mopidy_spotify.backend import SpotifyBackend
Jul 31 20:17:14 lkmusicbox mopidy[552]:   File "/usr/lib/python3/dist-packages/mopidy_spotify/backend.py", line 9, in <module>
Jul 31 20:17:14 lkmusicbox mopidy[552]:     from mopidy_spotify import Extension, library, playback, playlists, web
Jul 31 20:17:14 lkmusicbox mopidy[552]:   File "/usr/lib/python3/dist-packages/mopidy_spotify/library.py", line 5, in <module>
Jul 31 20:17:14 lkmusicbox mopidy[552]:     from mopidy_spotify import browse, distinct, images, lookup, search
Jul 31 20:17:14 lkmusicbox mopidy[552]:   File "/usr/lib/python3/dist-packages/mopidy_spotify/browse.py", line 38, in <module>
Jul 31 20:17:14 lkmusicbox mopidy[552]:     "albums": spotify.ToplistType.ALBUMS,
Jul 31 20:17:14 lkmusicbox mopidy[552]: AttributeError: module 'spotify' has no attribute 'ToplistType'
Jul 31 20:17:14 lkmusicbox mopidy[552]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Jul 31 20:17:14 lkmusicbox mopidy[552]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Jul 31 20:17:14 lkmusicbox mopidy[552]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Jul 31 20:17:14 lkmusicbox mopidy[552]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, LocalBackend
Jul 31 20:17:14 lkmusicbox mopidy[552]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Jul 31 20:17:14 lkmusicbox mopidy[552]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: HttpFrontend, IrisFrontend, MpdFrontend
Jul 31 20:17:14 lkmusicbox mopidy[552]: INFO     [HttpFrontend-9] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Jul 31 20:17:14 lkmusicbox mopidy[552]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Jul 31 20:17:14 lkmusicbox mopidy[552]: INFO     [IrisFrontend-10] mopidy_iris.core Starting Iris 3.45.0
Jul 31 20:17:14 lkmusicbox mopidy[552]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Jul 31 20:17:14 lkmusicbox mopidy[552]: INFO     [MainThread] mopidy.commands Starting GLib mainloop

```

Thanks!

s-martin, 2020-08-02:
The mopidy installation looks broken.

Please check with `ls -l ~/phoniebox_logs/` for existing logs of the installation script and post the content of the latest log here.

BzowK, 2020-08-03:
Thanks again for the reply...

Attached is the only log listed.  If performing a 2nd installation from scratch is suggested or would be easier, I'll do so.  To note, when installing rotary encoder service, I found some of the files it mentions adding in [this post](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Audio-RotaryKnobVolume) already existed so wasn't sure which to use (although I tested using both and got the same issue.)  Thanks!
[buster-install-default.sh_20200626_223002.log](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/5017152/buster-install-default.sh_20200626_223002.log)

BzowK, 2020-08-24:
I resolved this by using a different type of rotary encoder - the one recommended.  Thanks

closed 2020-08-24 by BzowK