MiczFlor/RPi-Jukebox-RFID #1046: Cannot get KKmoon Reader to work

BStramke, 2020-07-31:
I have the issue where i cannot see the CardID in the WebUI, but it will be sent to the Console Input directly (just like typing with a keyboard, but not hitting return key). I also changed the Reader.py.kkmoonRFIDreader to Reader.py

> pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ python3 RegisterDevice.py
Choose the reader from list
0 C-Media Electronics Inc. USB PnP Sound Device
1 HID 413d:2107
Device Number: 1

> deviceName.txt
HID 413d:2107

> pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ ls -la /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 80 Jul 31 13:40 .
drwxr-xr-x 4 root root 140 Jul 31 13:40 ..
lrwxrwxrwx 1 root root 9 Jul 31 13:40 usb-413d_2107-event-kbd -> ../event0
lrwxrwxrwx 1 root root 9 Jul 31 13:40 usb-C-Media_Electronics_Inc._USB_PnP_Sound_Device-event-if03 -> ../event1

> pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ lsusb
Bus 001 Device 006: ID 8086:0808 Intel Corp.
Bus 001 Device 007: ID 0781:5571 SanDisk Corp. Cruzer Fit
Bus 001 Device 005: ID 7392:7811 Edimax Technology Co., Ltd EW-7811Un 802.11n Wireless Adapter [Realtek RTL8188CUS]
Bus 001 Device 004: ID 413d:2107
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

> pi@raspberrypi:~ $ tail -n 500 /var/log/syslog
Jul 31 13:40:29 raspberrypi kernel: [    1.356070] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 249)
Jul 31 13:40:29 raspberrypi kernel: [    1.356117] io scheduler mq-deadline registered
Jul 31 13:40:29 raspberrypi kernel: [    1.356149] io scheduler kyber registered
Jul 31 13:40:29 raspberrypi kernel: [    1.359524] bcm2708_fb soc:fb: FB found 1 display(s)
Jul 31 13:40:29 raspberrypi kernel: [    1.376232] Console: switching to colour frame buffer device 90x30
Jul 31 13:40:29 raspberrypi kernel: [    1.385979] bcm2708_fb soc:fb: Registered framebuffer for display 0, size 720x480
Jul 31 13:40:29 raspberrypi kernel: [    1.395577] bcm2835-rng 3f104000.rng: hwrng registered
Jul 31 13:40:29 raspberrypi kernel: [    1.399508] vc-mem: phys_addr:0x00000000 mem_base=0x3ec00000 mem_size:0x40000000(1024 MiB)
Jul 31 13:40:29 raspberrypi kernel: [    1.406940] vc-sm: Videocore shared memory driver
Jul 31 13:40:29 raspberrypi kernel: [    1.411065] gpiomem-bcm2835 3f200000.gpiomem: Initialised: Registers at 0x3f200000
Jul 31 13:40:29 raspberrypi kernel: [    1.431566] brd: module loaded
Jul 31 13:40:29 raspberrypi kernel: [    1.451624] loop: module loaded
Jul 31 13:40:29 raspberrypi kernel: [    1.456964] Loading iSCSI transport class v2.0-870.
Jul 31 13:40:29 raspberrypi kernel: [    1.461737] libphy: Fixed MDIO Bus: probed
Jul 31 13:40:29 raspberrypi kernel: [    1.465371] usbcore: registered new interface driver lan78xx
Jul 31 13:40:29 raspberrypi kernel: [    1.469001] usbcore: registered new interface driver smsc95xx
Jul 31 13:40:29 raspberrypi kernel: [    1.472367] dwc_otg: version 3.00a 10-AUG-2012 (platform bus)
Jul 31 13:40:29 raspberrypi kernel: [    2.204039] Core Release: 2.80a
Jul 31 13:40:29 raspberrypi kernel: [    2.207377] Setting default values for core params
Jul 31 13:40:29 raspberrypi kernel: [    2.210882] Finished setting default values for core params
Jul 31 13:40:29 raspberrypi kernel: [    2.414763] Using Buffer DMA mode
Jul 31 13:40:29 raspberrypi kernel: [    2.418136] Periodic Transfer Interrupt Enhancement - disabled
Jul 31 13:40:29 raspberrypi kernel: [    2.421552] Multiprocessor Interrupt Enhancement - disabled
Jul 31 13:40:29 raspberrypi kernel: [    2.424928] OTG VER PARAM: 0, OTG VER FLAG: 0
Jul 31 13:40:29 raspberrypi kernel: [    2.428319] Dedicated Tx FIFOs mode
Jul 31 13:40:29 raspberrypi kernel: [    2.432214] WARN::dwc_otg_hcd_init:1074: FIQ DMA bounce buffers: virt = b7514000 dma = 0xf7514000 len=9024
Jul 31 13:40:29 raspberrypi kernel: [    2.439162] FIQ FSM acceleration enabled for :
Jul 31 13:40:29 raspberrypi kernel: [    2.439162] Non-periodic Split Transactions
Jul 31 13:40:29 raspberrypi kernel: [    2.439162] Periodic Split Transactions
Jul 31 13:40:29 raspberrypi kernel: [    2.439162] High-Speed Isochronous Endpoints
Jul 31 13:40:29 raspberrypi kernel: [    2.439162] Interrupt/Control Split Transaction hack enabled
Jul 31 13:40:29 raspberrypi kernel: [    2.455467] dwc_otg: Microframe scheduler enabled
Jul 31 13:40:29 raspberrypi kernel: [    2.455558] WARN::hcd_init_fiq:457: FIQ on core 1
Jul 31 13:40:29 raspberrypi kernel: [    2.458676] WARN::hcd_init_fiq:458: FIQ ASM at 806bdcbc length 36
Jul 31 13:40:29 raspberrypi kernel: [    2.461725] WARN::hcd_init_fiq:497: MPHI regs_base at bb810000
Jul 31 13:40:29 raspberrypi kernel: [    2.464735] dwc_otg 3f980000.usb: DWC OTG Controller
Jul 31 13:40:29 raspberrypi kernel: [    2.467809] dwc_otg 3f980000.usb: new USB bus registered, assigned bus number 1
Jul 31 13:40:29 raspberrypi kernel: [    2.470946] dwc_otg 3f980000.usb: irq 56, io mem 0x00000000
Jul 31 13:40:29 raspberrypi kernel: [    2.474034] Init: Port Power? op_state=1
Jul 31 13:40:29 raspberrypi kernel: [    2.477026] Init: Power Port (0)
Jul 31 13:40:29 raspberrypi kernel: [    2.480557] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.04
Jul 31 13:40:29 raspberrypi kernel: [    2.486786] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[... 433 lines cut ...]
Jul 31 13:40:58 raspberrypi systemd[1]: systemd-fsckd.service: Succeeded.
Jul 31 17:00:11 raspberrypi systemd-timesyncd[283]: Synchronized to time server for the first time 193.171.23.163:123 (2.debian.pool.ntp.org).
Jul 31 17:00:11 raspberrypi systemd[1]: Starting Clean php session files...
Jul 31 17:00:12 raspberrypi systemd[1]: phpsessionclean.service: Succeeded.
Jul 31 17:00:12 raspberrypi systemd[1]: Started Clean php session files.
Jul 31 17:00:44 raspberrypi systemd[1]: phoniebox-idle-watchdog.service: Succeeded.
Jul 31 17:01:04 raspberrypi systemd[1]: dev-serial1.device: Job dev-serial1.device/start timed out.
Jul 31 17:01:04 raspberrypi systemd[1]: Timed out waiting for device /dev/serial1.
Jul 31 17:01:04 raspberrypi systemd[1]: Dependency failed for Configure Bluetooth Modems connected by UART.
Jul 31 17:01:04 raspberrypi systemd[1]: hciuart.service: Job hciuart.service/start failed with result 'dependency'.
Jul 31 17:01:04 raspberrypi systemd[1]: dev-serial1.device: Job dev-serial1.device/start failed with result 'timeout'.
Jul 31 17:01:04 raspberrypi systemd[1]: Reached target Multi-User System.
Jul 31 17:01:04 raspberrypi systemd[1]: Starting Update UTMP about System Runlevel Changes...
Jul 31 17:01:04 raspberrypi systemd[1]: systemd-update-utmp-runlevel.service: Succeeded.
Jul 31 17:01:04 raspberrypi systemd[1]: Started Update UTMP about System Runlevel Changes.
Jul 31 17:01:04 raspberrypi systemd[1]: Startup finished in 4.552s (kernel) + 1min 32.381s (userspace) = 1min 36.933s.
Jul 31 17:02:39 raspberrypi systemd[1]: Started Session c2 of user pi.
Jul 31 17:09:01 raspberrypi CRON[1040]: (root) CMD (  [ -x /usr/lib/php/sessionclean ] && if [ ! -d /run/systemd/system ]; then /usr/lib/php/sessionclean; fi)
Jul 31 17:09:28 raspberrypi systemd[1]: Starting Clean php session files...
Jul 31 17:09:28 raspberrypi systemd[1]: phpsessionclean.service: Succeeded.
Jul 31 17:09:28 raspberrypi systemd[1]: Started Clean php session files.
Jul 31 17:14:56 raspberrypi systemd[1]: Starting Cleanup of Temporary Directories...
Jul 31 17:14:56 raspberrypi systemd[1]: systemd-tmpfiles-clean.service: Succeeded.
Jul 31 17:14:56 raspberrypi systemd[1]: Started Cleanup of Temporary Directories.
Jul 31 17:17:01 raspberrypi CRON[1103]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)

BStramke, 2020-07-31:
Output while configuring the card reader: 
>config card  1
config card  2
config card  3
download manual  www.jatsecuritz.com-easzmfßim.html
format description   
 1 ß 28  data format 
  e    qwertz kezboard
   f    ayertz kezboard
    h    quertz kezboard
	config reader ...
	when demand format is appeared plug off reader
	1 8hß10dße
	2 2hß3dß4hß5dße
	3 8hße
	4 6hße
	5 6hß8dße
	6 6hß10dße
	7 4hß5dß4hß5dße
	8 8hß10d
	9 2hß3dß4hß5d
	10 8h
	11 6h
	12 6hß8d
	13 6hß10d
	14 4hß5dß4hß5d
	15 8hß10dßrße
	16 2hß3dß4hß5dßrße
	17 8hßrße
	18 6hßrße
	19 6hß8dßrße
	20 6hß10dßrße
	21 4hß5dß4hß5dßrße
	22 8hß10dßr
	23 2hß3dß4hß5dßr
	24 8hßr
	25 6hßr
	26 6hß8dßr
	27 6hß10dßr
	28 4hß5dß4hß5dßr
	29 e

BStramke, 2020-07-31:
I set it to Format 1, and voila, it worked. This means out of the box not all KKmoon Readers will work and you have to follow the instructions on your configuration card that comes with the Reader. For a Small recap, you have to (First read all the points and make sure you understand them):

1. Open some Text-Editor (Notepad, MCEdit, Nano, ... Can be run in different OS Versions like Windows or on the Raspberry)
2. Swipe your Configuration Card 3 Times in a row. It will display "config card 1", "config card 2" and "config card 3"
3. You will see the Format Text running through. If you want to set it to Format 1 (as I did), you have to unplug your reader when you see the Line "1 8hß10dße"
4. Plugin your Reader (beep), then Plug it out again (less than 2 Seconds!)
5. Plugin your Reader (beep), then Plug it out again (less than 2 Seconds!)
6. Plugin your Reader (beep), then Plug it out again (less than 2 Seconds!)
7. Plugin your Reader. It should now be Configured as Format 1. (yes, you have to unplug it 3 times)

closed 2020-07-31 by BStramke

MiczFlor, 2020-07-31:
Hi @BStramke 
thanks for the input. You solved it and closed it fast :) great. I still took the information and added it to the wiki. Is that ok?:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/RFID_Reader_KKMOON_Info

BStramke, 2020-07-31:
Yep, its fine ;-)