MiczFlor/RPi-Jukebox-RFID #1048: 🐛 | BUG SUMMARY

jayefjay, 2020-08-07:
_##_ Bug

### What I did
I installed the raspberry pi with above mentioned buster image and ran the installer script. Spotify Version.

### What happened
When trying to enable the MQTT function according to https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Smart-Home-remote-control-with-MQTT an error (code=exited, status=1/FAILURE) after trying to start the service with sudo systemctl start phoniebox-mqtt-client.

Starting the .py file directly it throws the following error:
ModuleNotFoundError: No module named 'paho'
pi@raspberrypi:~/RPi-Jukebox-RFID/scripts$ /usr/bin/python3 /home/pi/RPi-Jukebox-RFID/scripts/daemon-mqtt-client.py
Traceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/scripts/daemon-mqtt-client.py", line 3, in <module>
    import paho.mqtt.client as mqtt
ModuleNotFoundError: No module named 'paho'

Installing the package 

sudo apt-get install python3-paho-mqtt

fixed this problem. This should be added to the documentation, but I am hasitant to just do so in the above referenced wiki page.

s-martin, 2020-08-07:
In the wiki is a link to the Readme of the mqtt component, which covers the installation:

https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components/smart-home-automation/MQTT-protocol

s-martin, 2020-08-16:
I added some more pointers in the wiki under the section **Installation**, so I close this for now.

closed 2020-08-16 by s-martin