MiczFlor/RPi-Jukebox-RFID #1049: Enable 'Stop on removal with USB RFID Reader' with v2's 'GPIO control'

thiesschneider, 2020-08-08:
Hey there,

for the old gpio.py there was a possibility to enable the [stop on removal of RFID Cards](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Stop-on-removal-with-USB-RFID-Reader) within the file. Now I am not sure how to enable this with the [GPIO control ini file](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/components/gpio_control/README.md).

Can anybody help me with this?

thx and cheers

manuelbachl, 2020-08-11:
I think the lines you're searching for are those:

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/a5edf05bcd653f485c25cd0663eeda9d727b82cf/misc/sampleconfigs/gpio-buttons.py.sample#L137
`-> uncomment that line and set the proper pin of your RFID reader`

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/a5edf05bcd653f485c25cd0663eeda9d727b82cf/misc/sampleconfigs/gpio-buttons.py.sample#L157
`-> uncomment that line`

and

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/a5edf05bcd653f485c25cd0663eeda9d727b82cf/misc/sampleconfigs/gpio-buttons.py.sample#L169
`-> uncomment that line`

It requires you to have that sample copied and renamed to `/home/RPi-Jukebox-RFID/gpio-buttons.py` and to use that as your config.

Hope that helps
Cheers

thiesschneider, 2020-08-13:
Hey @manuelbachl, thx for your tips, but I think this does not help for the new GPIO control ini file from version 2.0.

@veloxidSchweiz, as you made the pullrequest afaik, could you give me a hint?

veloxidSchweiz, 2020-08-15:
@thiesschneider : You will probably just need to edit the ini file.
Add something like:
'''[RFIDDevice]

enable: True
Type: Button
Pin: 27
pull_up: True
functionCall: functionCallrfidoff_action
'''
And add the 'functionCallrfidoff_action'  to the  function_calls file:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/components/gpio_control/function_calls.py

If this doesnt work we would need to add another Button Class in
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/components/gpio_control/GPIODevices/simple_button.py

I'll think about it in the next days in the next days.

thiesschneider, 2020-08-18:
Ah nice, thanks! As I am on vacation I will try it out, when I come back! 
Would be cool if this feature works 👍

dustMason, 2020-09-10:
Not sure if this helps your specific situation, but I ended up solving this a different way for my pn532 reader. You can see the changes in my fork https://github.com/dustMason/RPi-Jukebox-RFID

scharc, 2020-12-06:
Got this working with this added to gpio_settings.ini:
```
[RFIDDevice]
enable: True
Type: Button
Pin: 21
pull_up: True
functionCall: functionCallPlayerStop
```
I changed the Pin from 6 to 21

closed 2021-01-01 by s-martin