MiczFlor/RPi-Jukebox-RFID #1055: RFID Reader does not play audio

sandreas, 2020-08-13:
## Bug

### What I did
Installed a fresh and clean Phoniebox via default install script (`master` branch). Everything works but scanning an RFID card does not play the music that is linked with it.

The RFID Reader (Neuftech) itself works. Scanning cards result in writing the correct RFID to the login prompt, or, if logged in, to the command line.

Also the web interface is showing the last scanned card, but after linking it to an audio folder, it does not play its contents. 
Normal play over web interface works.

Help with this would be awesome.

## Software

### Base image and version
I used `2020-05-27-raspios-buster-lite-armhf.img`
```
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

### Branch / Release
`master`

### Installscript
`scripts/installscripts/buster-install-default.sh`

## Hardware

### RaspberryPi version
`Raspberry Pi 3 Model B Rev 1.2`

### RFID Reader
`16c0:27db HXGCoLtd Keyboard`

```
lsusb
Bus 001 Device 004: ID 16c0:27db Van Ooijen Technische Informatica Keyboard
```
-->

### Soundcard
Hifiberry Mini Amp

### Other notable hardware
Hifiberry Mini Amp requires gpio buttons to be changed:
GPIO Buttons:
```
btn_shut = 27 #3
btn_vol0 = 13
btn_volU = 16
btn_volD = 28 #19
btn_next = 26
btn_prev = 29 #20
btn_halt = 30 #21
```

s-martin, 2020-08-13:
Please post the result of `tail -n 500 /var/log/syslog`

sandreas, 2020-08-14:
Fixed it myself. The problem was, that `/var/log/mpd/mpd.log` was not writable and therefore mpd service could not be started. 

I used the mallet method:
```
sudo chmod 0777 /var/log/mpd/mpd.log
```

I'll leave the ticket open, because this seems to be something, that should be fixed?! If not, feel free to close the ticket.

s-martin, 2020-08-16:
PR #996 should fix this.

closed 2020-08-20 by MiczFlor