MiczFlor/RPi-Jukebox-RFID #1065: Phoniebox doesn't recognize commands when known WiFi isn't available

morres83, 2020-09-11:
Hello,
I've built the Oled-Phoniebox which works really great so far. Recently I took the phoniebox to our vacation and of course there is no known WiFi network available.
I recognized that the phoniebox doesn't react to any pushbuttons or card swipes for some minutes. After that time it works as usual.

With any known network available m, the phoniebox can be used after about 20 seconds.
Does anybody else have this behavior?

Thanks
morres

s-martin, 2020-09-12:
What’s the output of `tail -n 500 /var/log/syslog`?

morres83, 2020-09-19:
Now this is really strange, I have no syslog entries for the entire time without WiFi although the phoniebox was running every single day. See below. How is this possible??

```
Sep  8 02:58:38 starwarsbox systemd[1]: Started OpenBSD Secure Shell server.
Sep  8 02:58:38 starwarsbox systemd[1]: phoniebox-rotary-encoder.service: Service RestartSec=100ms expired, scheduling restart.
Sep  8 02:58:38 starwarsbox systemd[1]: phoniebox-rotary-encoder.service: Scheduled restart job, restart counter is at 1.
Sep  8 02:58:38 starwarsbox systemd[1]: Stopped Phoniebox Rotary Encoder Service.
Sep  8 02:58:38 starwarsbox systemd[1]: Started Phoniebox Rotary Encoder Service.
Sep  8 02:58:38 starwarsbox systemd[608]: phoniebox-rotary-encoder.service: Failed to execute command: No such file or directory
Sep  8 02:58:38 starwarsbox systemd[608]: phoniebox-rotary-encoder.service: Failed at step EXEC spawning /home/pi/RPi-Jukebox-RFID/scripts/rotary-encoder.py: No such file or directory
Sep  8 02:58:38 starwarsbox systemd[1]: phoniebox-rotary-encoder.service: Main process exited, code=exited, status=203/EXEC
Sep  8 02:58:38 starwarsbox systemd[1]: phoniebox-rotary-encoder.service: Failed with result 'exit-code'.
Sep  8 02:58:38 starwarsbox systemd[1]: phoniebox-rotary-encoder.service: Service RestartSec=100ms expired, scheduling restart.
Sep  8 02:58:38 starwarsbox systemd[1]: phoniebox-rotary-encoder.service: Scheduled restart job, restart counter is at 2.
Sep  8 02:58:38 starwarsbox systemd[1]: Stopped Phoniebox Rotary Encoder Service.
Sep  8 02:58:38 starwarsbox systemd[1]: Started Phoniebox Rotary Encoder Service.
Sep  8 02:58:38 starwarsbox systemd[620]: phoniebox-rotary-encoder.service: Failed to execute command: No such file or directory
Sep  8 02:58:38 starwarsbox systemd[620]: phoniebox-rotary-encoder.service: Failed at step EXEC spawning /home/pi/RPi-Jukebox-RFID/scripts/rotary-encoder.py: No such file or directory
Sep  8 02:58:38 starwarsbox systemd[1]: Started Samba NMB Daemon.
Sep  8 02:58:38 starwarsbox systemd[1]: phoniebox-rotary-encoder.service: Main process exited, code=exited, status=203/EXEC
Sep  8 02:58:38 starwarsbox systemd[1]: phoniebox-rotary-encoder.service: Failed with result 'exit-code'.
Sep  8 02:58:38 starwarsbox systemd[1]: Starting Samba SMB Daemon...
Sep  8 02:58:38 starwarsbox daemon_rfid_reader.py[595]: 2020-09-08 02:58:38,650 - root - INFO - Dir_PATH: /home/pi/RPi-Jukebox-RFID/scripts
Sep  8 02:58:38 starwarsbox systemd[1]: phoniebox-rotary-encoder.service: Service RestartSec=100ms expired, scheduling restart.
Sep  8 02:58:38 starwarsbox systemd[1]: phoniebox-rotary-encoder.service: Scheduled restart job, restart counter is at 3.
Sep  8 02:58:38 starwarsbox systemd[1]: Stopped Phoniebox Rotary Encoder Service.
Sep  8 02:58:38 starwarsbox systemd[1]: Started Phoniebox Rotary Encoder Service.
Sep  8 02:58:38 starwarsbox systemd[683]: phoniebox-rotary-encoder.service: Failed to execute command: No such file or directory
Sep  8 02:58:38 starwarsbox systemd[683]: phoniebox-rotary-encoder.service: Failed at step EXEC spawning /home/pi/RPi-Jukebox-RFID/scripts/rotary-encoder.py: No such file or directory
Sep  8 02:58:38 starwarsbox systemd[1]: phoniebox-rotary-encoder.service: Main process exited, code=exited, status=203/EXEC
Sep  8 02:58:38 starwarsbox systemd[1]: phoniebox-rotary-encoder.service: Failed with result 'exit-code'.
Sep  8 02:58:38 starwarsbox systemd[1]: phoniebox-rotary-encoder.service: Service RestartSec=100ms expired, scheduling restart.
Sep  8 02:58:38 starwarsbox systemd[1]: phoniebox-rotary-encoder.service: Scheduled restart job, restart counter is at 4.
Sep  8 02:58:38 starwarsbox systemd[1]: Started Lighttpd Daemon.
Sep  8 02:58:38 starwarsbox systemd[1]: Stopped Phoniebox Rotary Encoder Service.
Sep  8 02:58:38 starwarsbox systemd[1]: Started Phoniebox Rotary Encoder Service.
Sep  8 02:58:38 starwarsbox systemd[685]: phoniebox-rotary-encoder.service: Failed to execute command: No such file or directory
Sep  8 02:58:38 starwarsbox systemd[685]: phoniebox-rotary-encoder.service: Failed at step EXEC spawning /home/pi/RPi-Jukebox-RFID/scripts/rotary-encoder.py: No such file or directory
Sep  8 02:58:38 starwarsbox systemd[1]: phoniebox-rotary-encoder.service: Main process exited, code=exited, status=203/EXEC
Sep  8 02:58:38 starwarsbox systemd[1]: phoniebox-rotary-encoder.service: Failed with result 'exit-code'.
Sep  8 02:58:39 starwarsbox systemd[1]: Started Samba SMB Daemon.
Sep  8 02:58:39 starwarsbox systemd[1]: phoniebox-rotary-encoder.service: Service RestartSec=100ms expired, scheduling restart.
Sep  8 02:58:39 starwarsbox systemd[1]: phoniebox-rotary-encoder.service: Scheduled restart job, restart counter is at 5.
Sep  8 02:58:39 starwarsbox systemd[1]: Stopped Phoniebox Rotary Encoder Service.
Sep  8 02:58:39 starwarsbox systemd[1]: phoniebox-rotary-encoder.service: Start request repeated too quickly.
Sep  8 02:58:39 starwarsbox systemd[1]: phoniebox-rotary-encoder.service: Failed with result 'exit-code'.
Sep  8 02:58:39 starwarsbox systemd[1]: Failed to start Phoniebox Rotary Encoder Service.
Sep  8 02:58:39 starwarsbox systemd[1]: Reached target Multi-User System.
Sep  8 02:58:39 starwarsbox systemd[1]: Reached target Graphical Interface.
Sep  8 02:58:39 starwarsbox systemd[1]: Starting Update UTMP about System Runlevel Changes...
Sep  8 02:58:39 starwarsbox systemd[1]: systemd-update-utmp-runlevel.service: Succeeded.
[... cut ...]
Sep 19 20:05:19 starwarsbox systemd[1]: Starting Daily man-db regeneration...
Sep 19 20:05:20 starwarsbox logrotate[847]: error: skipping "/var/log/mpd/mpd.log" because parent directory has insecure permissions (It's world writable or writable by group which is not "root") Set "su" directive in config file to tell logrotate which user/group should be used for rotation.
Sep 19 20:05:20 starwarsbox systemd[1]: phpsessionclean.service: Succeeded.
Sep 19 20:05:20 starwarsbox systemd[1]: Started Clean php session files.

```

s-martin, 2023-05-09:
Version 2.4 was released a while ago, is this still an issue?

s-martin, 2023-12-05:
Closing as no interaction for a couple of months.
Feel free to reopen the issue, if the problem still occurs.

closed 2023-12-05 by s-martin