MiczFlor/RPi-Jukebox-RFID #1067: No Music played via RFID Card or web interface

j-gimbel, 2020-09-11:
## Bug

### What I did

raspian buster,
PRI-Jukebox version 2.0 - ed888da - master . 
with Spotify.

### What happened

I cannot play music via webinterface or with a RFID card.
    I hear sound when the system boots.
    Spotify Iris plays sounds,
    command line mpg123 also plays sounds.
    Card gets detected and cofigured ( I created a folder "test" with only one file "test.mp3" and connected it to the card.

But when I swipe the card: nothing happens (cards get recognized, but no sound, no play/pause button change.
When I click the play button nothing happens.

In the command line the command "mpc playlist test" lists the playlist.
But "mpc playlist" shows an empty list.
So it seems that somehow mpd does not get the playlist.

But when I manually add the file via "mpc add" I can start the music over the webinterface (not with the card).
Any ideas ?
Maybe the audio files need certain ownerships (I gave the 777 permissions to rule out access issues) .

### I expected this to happen

Play sound when swiping a registerd card.
Play sound when clicking on a sound file in the web interface.

### Further information that might help

-- Logs begin at Fri 2020-09-11 21:42:35 BST, end at Fri 2020-09-11 22:28:57 BST. --
Sep 11 21:43:08 raspberrypi systemd[1]: Starting Mopidy music server...
Sep 11 21:43:11 raspberrypi systemd[1]: Started Mopidy music server.
Sep 11 21:43:58 raspberrypi mopidy[564]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Sep 11 21:43:58 raspberrypi mopidy[564]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Sep 11 21:43:58 raspberrypi mopidy[564]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Sep 11 21:43:58 raspberrypi mopidy[564]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Sep 11 21:43:58 raspberrypi mopidy[564]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Sep 11 21:43:58 raspberrypi mopidy[564]: INFO     [MainThread] mopidy.config Loading config from command line options
Sep 11 21:43:58 raspberrypi mopidy[564]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Sep 11 21:44:02 raspberrypi mopidy[564]: INFO     [MainThread] mopidy.__main__ Enabled extensions: local, file, iris, m3u, http, mpd, softwaremixer, spotify, stream
Sep 11 21:44:02 raspberrypi mopidy[564]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Sep 11 21:44:04 raspberrypi mopidy[564]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Sep 11 21:44:04 raspberrypi mopidy[564]: INFO     [MainThread] mopidy.commands Mixer volume set to 70
Sep 11 21:44:04 raspberrypi mopidy[564]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Sep 11 21:44:04 raspberrypi mopidy[564]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBackend, LocalBackend
Sep 11 21:44:05 raspberrypi mopidy[564]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Sep 11 21:44:07 raspberrypi mopidy[564]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Authorization failed, not attempting Spotify API request. Please get new credentials from https://www.mopidy.com/authenticate and/or restart Mopidy to resolve this problem.
Sep 11 21:44:07 raspberrypi mopidy[564]: ERROR    [SpotifyBackend-6] mopidy_spotify.web OAuth token refresh failed: invalid_client Client not known.
Sep 11 21:44:07 raspberrypi mopidy[564]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Failed to load Spotify user profile
Sep 11 21:44:07 raspberrypi mopidy[564]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Sep 11 21:44:07 raspberrypi mopidy[564]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: IrisFrontend, HttpFrontend, MpdFrontend
Sep 11 21:44:07 raspberrypi mopidy[564]: INFO     [IrisFrontend-11] mopidy_iris.core Starting Iris 3.45.0
Sep 11 21:44:07 raspberrypi mopidy[564]: INFO     [HttpFrontend-13] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Sep 11 21:44:07 raspberrypi mopidy[564]: ERROR    [MainThread] mopidy.commands Frontend (MpdFrontend) initialization error: MPD server startup failed: [Errno 98] Address already in use
Sep 11 21:44:07 raspberrypi mopidy[564]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
Sep 11 21:44:08 raspberrypi mopidy[564]: INFO     [SpotifyEventLoop] mopidy_spotify.backend Logged in to Spotify in online mode
Sep 11 21:48:06 raspberrypi mopidy[564]: INFO     [MainThread] mopidy.commands GLib mainloop got SIGTERM. Exiting...
Sep 11 21:48:06 raspberrypi mopidy[564]: INFO     [MainThread] mopidy.commands Stopping Mopidy frontends
Sep 11 21:48:06 raspberrypi systemd[1]: Stopping Mopidy music server...
Sep 11 21:48:06 raspberrypi mopidy[564]: INFO     [IrisFrontend-11] mopidy_iris.core Stopping Iris
Sep 11 21:48:06 raspberrypi mopidy[564]: INFO     [MainThread] mopidy.commands Stopping Mopidy core
Sep 11 21:48:06 raspberrypi mopidy[564]: INFO     [MainThread] mopidy.commands Stopping Mopidy backends
Sep 11 21:48:06 raspberrypi mopidy[564]: INFO     [MainThread] mopidy.commands Stopping Mopidy audio
Sep 11 21:48:06 raspberrypi mopidy[564]: INFO     [MainThread] mopidy.commands Stopping Mopidy mixer
Sep 11 21:48:07 raspberrypi systemd[1]: mopidy.service: Succeeded.
[... 542 lines cut ...]
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Branch / Release

master

### Installscript
scripts/installscripts/buster-install-default.sh`

## Hardware

### RaspberryPi version

pi zero

### RFID Reader

RC522

### Soundcard

usb sound card

### Other notable hardware

s-martin, 2020-09-12:
You have an authorization error, please check your Spotify credentials:
```
Sep 11 21:48:58 raspberrypi mopidy[2225]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Authorization failed, not attempting Spotify API request. Please get new credentials from https://www.mopidy.com/authenticate and/or restart Mopidy to resolve this problem.
Sep 11 21:48:58 raspberrypi mopidy[2225]: ERROR    [SpotifyBackend-6] mopidy_spotify.web OAuth token refresh failed: invalid_client Client not known.
Sep 11 21:48:58 raspberrypi mopidy[2225]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Failed to load Spotify user profile
```

Indeed you have a rights issue, the webserver lighttpd cannot access the mp3 file:
`Sep 11 21:49:27 raspberrypi lighttpd[946]: error adding /home/pi/RPi-Jukebox-RFID/shared/audiofolders/test/test.mp3: Access denied`

What’s the output of `ls -l  /home/pi/RPi-Jukebox-RFID/shared/audiofolders/test/test.mp3`?

j-gimbel, 2020-09-12:
Sorry, didn't mean to rate this as a bug. I am quite new to github. 
Strangely I was able to use iris and see my music and play it via iris web interface.
I already fixed the file permissions issue (i created and copied the file via command line).
I will upload a new syslog tomorrow.

j-gimbel, 2020-09-13:
Ok, in syslog there was a mopidy error:

Sep 13 19:46:49 raspberrypi mopidy[614]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Authorization failed, not attempting Spotify API request. Please get new credentials from https://www.mopidy.com/authenticate and/or restart Mopidy to resolve this problem.
Sep 13 19:46:49 raspberrypi mopidy[614]: ERROR    [SpotifyBackend-6] mopidy_spotify.web OAuth token refresh failed: invalid_client Client not known.
Sep 13 19:46:49 raspberrypi mopidy[614]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Failed to load Spotify user profile

I authenticated mopidy with spotify and restarted the system.
I played with the phoniebox web clinet and tried to play music with an RFID card and by clicking on mp3 files in the web client.
Then I dumped syslog and journal of mopidy.

Sep 13 20:19:18 raspberrypi kernel: [    3.431314] usb 1-1: New USB device found, idVendor=1a40, idProduct=0101, bcdDevice= 1.11
Sep 13 20:19:18 raspberrypi kernel: [    3.440649] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
Sep 13 20:19:18 raspberrypi kernel: [    3.449632] usb 1-1: Product: USB 2.0 Hub
Sep 13 20:19:18 raspberrypi kernel: [    3.462862] hub 1-1:1.0: USB hub found
Sep 13 20:19:18 raspberrypi kernel: [    3.473535] hub 1-1:1.0: 4 ports detected
Sep 13 20:19:18 raspberrypi kernel: [    3.485500] mmc0: host does not support reading read-only switch, assuming write-enable
Sep 13 20:19:18 raspberrypi kernel: [    3.505936] mmc0: new high speed SDHC card at address aaaa
Sep 13 20:19:18 raspberrypi kernel: [    3.518505] mmcblk0: mmc0:aaaa SL16G 14.8 GiB
Sep 13 20:19:18 raspberrypi kernel: [    3.538162]  mmcblk0: p1 p2
Sep 13 20:19:18 raspberrypi kernel: [    3.588398] mmc1: new high speed SDIO card at address 0001
Sep 13 20:19:18 raspberrypi kernel: [    3.600591] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
Sep 13 20:19:18 raspberrypi kernel: [    3.612701] VFS: Mounted root (ext4 filesystem) readonly on device 179:2.
Sep 13 20:19:18 raspberrypi kernel: [    3.635642] devtmpfs: mounted
Sep 13 20:19:18 raspberrypi kernel: [    3.653250] Freeing unused kernel memory: 480K
Sep 13 20:19:18 raspberrypi kernel: [    3.662888] This architecture does not have kernel memory protection.
Sep 13 20:19:18 raspberrypi kernel: [    3.672139] Run /sbin/init as init process
Sep 13 20:19:18 raspberrypi kernel: [    3.920353] usb 1-1.1: new full-speed USB device number 3 using dwc_otg
Sep 13 20:19:18 raspberrypi kernel: [    4.169828] usb 1-1.1: New USB device found, idVendor=0d8c, idProduct=013c, bcdDevice= 1.00
Sep 13 20:19:18 raspberrypi kernel: [    4.180056] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Sep 13 20:19:18 raspberrypi kernel: [    4.189589] usb 1-1.1: Product: USB PnP Sound Device
Sep 13 20:19:18 raspberrypi kernel: [    4.199127] usb 1-1.1: Manufacturer: C-Media Electronics Inc.      
Sep 13 20:19:18 raspberrypi kernel: [    4.221397] input: C-Media Electronics Inc.       USB PnP Sound Device as /devices/platform/soc/20980000.usb/usb1/1-1/1-1.1/1-1.1:1.3/0003:0D8C:013C.0001/input/input0
Sep 13 20:19:18 raspberrypi kernel: [    4.300919] hid-generic 0003:0D8C:013C.0001: input,hidraw0: USB HID v1.00 Device [C-Media Electronics Inc.       USB PnP Sound Device] on usb-20980000.usb-1.1/input3
Sep 13 20:19:18 raspberrypi kernel: [    4.520340] usb 1-1.2: new high-speed USB device number 4 using dwc_otg
Sep 13 20:19:18 raspberrypi kernel: [    4.762832] usb 1-1.2: New USB device found, idVendor=0bda, idProduct=8152, bcdDevice=20.00
Sep 13 20:19:18 raspberrypi kernel: [    4.773412] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 13 20:19:18 raspberrypi kernel: [    4.783715] usb 1-1.2: Product: USB 10/100 LAN
Sep 13 20:19:18 raspberrypi kernel: [    4.794285] usb 1-1.2: Manufacturer: Realtek
Sep 13 20:19:18 raspberrypi kernel: [    4.804428] usb 1-1.2: SerialNumber: 00E04D3AA074
Sep 13 20:19:18 raspberrypi kernel: [    4.910770] NET: Registered protocol family 10
Sep 13 20:19:18 raspberrypi kernel: [    4.924104] Segment Routing with IPv6
Sep 13 20:19:18 raspberrypi kernel: [    8.847819] random: systemd: uninitialized urandom read (16 bytes read)
Sep 13 20:19:18 raspberrypi kernel: [    8.911257] random: systemd: uninitialized urandom read (16 bytes read)
Sep 13 20:19:18 raspberrypi kernel: [    8.955639] random: systemd: uninitialized urandom read (16 bytes read)
Sep 13 20:19:18 raspberrypi systemd[1]: Reached target Basic System.
Sep 13 20:19:18 raspberrypi systemd[1]: Starting rng-tools.service...
Sep 13 20:19:18 raspberrypi systemd[1]: Starting Avahi mDNS/DNS-SD Stack...
Sep 13 20:19:18 raspberrypi rng-tools[270]: Starting Hardware RNG entropy gatherer daemon: rngd.
Sep 13 20:19:18 raspberrypi rngd[276]: rngd 2-unofficial-mt.14 starting up...
Sep 13 20:19:18 raspberrypi systemd[1]: Starting LSB: Switch to ondemand cpu governor (unless shift key is pressed)...
Sep 13 20:19:18 raspberrypi systemd[1]: Condition check resulted in Copy user wpa_supplicant.conf being skipped.
Sep 13 20:19:18 raspberrypi systemd[1]: Starting dphys-swapfile - set up, mount/unmount, and delete a swap file...
Sep 13 20:19:18 raspberrypi rngd[276]: entropy feed to the kernel ready
Sep 13 20:19:18 raspberrypi avahi-daemon[272]: Found user 'avahi' (UID 108) and group 'avahi' (GID 113).
Sep 13 20:19:18 raspberrypi avahi-daemon[272]: Successfully dropped root privileges.
[... cut ...]
Sep 13 20:23:05 raspberrypi mopidy[603]: INFO     [IrisFrontend-11] mopidy_iris.core Starting Iris 3.45.0
Sep 13 20:23:05 raspberrypi mopidy[603]: INFO     [HttpFrontend-13] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Sep 13 20:23:05 raspberrypi mopidy[603]: ERROR    [MainThread] mopidy.commands Frontend (MpdFrontend) initialization error: MPD server startup failed: [Errno 98] Address already in use
Sep 13 20:23:05 raspberrypi mopidy[603]: INFO     [MainThread] mopidy.commands Starting GLib mainloop

j-gimbel, 2020-09-13:
Update: No audio is played except "LiveStream-Bayern2 " ...

Wiggum84, 2020-11-07:
Hi, 
Have you solved your problem?
I've got the exact same issue, but i'm using a Pi 3 Model B.

bubble1999, 2020-11-21:
I had the same issue. It helped to set up the system completely new as described in [https://forum-raspberrypi.de/forum/thread/48179-phoniebox-spotify-ich-bekomme-es-nicht-zum-laufen/](url):
I think the main point is installing the desktop packages.
Hope this will help you.

j-gimbel, 2020-11-21:
Thank you. I am currently developing my own jukebox

closed 2020-11-21 by j-gimbel