MiczFlor/RPi-Jukebox-RFID #1068: 🐛 Umzug der Phobiebox

Rei85An, 2020-09-20:
Nach längerem bauen funktioniert nun meine Phoniebox. Zwar mit 1-2 Bugs aber mit denen kann ich aktuelle leben. (Werde ich demnächst nochmals erläutern)

Was nun viel wichtiger für mich wäre ist das ich die Box nun in ein neues WLAN integrieren möchte. Am besten ohne viel Aufwand! Ich habe leider keinen Bildschirm oder Tastatur auf die ich zurückgreifen kann.

Gibt es dafür eine Lösung ohne das ich alles komplett neu aufsetzt muss?

Vielen Dank und Gruß von einem blutigem Anfänger der es immerhin bis hierher geschafft hat 👍

Freacly, 2020-10-04:
Über das Webinterface kannst du dies problemlos realisieren.

Rei85An, 2020-10-04:
Das Problem ist das ich nun im neuen WLAN sitze, der Pi aber noch ein anderes WLAN inne hat!

Ardios, 2020-10-04:
@Rei85An 

Da gibt es verschiedene Möglichkeiten...

Du kannst eine Tastatur, Maus und Monitor am Raspberry anschließen und dann das WLAN wechseln oder du greifst mit deinem PC/Book auf die Linux Partition der SD Karte zu und änderst dann manuell deine WLAN Daten (SSID, PW etc.) in der wpa_supplicant.conf

Es gibt mit Sicherheit noch andere Wege...dies sind nur meine persönlichen Erfahrungen ;)

Grüße

s-martin, 2020-10-05:
> und dann das WLAN wechseln oder du greifst mit deinem PC/Book auf die Linux Partition der SD Karte zu und änderst dann manuell deine WLAN Daten (SSID, PW etc.) in der wpa_supplicant.conf

So: https://raspberrypi.stackexchange.com/a/57023

closed 2020-12-15 by s-martin