MiczFlor/RPi-Jukebox-RFID #1069: LED Setup

UndIch, 2020-09-26:
Hi everyone, after almost 1 year with our phonibox I wanted to make some upgrades and came across one "issue" - maybe its just so trivial that it wasn't discussed yet.

Most of the jukebox pics I see ad GitHub use LED (in buttons in lights etc.) - but there is no explanation about those LED and the wiring (or did I just not come across it?).

So the issue is - I used 4 LED (2 yellow, 2 red) and wanted to add 2 more - they are all using the "status" script and are turned on as soon as box is fired up ... but now I came across several statements that even 2 normal LED (between 10-20 mA) could exceed the max amount of current healthy for the pi (max 30-50 mA?) - so even with a higher resistor you will reach that goal easily.

How did you solve that problem? I tried to "understand" transistor or MOSFET solutions but the days of my higher electrical/physical understandings are gone I guess.... 

So I just need 6 LEDs lighting up if the Box is on ;) 

Let me know if it is something dumb to ask... I googled all day (with a 3 year old in my ear all day :D - so not fully aware of what I read ;)

s-martin, 2020-09-27:
I agree there isn't really much documentation about LEDs, I think we should add some info.
Maybe #171 can provide some.

UndIch, 2020-09-27:
Yes - that’s the old script I used - but no information about the max. amount of LED etc - and most people trying to build the jukebox are electric noobs (like I was and still am - some physics in university but still noob...). Even with google it’s hard to find a step by step for using a transistor or a MOSFET for LEDs...

topas-rec, 2023-03-08:
There is an [article in the wiki](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Using-GPIO-hardware-status-led) about this topic. I just updated it but recognized that

- the documentation is not aligned with the [gpio_control](https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components/gpio_control) docs and
- the shutdown LED doesn't work (separate issue - I'll have to open another issue for that)

s-martin, 2023-05-02:
Wiki page was updated, so closing it for now.

closed 2023-05-02 by s-martin