MiczFlor/RPi-Jukebox-RFID #1070: RC522 reads, but ID does not show in webinterface

olli83, 2020-09-28:
## Bug
Hey! I am fairly new to the world of Raspberry and started this project for my little daughter.

I am using an RC522 RFID reader and initially had the issue described in  #1001, but got that fixed. I wrote a script to test the reader separately to phoniebox and it works fine, so it does not appear to be incorrectly wired or have any libraries missing.

In the web interface of Phoniebox the ID field stays empty when I bring a tag close to the reader.

Any hints?

### What I did

I installed the raspberry pi with above mentioned buster image and ran the installer script
Ran the steps to add an RC522 (libraries added and ran the steps described in the RC522 component readme)
Tested the RFID reader with a separate script
Fixed the issue described in #1001 
Double checked java is enabled in browser displaying the web interface

### What happened

No RFID ID showing up in web interface

The syslog usually notes that the channel for the RFID is already in use. Not sure if that is an issue. DId not do it this time when I created the log below.
There also appear to be some issues with MPD in the log and it states it failed to start. However, in other parts of the log it appears to be started.

At the moment there is a recurrent undervoltage issue, a new power supply is on order to fix this.

I am also surprised it takes 1m45s to start up. That appears to be rather long.

### I expected this to happen

RFID ID number in web interface

### Further information that might help

Sep 28 10:24:43 raspberrypi kernel: [    3.704169] NET: Registered protocol family 10
Sep 28 10:24:43 raspberrypi kernel: [    3.708648] Segment Routing with IPv6
Sep 28 10:24:43 raspberrypi kernel: [    4.787754] random: systemd: uninitialized urandom read (16 bytes read)
Sep 28 10:24:43 raspberrypi kernel: [    4.805539] random: systemd: uninitialized urandom read (16 bytes read)
Sep 28 10:24:43 raspberrypi kernel: [    4.816344] random: systemd: uninitialized urandom read (16 bytes read)
Sep 28 10:24:43 raspberrypi kernel: [    5.787132] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
Sep 28 10:24:43 raspberrypi kernel: [    6.497018] mc: Linux media interface: v0.10
Sep 28 10:24:43 raspberrypi kernel: [    6.499930] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
Sep 28 10:24:43 raspberrypi kernel: [    6.530001] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
Sep 28 10:24:43 raspberrypi kernel: [    6.530027] [vc_sm_connected_init]: start
Sep 28 10:24:43 raspberrypi kernel: [    6.537048] [vc_sm_connected_init]: installed successfully
Sep 28 10:24:43 raspberrypi kernel: [    6.552358] videodev: Linux video capture interface: v2.00
Sep 28 10:24:43 raspberrypi kernel: [    6.561734] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
Sep 28 10:24:43 raspberrypi kernel: [    6.570788] bcm2835_audio bcm2835_audio: card created with 8 channels
Sep 28 10:24:43 raspberrypi kernel: [    6.596594] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
Sep 28 10:24:43 raspberrypi kernel: [    6.622209] bcm2835_isp: module is from the staging directory, the quality is unknown, you have been warned.
Sep 28 10:24:43 raspberrypi kernel: [    6.639654] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
Sep 28 10:24:43 raspberrypi kernel: [    6.653947] bcm2835-isp bcm2835-isp: Device node output[0] registered as /dev/video13
Sep 28 10:24:43 raspberrypi kernel: [    6.654606] bcm2835-isp bcm2835-isp: Device node capture[0] registered as /dev/video14
Sep 28 10:24:43 raspberrypi kernel: [    6.655122] bcm2835-isp bcm2835-isp: Device node capture[1] registered as /dev/video15
Sep 28 10:24:43 raspberrypi kernel: [    6.655522] bcm2835-isp bcm2835-isp: Device node stats[2] registered as /dev/video16
Sep 28 10:24:43 raspberrypi kernel: [    6.655567] bcm2835-isp bcm2835-isp: Register output node 0 with media controller
Sep 28 10:24:43 raspberrypi kernel: [    6.655604] bcm2835-isp bcm2835-isp: Register capture node 1 with media controller
Sep 28 10:24:43 raspberrypi kernel: [    6.655624] bcm2835-isp bcm2835-isp: Register capture node 2 with media controller
Sep 28 10:24:43 raspberrypi kernel: [    6.655645] bcm2835-isp bcm2835-isp: Register capture node 3 with media controller
Sep 28 10:24:43 raspberrypi kernel: [    6.655922] bcm2835-isp bcm2835-isp: Loaded V4L2 bcm2835-isp
Sep 28 10:24:43 raspberrypi kernel: [    6.669292] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
Sep 28 10:24:43 raspberrypi kernel: [    6.717599] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
Sep 28 10:24:43 raspberrypi kernel: [    6.717651] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
Sep 28 10:24:43 raspberrypi kernel: [    6.725936] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
Sep 28 10:24:43 raspberrypi kernel: [    6.726100] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
Sep 28 10:24:43 raspberrypi kernel: [    6.752743] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
Sep 28 10:24:43 raspberrypi kernel: [    6.752922] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
Sep 28 10:24:43 raspberrypi kernel: [    7.142408] cfg80211: Loading compiled-in X.509 certificates for regulatory database
Sep 28 10:24:43 raspberrypi kernel: [    7.324241] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
Sep 28 10:24:43 raspberrypi kernel: [    7.463070] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[... 260 lines cut ...]
"Your branch is up to date with "origin/develop"

I am surprised it says develop, I understood the script installed the master.

### Installscript

Used the install script for Phoniebox.

## Hardware

### RaspberryPi version

Model 3 B

### RFID Reader

The RC522 RFID reader linked in the "shopping list" section

### Soundcard

none

### Other notable hardware

none

olli83, 2020-09-28:
Played around a bit further and found the reader service is not active. On retrospect I now also see that in the log above.

```
phoniebox-rfid-reader.service - Phoniebox RFID-Reader Service
   Loaded: loaded (/etc/systemd/system/phoniebox-rfid-reader.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2020-09-28 13:35:47 BST; 17min ago
  Process: 1359 ExecStart=/home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py (code=exited, status=1/FAILURE)
 Main PID: 1359 (code=exited, status=1/FAILURE)

Sep 28 13:35:47 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Service RestartSec=100ms expired, scheduling restart.
Sep 28 13:35:47 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Scheduled restart job, restart counter is at 5.
Sep 28 13:35:47 raspberrypi systemd[1]: Stopped Phoniebox RFID-Reader Service.
Sep 28 13:35:47 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Start request repeated too quickly.
Sep 28 13:35:47 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Failed with result 'exit-code'.
Sep 28 13:35:47 raspberrypi systemd[1]: Failed to start Phoniebox RFID-Reader Service.
```

stundenblume, 2020-10-05:
@olli83
Bei mir war folgendes Vorgehen notwendig (Stand 27.09.20):
1. SPI aktivieren (`sudo raspi-config`)

2. `sudo python3 -m pip install --upgrade --force-reinstall -r /home/pi/RPi-Jukebox-RFID/components/rfid-reader/PN53/requirements.txt`
    Quelle: https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/11#issuecomment-646578154

3. `sudo sh RPi-Jukebox-RFID/components/rfid-reader/RC522/setup_rc522.sh`
    Quelle: https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/11#issuecomment-646437875

Danach hat der RC522 gearbeitet.

s-martin, 2020-10-05:
As @stundenblume said, please try

`sudo sh RPi-Jukebox-RFID/components/rfid-reader/RC522/setup_rc522.sh`

This should be sufficient, the script is supposed to do all the manual steps described.

Erechtheus, 2020-10-11:
Hey folks! I recently (today) stumbled across the same issue. For me there was a missing dependency (i.e., py532lib) in the Reader.py.experimental script. So for me the script mentioned by @s-martin did not work!
This fixed my issue:
`pip3 install py532lib`

olli83, 2020-10-11:
I have commented out the py532lib references in the script, so this should not have been the issue. I started from scratch again to set up the RC522 standalone with some online tutorials. It worked and afterwards it also worked with the Phoniebox. I must have made a mistake somewhere along the line though I am not sure were.

Thanks for the support and also for removing the bug label. When starting a new thread it is not possible to assign any other label to it.

closed 2020-10-11 by olli83