MiczFlor/RPi-Jukebox-RFID #1071: GPIO Buttons do not work

schmonzolo, 2020-09-30:
Hey guys,

first of all: Thank you so much for this wonderful project! I had so much fun and I bet my daughter will have as well 👍 

I am facing an issue with the GPIO buttons. Here is what I did so far.
- Installed software with install script (worked and plays stuff via headphone output)
- wired an RC522 (works)
- follwed the instructions for wiring GPIOs and copy the relevant script. Also added relevant services to have it available after reboot

Now I wanted to connect some arcade buttons to control the box. Unfortuantely I receive on all buttons error messages:

Volume Down -> does not work
```
gpio-buttons.py:150: RuntimeWarning: A physical pull up resistor is fitted on this channel!
  GPIO.setup(btn_shut, GPIO.IN, pull_up_down=GPIO.PUD_UP)
gpio-buttons.py:158: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(led_power, GPIO.OUT)
/bin/sh: 1: .scripts/playout_controls.sh: not found
Traceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py", line 122, in volD_handler
    volD_action(channel)
  File "/home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py", line 60, in volD_action
    check_call(".scripts/playout_controls.sh -c=volumedown", shell=True)
  File "/usr/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '.scripts/playout_controls.sh -c=volumedown' returned non-zero exit status 127.
```

VolumeUp -> does not work
```
gpio-buttons.py:150: RuntimeWarning: A physical pull up resistor is fitted on this channel!
  GPIO.setup(btn_shut, GPIO.IN, pull_up_down=GPIO.PUD_UP)
gpio-buttons.py:158: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(led_power, GPIO.OUT)
/bin/sh: 1: ./scripts/playout_controls.sh: not found
Traceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py", line 113, in volU_handler
    volU_action(channel)
  File "/home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py", line 64, in volU_action
    check_call("./scripts/playout_controls.sh -c=volumeup", shell=True)
  File "/usr/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command './scripts/playout_controls.sh -c=volumeup' returned non-zero exit status 127.
```
Pause / Play -> DOES work eventhough there is an error
```
gpio-buttons.py:150: RuntimeWarning: A physical pull up resistor is fitted on this channel!
  GPIO.setup(btn_shut, GPIO.IN, pull_up_down=GPIO.PUD_UP)
gpio-buttons.py:158: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(led_power, GPIO.OUT)
/bin/sh: 1: ./scripts/playout_controls.sh: not found
Traceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py", line 76, in halt_action
    check_call("./scripts/playout_controls.sh -c=playerpause", shell=True)
  File "/usr/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command './scripts/playout_controls.sh -c=playerpause' returned non-zero exit status 127.
```
Next -> does not work
``` 
gpio-buttons.py:150: RuntimeWarning: A physical pull up resistor is fitted on this channel!
  GPIO.setup(btn_shut, GPIO.IN, pull_up_down=GPIO.PUD_UP)
gpio-buttons.py:158: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(led_power, GPIO.OUT)
/bin/sh: 1: ./scripts/playout_controls.sh: not found
Traceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py", line 68, in next_action
    check_call("./scripts/playout_controls.sh -c=playernext", shell=True)
  File "/usr/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command './scripts/playout_controls.sh -c=playernext' returned non-zero exit status 127.
```
Previous
```
gpio-buttons.py:150: RuntimeWarning: A physical pull up resistor is fitted on this channel!
  GPIO.setup(btn_shut, GPIO.IN, pull_up_down=GPIO.PUD_UP)
gpio-buttons.py:158: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(led_power, GPIO.OUT)
/bin/sh: 1: ./scripts/playout_controls.sh: not found
Traceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py", line 72, in prev_action
    check_call("./scripts/playout_controls.sh -c=playerprev", shell=True)
  File "/usr/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command './scripts/playout_controls.sh -c=playerprev' returned non-zero exit status 127.
```

The shutdown works, butit also throws an error. Unfortunately I am not fast enough to copy that from the console :) However, it basically looks the same as all the errors.

First, it looks like the playout_controls.sh cannot be found. I sued that command to get the sample GPIO script:

```
sudo cp /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/gpio-buttons.py.sample /home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py
```
Which results in the follwoing content in /home/pi/RPi-Jukebox-RFID/scripts ->

```
activate_amplifier.py             __pycache__
daemon_rfid_reader.py             python-phoniebox
deviceName.txt                    Reader.py
gpio-buttons.py                   Reader.py.experimental
helperscripts                     Reader.py.kkmoonRFIDreader
idle-watchdog-countdown.sh        Reader.py.original
idle-watchdog.sh                  RegisterDevice.py
inc.readArgsFromCommandLine.sh    resume_play.sh
inc.settingsFolderSpecific.sh     rfid_trigger_play.sh
inc.writeFolderConfig.sh          shuffle_play.sh
inc.writeGlobalConfig.sh          single_play.sh
__init__.py                       startup-scripts.sh
installscripts                    test
playlist_recursive_by_folder.php  userscripts
playout_controls.sh

```
The script seems to use the wrong file path, as it always says:
```
def volD_action(channel):
    check_call("./scripts/playout_controls.sh -c=mute", shell=True)
```
I guess the "./scripts" is to much here, sicne both scripts are in the same dir. However, when I change that to:
```
def volD_action(channel):
    check_call("./playout_controls.sh -c=mute", shell=True)
```
The buttons still do not work, but I get another error:
```
gpio-buttons.py:150: RuntimeWarning: A physical pull up resistor is fitted on this channel!
  GPIO.setup(btn_shut, GPIO.IN, pull_up_down=GPIO.PUD_UP)
gpio-buttons.py:158: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(led_power, GPIO.OUT)
expr: syntax error: unexpected argument ‘(’
./playout_controls.sh: line 276: [: -ge: unary operator expected
OK MPD 0.21.4
OK
```

If I try the different path, then the following buttons work without any issue:
- Play/Pause/Halt
- Shutdown
- Next
- Previous

The volume up -> does not work
```
gpio-buttons.py:150: RuntimeWarning: A physical pull up resistor is fitted on this channel!
  GPIO.setup(btn_shut, GPIO.IN, pull_up_down=GPIO.PUD_UP)
gpio-buttons.py:158: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
expr: syntax error: unexpected argument ‘3’
./playout_controls.sh: line 238: [: -le: unary operator expected
OK MPD 0.21.4
OK
```

The volume down -> does not work
```
gpio-buttons.py:150: RuntimeWarning: A physical pull up resistor is fitted on this channel!
  GPIO.setup(btn_shut, GPIO.IN, pull_up_down=GPIO.PUD_UP)
gpio-buttons.py:158: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(led_power, GPIO.OUT)
expr: syntax error: unexpected argument ‘(’
./playout_controls.sh: line 276: [: -ge: unary operator expected
OK MPD 0.21.4
OK
```

I am super stuck with the latter one, as my bash script knowledge is unfortunately not as profound as it should be :(

Also I am not sure what is going on with the runtime warning regarding the GPIOs, as that always appears.

It would be super great if somebody could help me out! Thanks!!!

s-martin, 2020-10-05:
The path `./scripts/playout_controls.sh` should be fine as the service for gpio-buttons.py runs in the root dir of Phoniebox.

Did you check that you edited the used pin numbers in gpio-buttons.py are correct?

s-martin, 2020-10-05:
What's the output of `tail -n 500 /var/log/syslog`?

s-martin, 2020-10-17:
Please see #1096, the paths for `gpio_settings.ini` were mixed up. This is now fixed in `develop` branch.

Please give some info, if these changes fix this issue.

schmonzolo, 2020-10-19:
Hey s-martin,

Thanks for your response! I am so sry for the late response from my side! My spam folder has eaten the git messages :(

I will have some time on Wednesday to implement #1096 and let you know. 

Regards

schmonzolo, 2020-10-21:
Hey s-martin, 

so I wired now four buttons:
- Start/Stop
- volUp
- volDown
- shutdown

The pins match the definition in gpio-buttons.py. I used the layout provided in the documentation and there was no need for adaption in gpio-buttons.py.

I started the phoniebox and pressed the buttons that I have wired now. The out of "tail -n 500 /var/log/syslog" is:

```
Oct 21 19:52:35 raspberrypi gpio-buttons.py[497]: volume: n/a   repeat: off   random: off   single: off   consume: off
Oct 21 19:52:36 raspberrypi gpio-buttons.py[497]: volume: n/a   repeat: off   random: off   single: off   consume: off
Oct 21 19:53:59 raspberrypi gpio-buttons.py[3850]: expr: syntax error: unexpected argument '3'
Oct 21 19:53:59 raspberrypi gpio-buttons.py[3850]: ./scripts/playout_controls.sh: line 238: [: -le: unary operator expected
Oct 21 19:53:59 raspberrypi gpio-buttons.py[3850]: OK MPD 0.21.4
Oct 21 19:53:59 raspberrypi gpio-buttons.py[3850]: OK
Oct 21 19:53:59 raspberrypi gpio-buttons.py[3850]: expr: syntax error: unexpected argument '3'
Oct 21 19:53:59 raspberrypi gpio-buttons.py[3850]: ./scripts/playout_controls.sh: line 238: [: -le: unary operator expected
Oct 21 19:53:59 raspberrypi gpio-buttons.py[3850]: OK MPD 0.21.4
Oct 21 19:53:59 raspberrypi gpio-buttons.py[3850]: OK
Oct 21 19:54:17 raspberrypi gpio-buttons.py[3850]: volume: n/a   repeat: off   random: off   single: off   consume: off
Oct 21 19:54:19 raspberrypi gpio-buttons.py[3850]: expr: syntax error: unexpected argument '3'
Oct 21 19:54:19 raspberrypi gpio-buttons.py[3850]: ./scripts/playout_controls.sh: line 238: [: -le: unary operator expected
Oct 21 19:54:19 raspberrypi gpio-buttons.py[3850]: OK MPD 0.21.4
Oct 21 19:54:19 raspberrypi gpio-buttons.py[3850]: OK
Oct 21 19:54:20 raspberrypi gpio-buttons.py[3850]: expr: syntax error: unexpected argument '('
Oct 21 19:54:20 raspberrypi gpio-buttons.py[3850]: ./scripts/playout_controls.sh: line 276: [: -ge: unary operator expected
Oct 21 19:54:20 raspberrypi gpio-buttons.py[3850]: OK MPD 0.21.4
Oct 21 19:54:20 raspberrypi gpio-buttons.py[3850]: OK

```

Of course I cannot show you what happened when I press the shutdown, since the shutdown works ;)

I also implemented the change from #1096 but that did not do the trick for me :(

I have a **workaround that works for me** so far in order to control volume via buttons. Unfortunately I cannot use the app for volume control with that solution. In the gpio-buttons.py I do the following changes:

```
def volD_action(channel):
    check_call("amixer -q -M sset Headphone 3%-", shell=True)
    #check_call("./scripts/playout_controls.sh -c=volumedown", shell=True) -> comment this line

def volU_action(channel):
    check_call("amixer -q -M sset Headphone 3%+", shell=True)
    #check_call("./scripts/playout_controls.sh -c=volumeup", shell=True) -> comment this line
```
Obviously one needs to know how to feed the amixer api for this. I found that I am not the first one asking how to do this :) : https://unix.stackexchange.com/questions/32206/set-volume-from-terminal

s-martin, 2020-12-18:
Since 2.1 we use GPIO control. Could you please check, if this fixes your issue?

s-martin, 2022-12-20:
Closing as there was no activity for a very long time.

Please check the current release 2.4 or our new implementation 3.2, if this issue still persists.

Feel free to open a new issue, if necessary.

closed 2022-12-20 by s-martin