MiczFlor/RPi-Jukebox-RFID #1072: Getting my HifiBerry Mini Amp running again

georgbachmann, 2020-10-02:
I have my phoniebox for quite a while now, but recently the card reader stopped working. So today I finally found some time and tried out various thing. The cards would just not read :(
So I simply went ahead and reinstalled the Phoniebox Software using the command line script. It was asking me something about audio right in the beginning which I skipped, cause audio was working before (I did not completely wipe the whole raspberry)...
Then I first went ahead and made sure the rfid thinggy was working again (which it was... jippie) but then I found out that the sound is not working again.
I remember that this was quite a pain when I initially did set it up... I even had some backups where I could see my config. Nothing worked

So now after about 2 hours of not really knowing anything about audio and linux and playing around with various suggestions I was reading on the internet, I have to give up... And would be super happy if I can find somebody smarter than me to help me :)

So... my current config is this:

/etc/asound.conf
```
pcm.hifiberry {
    type softvol
    slave.pcm "plughw:0"
    control.name "Master"
    control.card 0
}
pcm.!default {
    type       plug
    slave.pcm  "hifiberry"
}
```

/etc/mpd.conf
```
audio_output {
        type            "alsa"
        name            "My ALSA Device"
#       device          "hw:0,0"        # optional
#       mixer_type      "software"      # optional
#       mixer_device    "default"       # optional
#       mixer_control   "Master"        # optional
#       mixer_index     "0"             # optional
}

audio_output {
        enabled         "yes"
        type            "alsa"
        name            "HiFiBerry DAC"
        device          "hifiberry"
        auto_resample   "no"
        auto_channels   "no"
        auto_format     "no"
        dop             "no"
}
```

/boot/config.txt
```
device_tree_param=spi=on
dtoverlay=spi-bcm2708

dtoverlay=hifiberry-dac
dtoverlay=dwc2
```

`speaker-test` does at least pretend do be doing something... but I cant hear anything

`aplay -l` gives me
```

**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_dac], device 0: HifiBerry DAC HiFi pcm5102a-hifi-0 [HifiBerry DAC HiFi pcm5102a-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

I don't know if there is anything else relevant... 
I'd be super super happy for any help as I have no clue what else I need to try :(

lj2000, 2020-10-02:
On my side I have two phonieboxes running which both have a Hifiberry DAC.

I don't know if the problem still exists after the rework sessions of the phoniebox software in the last months.
But in the past you always had to change the default gpio configuration to get the Hifiberry DAC running with the Phoniebox. 

Using the standard gpio configuration leads to a reconfiguration of the I2C pin(s) to gpio functionality. But the Hifiberry DAC needs the  I2C hardware function. So in my opinion you must disable the gpio service if you don't use any buttons or you have to reconfigure the gpio settings script to your hardware situation

lj2000, 2020-10-02:
See also:
[https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/544](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/544)

lj2000, 2020-10-02:
It's for another Audio HAT, but the problem was the same:
[https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/549](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/549)

georgbachmann, 2020-10-03:
Ahhhhhh... super cool hint! Thank you very much! I hope I'll have time tomorrow evening to try it out, but will keep you posted! Thanks a lot!!!

georgbachmann, 2020-10-05:
@lj2000 yeah yeah yeah... that did the trick!!! awesome! wouldn't have thought about that :) cool!!! thanks a lot!!!

closed 2020-10-05 by georgbachmann