MiczFlor/RPi-Jukebox-RFID #1073: No volume control with Hifiberry Miniamp

jackwidu, 2020-10-03:
Hello Everybody, 
I know, there is a closed issue with the same topic, but it didn't work for me and I hope you guys can help.
Starting with a 100% working RPi-Jukebox (Raspi 2B, MFRC522, even a OLED, clean installation 3 months ago) I decided to improve sound quality with a Hifiberry Miniamp.
I changed the gpio buttons (prev/halt/next/volu/vold -> GPIO 17/27/22/23/4) in the RPi-Jukebox-RFID/scripts/gpio-buttons.py.
The MFRC522 uses GPIO 8/9/10/11/24/25 so there was no problem.
For installation of the hifiberry I followed https://forum-raspberrypi.de/forum/thread/36628-hifiberry-miniamp-aktivieren/ which made me change /boot/config.txt to `# Enable audio (loads snd_bcm2835)
####dtparam=audio=on
dtoverlay=hifiberry-dac`
and /etc/asound.conf (which had to be created)
`#pcm.!default {
#type hw card 0
#}
#ctl.!default {
#type hw card 0
#}

pcm.hifiberryMiniAmp {
    type softvol
    slave.pcm "plughw:0"
    control.name "Master"
    control.card 0
}
pcm.!default {
    type       plug
    slave.pcm  "hifiberryMiniAmp"
}`

Following https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/HiFiBerry-Soundcard-Details I then changed /etc/mpd.conf
I made sure the following was uncommented `audio_output {
        type               "alsa"
        name            "My ALSA Device"
}`
and added two lines below: `audio_output {
        enabled         "yes"
        type            "alsa"
        name            "HiFiBerry DAC+ Lite"
        device          "hifiberry"
        auto_resample   "no"
        auto_channels   "no"
        auto_format     "no"
        dop             "no"
}`

And now it works  (reads card, plays sound, button for halt and next), but I cannot change volume. Not with the buttons and not in the webinterface. I can change the volume with alsamixer. The selected card's name is "snd_rpi_hifiberry_dac" (I think i changed from default at certain point, but don't know when). 

What I tried:
- added `mixer_type      "software"`in /etc/mpd.conf, as well as in the longer paragraph with HiFiBerry DAC+ Lite and in the one above.
- added `mixer_type "digital"`
- changed the gpio in gpio-buttons.py for volU and volD

https://splittscheid.de/faqs-zu-meiner-phoniebox/#MiniAmp suggests to change the Audio_iFace_Name, but I don't understand what I should do. I am not sure, what I named the sound device in the inital installation, I think it was Headphone and not Master. Maybe this is the problem? Should I change it to "Master"? If, how? `amixer scontrols` gives me `Simple mixer control 'Master',0`
I also tried to change /etc/mpd.conf from `name            "My ALSA Device"` to `name            "Master"`

For my understanding this is not a hardware problem, as I can change volume by alsamixer. How can I make the webface an the GPIO work?

Thank you very much
  Jakob

PS. Is it true, that I cannot useboth hifiberry and a oled-display as they both need the GPIO 2and3for I2C?

lj2000, 2020-10-03:
As far as I can remember I had a similar problem.
In my installation notes I found the following: 
1. Ensure that the gpio service is disabled via systemctl (switching it off in the web interface is not enough) or configure the gpio script in way that the I2S pins of the RaspberryPi/HifiberryMiniamp are not used for gpio function. 
2. Modify `/boot/config.txt`: 
```
#dtparam=audio=on
dtoverlay=hifiberry-dac
```
3. Create/edit `/etc/asound.conf ` in the following way:
```
	#pcm.!default {
	# type hw card 0
	#}
	#ctl.!default {
	# type hw card 0
	#}

	pcm.hifiberryMiniAmp {
	    type softvol
	    slave.pcm "plughw:0"
	    control.name "Master"
	    control.card 0
	}
	pcm.!default {
	    type       plug
	    slave.pcm  "hifiberryMiniAmp"
	}
```
3. Reboot: `sudo reboot`
4. Perform the following command:
**`speaker-test -D hifiberryMiniAmp -c 2`**

**After these steps you should see the MASTER slider in the alsa-mixer.**

For my 2 boxes which also have the Miniamp in service I didn't have to do anything else than the steps described above. I didn't have to change any other setting (i.e. mpd.conf)

If the problem is still present and if you are not shure if the I2S bus isn't modified by the gpio script/service you can have a look in the system log using the `dmesg` command. If anything is wrong you should see an I2S error message.

jackwidu, 2020-10-04:
Thank you lj2000, but for my understanding this does not Adress my problem as the speaker DOES work and I already see in alsa the Master slider (and it works)

lj2000, 2020-10-04:
Ok. It could be that this is not the solution for your problem.

BUT: 
As I wrote above: I have two boxes with a Miniamp which work perfectly.
And I could remember that I had the same problem with both boxes as you described. The Hifiberry itself worked on the command line and I couldn't control the volume using the web client.

So I created a new SD card with raspbian buster lite.
After that I performed the steps described above.
And for the final step I ran the phoniebox installer.

After that the audio control and output of the Phonieboxes worked WITHOUT any further modification in mpd.conf etc.

ueb, 2020-10-04:
Try to add `mixer_control	"Master"` to the audio_output "My ALSA Device" in mpd.conf

jackwidu, 2020-10-05:
YES!!!! Thank you so much! Now it works :)
Now only the little additional question:Is it true, that I can only use one I2C device at a time?That means no OLED while Hifiberry?
Thanks again.

s-martin, 2020-10-05:
> YES!!!! Thank you so much! Now it works :)
> 
> Now only the little additional question:Is it true, that I can only use one I2C device at a time?That means no OLED while Hifiberry?
> 
> Thanks again.

If you know German, @splitti has an extensive howto, also OLED with Hifiberry:
https://splittscheid.de/selfmade-phoniebox/#5_3

closed 2020-12-15 by s-martin