MiczFlor/RPi-Jukebox-RFID #1074: 🐛 RC522 erscheint nicht in Phoniebox

clemsonfire, 2020-10-04:
Hallo!

Erstmals herzliche Gratulation zu diesem coolen Projekt!
Ich bin Umsteiger von Tonuino - da fehlte mir die Online-Option, weil die Kinder gerne Webradios hören.

Ich benutze einen Raspi Zero (raspbian lite, buster) und habe einen RC522 angeschlossen.
Im Webinterface lässt sich RFID Reader nicht dauerhaft aktivieren. Vor den Reader gehaltene Cards tauchen nicht im WebIf mit ihren CardIDs auf. Wenn ich reader im WebIf aktiviere, dann ist er nach reload wieder deaktivivert.

Nach dem Lesen - nicht für phoniebox zutreffender Howtos - habe ich nun den RC522 mit 8Pins connected, zuvor wegen nicht idealer Anleitung nur mit 7.

Mit einem Script laut einer Anleitung sehe ich auf Konsole, dass der Reader arbeitet; Karten IDs werden mir mit diesem Mini-Script angezeigt; wie gesagt, nur auf der Konsole, nicht im WebIf.

Reader.py.experimental nutze ich als Reader.py, doch da erhalte ich diese Fehlermeldung:

pi@phoniebox:~ $ python /home/pi/pi-rc522/examples/Read.py
Traceback (most recent call last):
  File "/home/pi/pi-rc522/examples/Read.py", line 7, in <module>
    from pirc522 import RFID
ModuleNotFoundError: No module named 'pirc522'

und bei 
pi@phoniebox:~ $ sudo pip install spidev
sudo: pip: Befehl nicht gefunden
pi@phoniebox:~ $ sudo  pip install pi-rc522
sudo: pip: Befehl nicht gefunden´

Bin leider mangels Linux-Kentnissen auf drag'n'drop von Befehlen angewiesen.

Bitte könntet Ihr mir sagen, wo ich ansetzen muß, dass reader von der Phoniebox erkannt und genutzt wird?

Vielen herzlichen Dank im voraus!

Lg. clemsonfire

## Hardware

**RaspberryPi version**
Zero
**RFID Reader**
RC522
**Soundcard**
Audio Pirate 3W

Ardios, 2020-10-04:
@clemsonfire 
Nabend,

schau dir mal folgenden Link an, dies sollte deine Probleme eigentlich beheben (8 Pin ist ja schon angeschlossen)

[https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components/rfid-reader/RC522](https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components/rfid-reader/RC522)

`sudo python3 -m pip install -q -r /home/pi/RPi-Jukebox-RFID/components/rfid-reader/RC522/requirements.txt`

Damit sollte alles fehlende installiert werden

LG

clemsonfire, 2020-10-05:
Hallo!

Vielen Dank für deine Antwort mit der super Seite und dem Verweis auf die Requirements! Ich war immer im Scripts Ordner... :-(

Auf dieser WIKI Seite habe ich das Fehlen des 8. Pins entdeckt! ..ich verbringe mittlerweile Tage auf der Suche nach meinem Fehler ;-) , lerne aber dabei viel dazu - sehr angenehmer Nebeneffekt! 

Ich habe zuerst mal manuell das Fehlende nachinstalliert und dann neu die Reader eingelesen.
Ergbnis:
`python3 RegisterDevice.py
Traceback (most recent call last):
  File "RegisterDevice.py", line 4, in <module>
    from Reader import get_devices
  File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 17, in <module>
    from py532lib.i2c import *
ModuleNotFoundError: No module named 'py532lib'
`
Also habe ich dann auch noch die setup_rc522.sh aufgerufen. 
Ergbnis:
`sudo ./setup_rc522.sh 
Please make sure that the RC522 reader is wired up correctly to the GPIO ports before continuing...
Continue (y/n)? y
Installing Python requirements for RC522...
Configure RFID reader in Phoniebox...
Restarting phoniebox-rfid-reader service...
Done.
`
Dachte, dass das nun passt. Im WebIf wird aber nach wie vor der Reader als deaktiviert angeführt. Hat wohl mit der fehlenden py532lib zu tun!?

Zu der Auswahl, welchen Reader ich benutzen möchte, kam es leider nicht.

Soll ich 

   `Did you run components/rfid-reader/RC522/setup_rc522.sh?
    Then you don't need to run RegisterDevice.py, which currently contains a bug.`
machen?

Würde mich freuen, von Dir/ Euch weitere Hinweise zu bekommen!

Besten Dank im Voraus!
C.

Ardios, 2020-10-05:
@clemsonfire 

Sorry für die späte Antwort...bitte installiere das hier noch nach:
`sudo pip install py532lib`

Danach dann bitte noch:
`python /home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py`

Und dort dann die "0" für MFRC522 wählen, dann nur noch im Webinterface den Reader aktivieren und dann sollte es eigentlich laufen

Viele Grüße

stundenblume, 2020-10-05:
@clemsonfire
bei mir war folgendes Vorgehen notwendig:
1. SPI aktivieren (`sudo raspi-config`)

2. `sudo python3 -m pip install --upgrade --force-reinstall -r /home/pi/RPi-Jukebox-RFID/components/rfid-reader/PN53/requirements.txt`
    Quelle: https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/11#issuecomment-646578154

3. `sudo sh RPi-Jukebox-RFID/components/rfid-reader/RC522/setup_rc522.sh`
    Quelle: https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/11#issuecomment-646437875

Danach hat der RC522 gearbeitet.

Ardios, 2020-10-05:
@stundenblume 

Bei Punkt 2. fehlt glaube ich eine 2... 

`/home/pi/RPi-Jukebox-RFID/components/rfid-reader/PN532/requirements.txt` anstatt `/home/pi/RPi-Jukebox-RFID/components/rfid-reader/PN53/requirements.txt`

Aber sonst funktioniert dein Weg genauso :)

Viele Grüße

stundenblume, 2020-10-05:
@Ardios
Stimmt. Ist korrigiert. Danke! :)

clemsonfire, 2020-10-05:
Hallo!
Na Ihr seids der Wahnsinn! Danke für die superschnellen Antworten! 

So, habe Eure Hinweise umgesetzt:
`./setup_rc522.sh
Please make sure that the RC522 reader is wired up correctly to the GPIO ports before continuing...
Continue (y/n)? y
Installing Python requirements for RC522...
Configure RFID reader in Phoniebox...
Restarting phoniebox-rfid-reader service...
Done.
`
Das "Done" klingt ja gut, ABER: Ich werde leider nicht gefragt, welchen Reader ich verwenden möchte. Also nichts, wo ich "0" eingeben könnte.

Der Reader hängt natürlich nach wie vor am Raspi.
Bitte höflich wiederum um Eure Ideen!

Danke im voraus! C.

Ardios, 2020-10-05:
Hey @clemsonfire 

Hast du die beiden Befehle von mir ausprobiert? 

`sudo pip install py532lib` und danach dann `python /home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py`  ?

Wenn ja, was steht denn da wenn du `python /home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py` eingibst?

Viele Grüße

clemsonfire, 2020-10-05:
HURRA! Es klappt! Soeben ist im WebIf - wo der Reader bereits als aktiviert vermerkt war - die gescannte Karten ID aufgetaucht!

Euch vielen, vielen herzlichen Dank!

Da kommt Freude auf! Jetzt gehts an die weiteren Vorbereitungen und dann ans Kiste bauen :-)

Lg. C.

Ardios, 2020-10-05:
Perfekt, dann kannst du meinen letzten Post natürlich ignorieren xD

Freut mich das es nun klappt 👍 

Viele Grüße

s-martin, 2020-10-05:
> Hallo!
> Na Ihr seids der Wahnsinn! Danke für die superschnellen Antworten!
>
> So, habe Eure Hinweise umgesetzt:
>
> ```
> ./setup_rc522.sh
> Please make sure that the RC522 reader is wired up correctly to [...]

Ja, das ist so gewollt. Das Script installiert vja schon diesen Reader, da brauchst du keinen mehr auswählen.

s-martin, 2020-10-05:
Wenn's funktioniert, bitte Issue schließen ;)

clemsonfire, 2020-10-05:
Hi!
Alles klar, @s-martin , das macht Sinn! Ich habe nur immer auf diese Abfrage gewartet.

Vielen Dank allen Lesern und Helfern! Macht bitte weiter so super Arbeit!
Thread wird geschlossen!
Lg. clemsonfire

closed 2020-10-05 by clemsonfire