MiczFlor/RPi-Jukebox-RFID #1075: Spotify playing via IRIS but not from Phoniebox webinterface or RFID Tags

stundenblume, 2020-10-05:
Phoniebox is running well with uploaded .mp3 files, and web streams.
Spotify is working great with Iris Webinterface.

If I give the RFID tag for a Spotify playlist or use the Phoniebox web interface the current playing song/stream stops and nothing more is happening.

Fresh Buster install, RC522, one line installation (27.09.20).

s-martin, 2020-10-05:
That's exactly the place to ask questions. :)

What's the output of `journalctl -u mopidy`?

stundenblume, 2020-10-07:
@s-martin
Sorry for my late reply! I didn't notice your reply.
Output of `journalctl -u mopidy`

> Oct 06 18:34:44 raspberrypi mopidy[678]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Authorization failed, not attempting Spotify API request. Please get new credentials from https://www.mo$
> Oct 06 18:34:44 raspberrypi mopidy[678]: ERROR    [SpotifyBackend-6] mopidy_spotify.web OAuth token refresh failed: invalid_client Client not known.
> Oct 06 18:34:44 raspberrypi mopidy[678]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Failed to load Spotify user profile

I than checked my credentials again and I had the wrong ones inserted.

So: Thank you for giving me the right hint!

closed 2020-10-07 by stundenblume