MiczFlor/RPi-Jukebox-RFID #1078: 🐛 | GPIO Buttons do not work (python script error)

toxidle, 2020-10-07:
## Bug
GPIO Buttons do not work (python script error)

### What I did
hardware is a raspberry pi 3 with a hifiberry amp and a rc522, using 223B touch buttons for gpio input.
everything works except the gpio-buttons

### What happened

installed the gpio-buttons 
i get the following erros when i start the daemon

```
pi@raspberrypi:~/RPi-Jukebox-RFID/components/gpio_control $ ./gpio_control.py 
INFO:root:['VolumeControl', 'PrevNextControl', 'PlayPause', 'Shutdown', 'Volume0', 'VolumeUp', 'VolumeDown', 'NextSong', 'PrevSong', 'Halt']
INFO:root:adding GPIO-Device, VolumeControl
VolumeControl
INFO:gpio_control:VolumeControl as TwoButtonControl
INFO:GPIODevices.two_button_control:adding new action
INFO:GPIODevices.simple_button:{}: set when_pressed
INFO:GPIODevices.simple_button:add new action
INFO:GPIODevices.simple_button:{}: set when_pressed
INFO:GPIODevices.simple_button:add new action
INFO:root:adding GPIO-Device, PrevNextControl
PrevNextControl
INFO:root:adding TwoButtonControl
INFO:GPIODevices.two_button_control:adding new action
INFO:GPIODevices.simple_button:{}: set when_pressed
INFO:GPIODevices.simple_button:add new action
INFO:GPIODevices.simple_button:{}: set when_pressed
INFO:GPIODevices.simple_button:add new action
INFO:root:adding GPIO-Device, PlayPause
PlayPause
Traceback (most recent call last):
  File "./gpio_control.py", line 105, in <module>
    devices = get_all_devices(config)
  File "./gpio_control.py", line 83, in get_all_devices
    device = generate_device(config[section], section)
  File "./gpio_control.py", line 54, in generate_device
    pull_up_down=config.get('pull_up_down', fallback='GPIO.PUD_UP'))
  File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/simple_button.py", line 66, in __init__
    self.pull_up_down = parse_pull_up_down(pull_up_down)
  File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/simple_button.py", line 34, in parse_pull_up_down
    raise KeyError('Unknown Pull Up/Down type {pull_up_down}'.format(pull_up_down=pull_up_down))
KeyError: 'Unknown Pull Up/Down type GPIO.PUD_UP'
Exception ignored in: <function SimpleButton.__del__ at 0x765ca3d8>
Traceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/simple_button.py", line 110, in __del__
    GPIO.remove_event_detect(self.pin)
AttributeError: 'SimpleButton' object has no attribute 'pin'
```

### Further information that might help

tested the functionality with simple_button.py and it worked

pi@raspberrypi:~/RPi-Jukebox-RFID/components/gpio_control/GPIODevices $ python simple_button.py 
please enter pin no to test
27
FunctionCall with (27,)
FunctionCall with (27,)
FunctionCall with (27,)
FunctionCall with (27,)
FunctionCall with (27,)

## Software

### Base image and version

pi@raspberrypi:~/RPi-Jukebox-RFID/components/gpio_control/GPIODevices $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Branch / Release

pi@raspberrypi:~/RPi-Jukebox-RFID/components/gpio_control/GPIODevices $ cd /home/pi/RPi-Jukebox-RFID/ && git status | head -2
Refresh index: 100% (390/390), done.
On branch develop
Your branch is up to date with 'origin/develop'.

### Installscript

i.e. `scripts/installscripts/buster-install-default.sh`
without spotify support

## Hardware

### RaspberryPi version

Raspberry Pi 3 Model B Rev 1.2

### RFID Reader

RC522

### Soundcard

Hifi Berry Amp

### Other notable hardware

default settings imported to /home/pi/.config/phoniebox/gpio-settings.ini

s-martin, 2020-10-12:
This should been fixed, see https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/5c93f67e9799900709bfbce9a756ac4f5c68d7e1#diff-7de2980ae0dfe3a67ec69f2e922125be

We just released 2.1, which should fix this issue.

drocx, 2020-10-13:
Hi. 

I have upgraded to version 2.1. 

Now the gpio buttons are not working anymore.

Before the upgrade the buttons worked fine. 

GPIO Buttons are enabled on the settings Page.

s-martin, 2020-10-13:
What's the output of  `tail -n 500 /var/log/syslog`?

Berdsen, 2020-10-14:
Hi there,

after updating from 2.0 to 2.1 the buttons didn't work anymore. I also made a clean install and got this issue too. I didn't changed the configuration in any way.

The output of the command `tail -n 500 /var/log/syslog` is

`Oct 14 13:00:27 phoniebox PhonieboxGPIOControl[11157]: INFO:root:adding GPIO-Device, PlayPause
Oct 14 13:00:27 phoniebox PhonieboxGPIOControl[11157]: VolumeControl
Oct 14 13:00:27 phoniebox PhonieboxGPIOControl[11157]: PrevNextControl
Oct 14 13:00:27 phoniebox PhonieboxGPIOControl[11157]: PlayPause
Oct 14 13:00:27 phoniebox PhonieboxGPIOControl[11157]: Traceback (most recent call last):
Oct 14 13:00:27 phoniebox PhonieboxGPIOControl[11157]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 105, in <module>
Oct 14 13:00:27 phoniebox PhonieboxGPIOControl[11157]:     devices = get_all_devices(config)
Oct 14 13:00:27 phoniebox PhonieboxGPIOControl[11157]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 83, in get_all_devices
Oct 14 13:00:27 phoniebox PhonieboxGPIOControl[11157]:     device = generate_device(config[section], section)
Oct 14 13:00:27 phoniebox PhonieboxGPIOControl[11157]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 54, in generate_device
Oct 14 13:00:27 phoniebox PhonieboxGPIOControl[11157]:     pull_up_down=config.get('pull_up_down', fallback='GPIO.PUD_UP'))
Oct 14 13:00:27 phoniebox PhonieboxGPIOControl[11157]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/simple_button.py", line 66, in __init__
Oct 14 13:00:27 phoniebox PhonieboxGPIOControl[11157]:     self.pull_up_down = parse_pull_up_down(pull_up_down)
Oct 14 13:00:27 phoniebox PhonieboxGPIOControl[11157]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/simple_button.py", line 34, in parse_pull_up_down
Oct 14 13:00:27 phoniebox PhonieboxGPIOControl[11157]:     raise KeyError('Unknown Pull Up/Down type {pull_up_down}'.format(pull_up_down=pull_up_down))
Oct 14 13:00:27 phoniebox PhonieboxGPIOControl[11157]: KeyError: 'Unknown Pull Up/Down type GPIO.PUD_UP'
Oct 14 13:00:27 phoniebox systemd[1]: phoniebox-gpio-control.service: Main process exited, code=exited, status=1/FAILURE
Oct 14 13:00:27 phoniebox systemd[1]: phoniebox-gpio-control.service: Failed with result 'exit-code'.`

Berdsen, 2020-10-14:
So I could bring it back to work with changing the following line:
```gpio_control.py
pull_up_down=config.get('pull_up_down', fallback=GPIO.PUD_UP))
```

s-martin, 2020-10-14:
Thanks for digging into that.
Could you open a pull request fixing this?

Berdsen, 2020-10-14:
@s-martin I will fix this today in the evening and create a PR. Furthermore I found an issue with the volume up or down not using the configured volume step. Will perhaps also take a look into this.

PR #1088 (2020-10-14): fixed an issue where gpio controls didn't work properly
fixes #1078
commit b3fa3c34 (2020-10-14): fixed an issue where gpio controls didn't work properly
fixes #1078
closed 2020-10-14 by s-martin