MiczFlor/RPi-Jukebox-RFID #1080: I2C for LCD not working after phoniebox script installer runs and reboots

DSteg, 2020-10-08:
## System
Rasberry Pi 3A+
Raspbian Buster Lite

## Bug
I2C for LCD not working after phoniebox script installer and reboot

### What I did
On a fresh rasbian buster the i2c can be enabled and the display is detected by 'sudo i2cdetect -y 1' with ID 27
After reboot, display still found with 'sudo i2cdetect -y 1'
I can write text on the display with python code too.

No I am installing phoniebox with the script installer
After script is finished, I reboot the system.

### What happened
After system is rebooted, 'sudo i2cdetect -y 1' can't detect the display again.  'sudo i2cdetect -y 1' runs slow

If I disable and enable I2C again, 'sudo i2cdetect -y 1' can detect the display again.

I tried this many times, always the same problem occures.

Can someone else confirm this bug?

Thank you and regards
Dominik

And sorry for the bad english. :-)

domu83, 2020-10-09:
@DSteg you may want to check out issue #656

DSteg, 2020-10-09:
Thank you very much.

Yes, it is the same issue.
I have not recognized it myself, because I have the buttons still not connected.

If the GPIO3 (PIN5,SCL) is set to high, i2c is not working any longer.
So it is not possible to use the GPIO3 for shutdown + wake-up and any device with i2c at the same time.

In my opinion, this is a design failure of the Raspberry Pi constructors to set two functions on the same GPIO.
It would be better if the wake up function would be available on a free GPIO without a special function.

I have now set the shutdown GPIO to GPIO17 PIN11 and will try to create a switch function with a relay to use GPIO3 to wake up and GPIO 17 to shutdown.

Thank you again for the hint on issue #656

closed 2020-10-09 by DSteg