MiczFlor/RPi-Jukebox-RFID #1083: RFID is working only sometimes

Barzille, 2020-10-12:
## Bug

This is a strange behaviour I could not track down.
When we start or restart the raspberry it seems as if everything is initialized correctly. The rfid is beeping, the leds are blinking and also the jukebox startup sound is playing.

But every now and then the rfid reader is not recognizing any cards. It's not beeping when holding a card to it while the other time when we switched the raspi on it worked flawlessly. 

None of the logs show any errors or warnings which we could connect to the rfid reader.

### What I did

- I installed the raspberry pi with above mentioned buster image and ran the installer script
- Installed a on/off shim
- working with a power bank or wired current
- sed -i -e 's/sudo poweroff/sudo shutdown -h 0/g' /home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh

### What happened

unknown but even another RFID reader (same product type) is producing the same outcome

### I expected this to happen

RFID Reader is working every time correctly

### Further information that might help

tail -n 500 /var/log/syslog
Oct 12 20:35:12 raspberrypi kernel: [    2.596711] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
Oct 12 20:35:12 raspberrypi kernel: [    2.600572] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
Oct 12 20:35:12 raspberrypi kernel: [    2.605152] mmc0: sdhost-bcm2835 loaded - DMA enabled (>1)
Oct 12 20:35:12 raspberrypi kernel: [    2.609883] of_cfs_init
Oct 12 20:35:12 raspberrypi kernel: [    2.612255] of_cfs_init: OK
Oct 12 20:35:12 raspberrypi kernel: [    2.627939] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
Oct 12 20:35:12 raspberrypi kernel: [    2.628068] Waiting for root device PARTUUID=50d7c220-02...
Oct 12 20:35:12 raspberrypi kernel: [    2.636460] Indeed it is in host mode hprt0 = 00021501
Oct 12 20:35:12 raspberrypi kernel: [    2.676188] random: fast init done
Oct 12 20:35:12 raspberrypi kernel: [    2.697251] mmc0: host does not support reading read-only switch, assuming write-enable
Oct 12 20:35:12 raspberrypi kernel: [    2.705805] mmc0: new high speed SDHC card at address aaaa
Oct 12 20:35:12 raspberrypi kernel: [    2.709563] mmcblk0: mmc0:aaaa SC32G 29.7 GiB
Oct 12 20:35:12 raspberrypi kernel: [    2.715919]  mmcblk0: p1 p2
Oct 12 20:35:12 raspberrypi kernel: [    2.742023] mmc1: new high speed SDIO card at address 0001
Oct 12 20:35:12 raspberrypi kernel: [    2.761546] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
Oct 12 20:35:12 raspberrypi kernel: [    2.766016] VFS: Mounted root (ext4 filesystem) readonly on device 179:2.
Oct 12 20:35:12 raspberrypi kernel: [    2.778504] devtmpfs: mounted
Oct 12 20:35:12 raspberrypi kernel: [    2.788804] Freeing unused kernel memory: 1024K
Oct 12 20:35:12 raspberrypi kernel: [    2.796725] Run /sbin/init as init process
Oct 12 20:35:12 raspberrypi kernel: [    2.846336] usb 1-1: new high-speed USB device number 2 using dwc_otg
Oct 12 20:35:12 raspberrypi kernel: [    2.849067] Indeed it is in host mode hprt0 = 00001101
Oct 12 20:35:12 raspberrypi kernel: [    3.086698] usb 1-1: New USB device found, idVendor=0424, idProduct=9514, bcdDevice= 2.00
Oct 12 20:35:12 raspberrypi kernel: [    3.092002] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Oct 12 20:35:12 raspberrypi kernel: [    3.095633] hub 1-1:1.0: USB hub found
Oct 12 20:35:12 raspberrypi kernel: [    3.098576] hub 1-1:1.0: 5 ports detected
Oct 12 20:35:12 raspberrypi kernel: [    3.416330] usb 1-1.1: new high-speed USB device number 3 using dwc_otg
Oct 12 20:35:12 raspberrypi kernel: [    3.546704] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00, bcdDevice= 2.00
Oct 12 20:35:12 raspberrypi kernel: [    3.552173] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Oct 12 20:35:12 raspberrypi kernel: [    3.557854] smsc95xx v1.0.6
Oct 12 20:35:12 raspberrypi kernel: [    3.594586] NET: Registered protocol family 10
Oct 12 20:35:12 raspberrypi kernel: [    3.598834] Segment Routing with IPv6
Oct 12 20:35:12 raspberrypi kernel: [    3.653814] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:39:a5:84
Oct 12 20:35:12 raspberrypi kernel: [    3.660125] usb 1-1-port2: over-current change #1
Oct 12 20:35:12 raspberrypi kernel: [    4.216342] usb 1-1.3: new full-speed USB device number 4 using dwc_otg
Oct 12 20:35:12 raspberrypi kernel: [    4.238387] random: lvmconfig: uninitialized urandom read (4 bytes read)
Oct 12 20:35:12 raspberrypi kernel: [    4.354200] usb 1-1.3: New USB device found, idVendor=8086, idProduct=0808, bcdDevice= 1.00
Oct 12 20:35:12 raspberrypi kernel: [    4.360093] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Oct 12 20:35:12 raspberrypi kernel: [    4.363094] usb 1-1.3: Product: USB PnP Sound Device
Oct 12 20:35:12 raspberrypi kernel: [    4.365989] usb 1-1.3: Manufacturer: C-Media Electronics Inc.      
Oct 12 20:35:12 raspberrypi kernel: [    4.374653] input: C-Media Electronics Inc.       USB PnP Sound Device as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.3/0003:8086:0808.0001/input/input0
Oct 12 20:35:12 raspberrypi kernel: [    4.446613] hid-generic 0003:8086:0808.0001: input,hidraw0: USB HID v1.00 Device [C-Media Electronics Inc.       USB PnP Sound Device] on usb-3f980000.usb-1.3/input3
Oct 12 20:35:12 raspberrypi systemd[1]: Starting LSB: Switch to ondemand cpu governor (unless shift key is pressed)...
Oct 12 20:35:12 raspberrypi rngd[446]: entropy feed to the kernel ready
[... 641 lines cut ...]
GPIO.setup(btn_volU, GPIO.IN, pull_up_down=GPIO.PUD_UP)
GPIO.setup(btn_volD, GPIO.IN, pull_up_down=GPIO.PUD_UP)
GPIO.setup(btn_next, GPIO.IN, pull_up_down=GPIO.PUD_UP)
GPIO.setup(btn_prev, GPIO.IN, pull_up_down=GPIO.PUD_UP)
GPIO.setup(btn_halt, GPIO.IN, pull_up_down=GPIO.PUD_UP)
# GPIO.setup(btn_rfidoff, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)
GPIO.setup(led_power, GPIO.OUT)

# Set standard events for the buttons. Callback functions define the actions of the events (THey are defined above)
GPIO.add_event_detect(btn_shut, GPIO.FALLING, callback=shutdown_handler, bouncetime=bouncetime)
GPIO.add_event_detect(btn_vol0, GPIO.FALLING, callback=vol0_action, bouncetime=bouncetime)
GPIO.add_event_detect(btn_volU, GPIO.FALLING, callback=volU_handler, bouncetime=bouncetime)
GPIO.add_event_detect(btn_volD, GPIO.FALLING, callback=volD_handler, bouncetime=bouncetime)
GPIO.add_event_detect(btn_next, GPIO.FALLING, callback=next_action, bouncetime=bouncetime)
GPIO.add_event_detect(btn_prev, GPIO.FALLING, callback=prev_action, bouncetime=bouncetime)
GPIO.add_event_detect(btn_halt, GPIO.FALLING, callback=halt_action, bouncetime=bouncetime)
# GPIO.add_event_detect(btn_rfidoff, GPIO.FALLING, callback=rfidoff_action, bouncetime=bouncetime)
# reco.when_pressed = recordstart_action
# reco.when_released = recordstop_action
# play.when_pressed = recordplaylatest_action

# Switch on power led after boot to indicate state "on" for phoniebox
GPIO.output(led_power, GPIO.HIGH)

pause()

s-martin, 2020-10-12:
The reader seems to disconnect, see here:

```
Oct 12 20:38:29 raspberrypi kernel: [  149.126913] usb 1-1.2: USB disconnect, device number 5
Oct 12 20:38:29 raspberrypi thd[440]: Error reading device '/dev/input/event1'
Oct 12 20:38:29 raspberrypi daemon_rfid_reader.py[1282]: Traceback (most recent call last):
Oct 12 20:38:29 raspberrypi daemon_rfid_reader.py[1282]:   File "/home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py", line 38, in <module>
Oct 12 20:38:29 raspberrypi daemon_rfid_reader.py[1282]:     cardid = reader.reader.readCard()
Oct 12 20:38:29 raspberrypi daemon_rfid_reader.py[1282]:   File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 58, in readCard
Oct 12 20:38:29 raspberrypi daemon_rfid_reader.py[1282]:     for event in self.dev.read():
Oct 12 20:38:29 raspberrypi daemon_rfid_reader.py[1282]:   File "/usr/local/lib/python3.7/dist-packages/evdev/eventio.py", line 67, in read
Oct 12 20:38:29 raspberrypi daemon_rfid_reader.py[1282]:     events = _input.device_read_many(self.fd)
Oct 12 20:38:29 raspberrypi daemon_rfid_reader.py[1282]: OSError: [Errno 19] No such device
```

Could you please check, if there's an issue with the hardware, e.g. loose cable, something broken, etc?

Barzille, 2020-10-12:
I believe this one of our attempts to remove and reconnect the device.
We hav meanwhile updated to 2.1 and we use currently the Reader.py.experimental file.

What we currently see is, that after booting it is not working (same as in 2.0) but when we disconnect and reconnect the reader it seems to work.
What we also see is that the Reader Service is stopped after disconnecting the reader, so we have to enable it manually.

I have an assumption that maybe the reader is initialized too early in the boot process, is this possible?

One ting that's also strange: If we shutdown the RPI while music is playing and we start up again we first get the rfid beep, then the music starts and then - while the music plays - we get a startup sound melody. Maybe something in disorder?

s-martin, 2020-11-25:
Are you experiencing this issue in combination with resume or shuffle?

We released 2.2 a couple of days ago with fixes for both resume and shuffle, so you might want to check, if this fixes also your issue.

Barzille, 2020-12-23:
It's working fine meanwhile

closed 2020-12-23 by Barzille