MiczFlor/RPi-Jukebox-RFID #1084: 🐛 | Playbacklatest not working using RFID

alexander-sterzer, 2020-10-12:
## Bug

### What I did

I connected an usb microphone and wanted to play back the latest recording using an RFID token

### What happened

Latest recording was not played.

### I expected this to happen

Latest recording should play.

### Further information that might help

I found that in \scripts\rfid_trigger_play.sh
Line 269:             $PATHDATA/playout_controls.sh -c=recordplay**back**latest is called

However, in playout_controls.sh there is no recordplaybacklatest but only
Line 854:         recordplaylatest)

Therefore the RFID call goes nowhere.

## Software

### Base image and version
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Branch / Release
On branch master
Your branch is up to date with 'origin/master'.

### Installscript

scripts/installscripts/buster-install-default.sh

## Hardware

### RaspberryPi version

3 B+

### RFID Reader

16c0:27db HXGCoLtd Keyboard`

### Soundcard
n/a
### Other notable hardware

s-martin, 2020-10-12:
Thanks for finding this bug.

I opened a PR #1085 to fix this.

Could you check, if these changes fix your issue? Proably the easiest would be to just manually change this on your machine.

alexander-sterzer, 2020-10-13:
I changed it according to your fix and it solved the issue. Thanks!

closed 2020-10-13 by s-martin