MiczFlor/RPi-Jukebox-RFID #1087: Sometimes slow reaction on RFID scan or using the webbrowser control buttons

Huseriato, 2020-10-14:
## Bug

### What I did

I installed raspberry pi with the official raspberry os headless image and ran the installer script. I switched to the experimental RC522 script using this guide: https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components/rfid-reader/RC522

I uploaded two folders and some mp3 files using the webinterface and assigned an RFID card to each of this folders. Now I'm opening the player tab to see what's going on.

Now I'm switching between the cards (holding them in front of the rfid reader). 

Spotify Premium is connected and iris/mopidy is playing songs without any trouble (but Spotify is not used in this scenario).

### What happened

Sometimes all is fine and the tracks are playing instantly. Sometimes, there is a delay of some seconds, before a track starts. This is true, for RFID reading as for using the web interface to play a song.

If I use the iris interface to play the files, there is no lag between clicking and playing.

### I expected this to happen

Instant playback of the track.

### Further information that might help

syslog: https://paste.ubuntu.com/p/prPWwvS6fq/

You can see multiple detection of the card within 1 second. At the end you can see a detection, some seconds of waiting and than a final ok.

## Software

### Base image and version

    PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
    NAME="Raspbian GNU/Linux"
    VERSION_ID="10"
    VERSION="10 (buster)"
    VERSION_CODENAME=buster
    ID=raspbian
    ID_LIKE=debian

### Branch / Release

    Refresh index: 100% (390/390), done.
    On branch master
    Your branch is up to date with 'origin/master'.

### Installscript

I used variant A of this introduction via SSH: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL-stretch#install

## Hardware

### RaspberryPi version

Raspberry Pi 4 Model B Rev 1.2

### RFID Reader

RFID-RC522 made by joyit (8 pins, all connected)

### Soundcard

I'm currently using the headphone jack of the Raspberry PI for testing.

s-martin, 2020-10-14:
What’s the output of `cat /home/pi/RPi-Jukebox-RFID/settings/version-number`?

s-martin, 2020-10-14:
Could you check, if that’s the same issue as #1042? In 2.1 are some fixes from #1030 integrated.

Huseriato, 2020-10-15:
The version number is 2.1 - The fix from #1030 seems to be there. My setting is "ignore the same card for 2 seconds". I've no control cards used relating to my issue. But I tried now with a Volume Up card, which works finde (I configured 3% and got 3%). 

I think the repeating entries for scanning the cards are handled somehow. But the massive delay is real. You can see this on the end of the syslog.

This log is from starting one track. There are two errors, which may consume time:

```
Oct 15 08:01:28 kidsmusic daemon_rfid_reader.py[1001]: 2020-10-15 08:01:28,244 - Reader - INFO - Card detected.
Oct 15 08:01:28 kidsmusic daemon_rfid_reader.py[1001]: 2020-10-15 08:01:28,250 - Reader - INFO - 1792071111231
Oct 15 08:01:28 kidsmusic daemon_rfid_reader.py[1001]: 2020-10-15 08:01:28,250 - root - INFO - Trigger Play Cardid=1792071111231
Oct 15 08:01:28 kidsmusic mopidy[1048]: INFO     [MpdSession-105] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33130
Oct 15 08:01:28 kidsmusic mopidy[1048]: INFO     [MpdSession-106] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33134
Oct 15 08:01:30 kidsmusic mopidy[1048]: INFO     [MpdSession-107] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33138
Oct 15 08:01:30 kidsmusic mopidy[1048]: INFO     [MpdSession-108] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33142
Oct 15 08:01:31 kidsmusic mopidy[1048]: INFO     [MpdSession-109] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33146
Oct 15 08:01:31 kidsmusic daemon_rfid_reader.py[1001]: volume:100%   repeat: off   random: off   single: off   consume: off
Oct 15 08:01:31 kidsmusic mopidy[1048]: INFO     [MpdSession-110] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33150
Oct 15 08:01:31 kidsmusic mopidy[1048]: ERROR    [IrisFrontend-14] mopidy.listener Triggering event failed: tracklist_changed()
Oct 15 08:01:31 kidsmusic mopidy[1048]: Traceback (most recent call last):
Oct 15 08:01:31 kidsmusic mopidy[1048]:   File "/usr/local/lib/python3.7/dist-packages/mopidy/listener.py", line 40, in on_event
Oct 15 08:01:31 kidsmusic mopidy[1048]:     getattr(self, event)(**kwargs)
Oct 15 08:01:31 kidsmusic mopidy[1048]:   File "/usr/local/lib/python3.7/dist-packages/mopidy_iris/frontend.py", line 29, in tracklist_changed
Oct 15 08:01:31 kidsmusic mopidy[1048]:     iris.ioloop.add_callback(functools.partial(iris.clean_queue_metadata))
Oct 15 08:01:31 kidsmusic mopidy[1048]: AttributeError: 'NoneType' object has no attribute 'add_callback'
Oct 15 08:01:31 kidsmusic daemon_rfid_reader.py[1001]: volume:100%   repeat: off   random: off   single: off   consume: off
Oct 15 08:01:31 kidsmusic mopidy[1048]: INFO     [MpdSession-111] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33154
Oct 15 08:01:33 kidsmusic mopidy[1048]: ERROR    [IrisFrontend-14] mopidy.listener Triggering event failed: tracklist_changed()
Oct 15 08:01:33 kidsmusic mopidy[1048]: Traceback (most recent call last):
Oct 15 08:01:33 kidsmusic mopidy[1048]:   File "/usr/local/lib/python3.7/dist-packages/mopidy/listener.py", line 40, in on_event
Oct 15 08:01:33 kidsmusic mopidy[1048]:     getattr(self, event)(**kwargs)
Oct 15 08:01:33 kidsmusic mopidy[1048]:   File "/usr/local/lib/python3.7/dist-packages/mopidy_iris/frontend.py", line 29, in tracklist_changed
Oct 15 08:01:33 kidsmusic mopidy[1048]:     iris.ioloop.add_callback(functools.partial(iris.clean_queue_metadata))
Oct 15 08:01:33 kidsmusic mopidy[1048]: AttributeError: 'NoneType' object has no attribute 'add_callback'
Oct 15 08:01:33 kidsmusic daemon_rfid_reader.py[1001]: loading: Drei Fragezeichen % 001_und der Super-Papagei
Oct 15 08:01:33 kidsmusic mopidy[1048]: INFO     [MpdSession-112] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33158
Oct 15 08:01:33 kidsmusic daemon_rfid_reader.py[1001]: volume:100%   repeat: off   random: off   single: off   consume: off
Oct 15 08:01:33 kidsmusic mopidy[1048]: INFO     [MpdSession-113] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33162
Oct 15 08:01:33 kidsmusic daemon_rfid_reader.py[1001]: volume:100%   repeat: off   random: off   single: off   consume: off
Oct 15 08:01:33 kidsmusic mopidy[1048]: INFO     [MpdSession-114] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33166
Oct 15 08:01:33 kidsmusic daemon_rfid_reader.py[1001]: OK MPD 0.19.0
Oct 15 08:01:33 kidsmusic daemon_rfid_reader.py[1001]: OK
```

There are situations with the near same output, that are fast. But most of the time I have to wait several seconds before the track is played.

s-martin, 2020-10-15:
Could you try to check with `top`, which processes are running when swiping a card?

Pi 4 is the most powerful platform and as Phoniebox is supposed to run also on Pi Zero, so I'm wondering where this performance issue is coming from.

Huseriato, 2020-10-15:
mopidy is popping in with a load of ~20% for round about 1 second (I'm looking via ssh, so may be this is not really accurate). Sometimes there is python3 with ~20%. When running a playback mopidy has a cpu load ov ~5-7%.

Is there anything about the errors? May be it's a network problem and the script is waiting for something. That might explain the missing load. Is there any command which I can use to check the sdcard load?

Huseriato, 2020-10-15:
I just tested version 2.1.1 (updated via `git pull origin master`). The problem remains.

Huseriato, 2020-10-19:
Okay, sorry for any inconvenience. I think the problem had a very simple cause. I put my Phoniebox back on again from the very beginning, using a different SD card. It seems that the problem is somehow related to the older SD card. At the moment the delays do not occur anymore.

The error `ERROR    [IrisFrontend-14] mopidy.listener Triggering event failed: tracklist_changed()` still exists. I don't know, if this is okay or not. But I'm closing this now. To all other which have the same issues: Try to use a class 10 SD-Card.

closed 2020-10-19 by Huseriato