MiczFlor/RPi-Jukebox-RFID #1091: 🐛 | VolumeUp / VolumeDown not working properly with GPIO Buttons since 2.1

Berdsen, 2020-10-14:
## Bug
Since 2.1 and the new gpio buttons implementation the VolumeUp VolumeDown commands are not working properly. When I start and have a base sound level of 15 and I push the button up, it will become 63.

### What I did
I made a clean install (buster) to verify #1078 and ran into this issue. Having set up all as default and the buttons connected to the pins as mentioned in the old default setup guide.

### What happened
Sound is increased / decreased by wrong values

### Further information that might help
Debugged the playout_controls.sh and found out, that in the line with `VOLPERCENT=´expr ${VOLPERCENT} + \( ${AUDIOVOLCHANGESTEP} \* ${VALUE} \)´` value seems to be the value of the defined pin (e.g. 16 or 19 for VolumeUp or VolumeDown)

## Software

### Base image and version
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Branch / Release
master /fresh version 2.1

### Installscript
* scripts/installscripts/buster-install-default.sh

## Hardware

### RaspberryPi version

Raspberry Pi 3 Model B Rev 1.2

### RFID Reader
`16c0:27db HXGCoLtd Keyboard`

### Soundcard
i.e. `0d8c:0014 C-Media Electronics, Inc. Audio Adapter (Unitek Y-247A)`

### Other notable hardware
default gpio_settings but with enabled VolumeUp / VolumeDown buttons

Berdsen, 2020-10-14:
So I guess I found the problem. In simple_button.py the callbackFunctionHandler `def callbackFunctionHandler(self, *args)` will get the pressed pin as first argument (e.g. config volUp volDown on pin 16 and 19 and pressing them, I'll get args[0] 16 or 19). So we have to shift if the first argument if it is the same as self.pin.
```
    def callbackFunctionHandler(self, *args):
        logger.info('callbackFunctionHandler args: {}'.format(*args))
        if (len(args) > 0 and args[0] == self.pin):
            args = args[1:]

        if self.hold_repeat:
            return self.holdAndRepeatHandler(*args)
        logger.info('{}: executre callback'.format(self.name))
        return self.when_pressed(*args)
```

s-martin, 2020-10-14:
@veloxidSchweiz, can you help out here?

PR #1092 (2020-10-15): fixed an issue where arguments for playout_commands have the wrong value
fixes #1091
commit b892eda1 (2020-10-15): fixed an issue where arguments for playout_commands have the wrong value
fixes #1091
PR #1098 (2020-10-18): fixed unit tests
fixes #1091
commit 02a039d9 (2020-10-18): fixed unit tests
fixes #1091
closed 2020-10-15 by MiczFlor