MiczFlor/RPi-Jukebox-RFID #1093: 🐛 | GPIO scripts missing after installation

alexander-sterzer, 2020-10-16:
## Bug

Neither /RPi-Jukebox-RFID/scripts/gpio-buttons.py nor /RPi-Jukebox-RFID/misc/sampleconfigs/gpio-buttons.py.sample exists after using the one line installer --> GPIO Buttons cannot be integrated with Phoniebox.

### What I did

Followed the instructions from https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL-stretch to install the latest Raspian Buster/Phoniebox SW + Spotify. When asked about GPIO support I selected yes. 

### What happened

No GPIO scripts are available after installation. Both /RPi-Jukebox-RFID/misc/sampleconfigs/gpio-buttons.py.sample  and /RPi-Jukebox-RFID/scripts/gpio-buttons.py are not available to configure GPIOs.

### I expected this to happen

/RPi-Jukebox-RFID/scripts/gpio-buttons.py should be available after installation (as it was with previous installations).

### Further information that might help

pi@raspberrypi:~ $ journalctl -u mopidy
-- Logs begin at Fri 2020-10-16 11:53:30 BST, end at Fri 2020-10-16 12:17:09 BST
-- No entries --

## Software

### Base image and version

Otherwise the output of `cat /etc/os-release`

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Branch / Release

master

### Installscript

cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh

## Hardware

### RaspberryPi version

3 B+

### RFID Reader

16c0:27db HXGCoLtd Keyboard

### Soundcard

Hifiberry Miniamp

### Other notable hardware

GPIO settings not available

s-martin, 2020-10-16:
Since 2.0 these files are not used anymore, but [GPIO control](https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components/gpio_control) is used.

But as this isn't updated in the wiki yet, I consider this an documentation bug.

Thanks for the reminder ☺️

s-martin, 2020-10-18:
I added some updates to the wiki:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Using-GPIO-hardware-buttons

Dodo121188, 2020-10-19:
@s-martin  can you give me the path, where i can change the pins of the buttons, please?

s-martin, 2020-10-19:
Update the pins here:
 `~/RPi-Jukebox-RFID/settings/gpio_settings.ini` 

You can find more examples here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components/gpio_control/example_configs

Then restarting the service with `sudo systemctl restart phoniebox-gpio-control` will activate the new settings.

0x0lli, 2020-10-19:
nothing works for me.
RPI 3B+ / buster-full / installscript 2.11 / miniamp / 0,96" OLED / RFID RC-522 all up and running
only the rotary encoder isn't. Its connected à la https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Audio-RotaryKnobVolume without capacitors. I edited 
sudo nano /home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py
    config = configparser.ConfigParser(inline_comment_prefixes=";")
    config_path = os.path.expanduser('home/pi/RPi-Jukebox-RFID/setting/gpio_settings.ini')
#tried the above line using a slash before home ( config_path = os.path.expanduser('home/pi/RPi-Jukebox-RFID/setting/gpio_settings.ini') )
 
sudo nano /home/pi/RPi-Jukebox-RFID/settings/gpio_settings.ini:
[DEFAULT]
enabled: True
[VolumeControl]
enabled: True
Type: RotaryEncoder
PinUp: 5
PinDown: 6
timeBase: 0.02
functionCallDown: functionCallVolD
functionCallUp: functionCallVolU
#everything else is commented in bash

sudo nano /home/pi/RPi-Jukebox-RFID/components/gpio_control/install.sh
if [[ ! -f ~/.config/phoniebox/gpio_settings.ini ]]; then
    mkdir -p ~/.config/phoniebox && cp /home/pi/RPi-Jukebox-RFID/settings/gpio_settings.ini ~/.config/phoniebox/gpio_settings.ini

what does that line in the installer mean anyway ? there is nothing before or after the installer that is related to "config/phoniebox" at all.

Huseriato, 2020-10-20:
I saw your comment on #1101: I'm currently documenting my setup. I'm not sure, if I did something special about it, but I used another wiring than in the documentation and created a new gpio_settings.ini from scratch.

First of all, I'm using `DEBO ROT SWITCH` from reichelt. I don't know, if this is a different version (it's from JoyIt).

This is my wiring:

Rotary Encoder | PIN Raspberry PI | Description
-- | -- | --
GND | Pin 14 | Ground
VCC (+) | Pin 17 | 3.3V
DT | Pin 16 | GPIO 23
CLK | Pin 32 | GPIO 12

SW is unused. Theoretically this is pressing the button. But I think the press is nothing for the kits (they will rotate the knob, white pressing).

This is my VolumeControl section in the gpio_settings.ini:

```ini
[VolumeControl]
enabled: True
Type: RotaryEncoder
PinUp: 12
PinDown: 23
pull_up: True
hold_time: 0.3
timeBase: 0.1
functionCallUp: functionCallVolU
functionCallDown: functionCallVolD```

If this does not work, you can consider scripting a little bit. Because you might see better, what's happening behind the scenes:
https://thepihut.com/blogs/raspberry-pi-tutorials/how-to-use-a-rotary-encoder-with-the-raspberry-pi (this is not the solution, but more you can do this, to analyse the problem better).

Dodo121188, 2020-10-20:
@s-martin 
thx for help! 
if i change the "Vol.Change%" in settings to 5% and press the vol button, the volume goes 25% up, not 5%. Only if i set settings to 1% and in Editor "hold_time" to 0.1, the button turns the volume 5%. Is there another way to raise the volume to 5% per button when i set it ro 5% in settings,too?

s-martin, 2020-10-20:
@0x0lli:
It's important that you edit the gpio_settings.ini file you reference in gpio_control.py.

I suggest you use  `/home/pi/RPi-Jukebox-RFID/settings/gpio_settings.ini` (**first slash is necessary**)

What's the output of `tail -n 500 /var/log/syslog`?

0x0lli, 2020-10-20:
# @Huseriato @s-martin thanks for your help and patience!!!
after all, it was a cheap but oh-so-bitter-mistake that I made; apparently, I didnt copy-paste the directory information but wrote a part of it myself. an "s" was missing in .../setting/... 
shame on me

alexander-sterzer, 2020-10-22:
> sudo systemctl

> Update the pins here:
> `~/RPi-Jukebox-RFID/settings/gpio_settings.ini`
>
> You can find more examples here: [...]

I guess it would help many people to add the command how to restart the gpio service to the gpio wiki. I was doing many (time consuming) reboots when playing around with the gpio settings before finding this command.

themorlan, 2020-10-22:
@s-martin I wrote updates to the wiki pages. How can I submit them to you ?

s-martin, 2020-10-22:
> @s-martin I wrote updates to the wiki pages. How can I submit them to you ?

Can't you just click Edit on the page? I don't know, if special permissions are required.

Otherwise just post them here in this issue, if you like.

themorlan, 2020-10-22:
I did not want to deploy them before releasing 2.0 because they dont apply to current master version.

# Control Jukebox with buttons / GPIO

**Add buttons to your jukebox to control volume, skip tracks and more.**

Before we start:
One of the plus points about this projects, at least in my mind, 
was the fact that you don't need a soldering iron to build it.
Everything is USB, plug and play, thank you, boot and go.

Many, many fellow jukebox tweakers have contacted me to push
the envelope a bit further and add buttons to the jukebox.
Buttons to change the volume and skip between tracks in a playlist.

Enough said, here we go.

---

## Pin numbering on the RPi

On your RPi there are pin numbers printed on the board. In the following we are not referring to the board numbers, but the Broadcom (BCM) pin numbering for the GPIO pins. You can find more information on this issue on the [pin numbering](https://gpiozero.readthedocs.io/en/stable/recipes.html#pin-numbering) section of the GPIO Zero documentation site.

![Any pin marked “GPIO” in the diagram below can be used as a pin number. For example, if an LED was attached to “GPIO17” you would specify the pin number as 17 rather than 11.](img/GPIO-pin-numbering.png
 "Any pin marked “GPIO” in the diagram below can be used as a pin number. For example, if an LED was attached to “GPIO17” you would specify the pin number as 17 rather than 11.")

Any pin marked “GPIO” in the diagram above can be used as a pin number. A button attached to “GPIO17” would be specified in the script as pin number 17 rather than 11 (which would be the count on the board).

---

## Update September 2020
The service which provides the functionality of GPIO-Control-Devices has seen numerous changes in the past. The most recent version was originally designed by @veloxidSchweiz und @veloxid. It has now been integrated in the one-line installscript and support also RotaryEncoders in addition to normal buttons.

## How to connect the buttons
In the default configuration the GPIO-control-service uses the [new pinout](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Hardware-Pinout-Overview#flexible-pin-out) for minimal interferance with other GPIO-Devices, which are fixed on some PINs.

Here is how to connect the buttons using the new pinout:

* **Volume Down** GPIO6 (PIN35) and GND
* **Volume Up** GPIO5 (PIN29) and GND
* **Play/Pause/Halt (or how you call it)** GPIO27 (PIN13) and GND
* **Next** GPIO23 (PIN16) and GND
* **Previous** GPIO22 (PIN15) and GND
* **Shutdown (you need to hold button for 2 secs for shutdown)** GPIO3 (PIN5) and GND

**!!! IMPORTANT Only when using the above listed pins for wiring your shutdown pin you will be able to power-up the Raspberry PI from firmware halt. !!!**

## How to install the GPIO-Control-Service
When using the interactive version of the one-line installscript you will be asked if you want to activate the GPIO-Control-Service. 
If you want to install the service outside the one-line installscript you need the command `sudo bash /home/pi/RPi-Jukebox-RFID/components/gpio_control/install.sh` 

## Where to find the configuration of the GPIO-Service
The configuration file can be found here: `/home/pi/RPi-Jukebox-RFID/settings/gpio_settings.ini`

## Additional information:

* [Getting started with soldering on raspberrypi.org](https://www.raspberrypi.org/blog/getting-started-soldering/)
* [Switch and buttons basics on sparkfun.com](https://learn.sparkfun.com/tutorials/switch-basics)

### Circuit example(s) for the button wiring

You will be using push buttons, which are essentially the same as arcade buttons, meaning: when you press them down, they are ON, when you let go, the are OFF. So skipping a track is tapping a button once, changing the volume, each tap changes it a bit. The updated GPIO from pullrequest #306 added the option to hold down the volume buttons and change every 0.3 seconds the volume according to the configured volume steps.

There are a number of different ways to connect a button. The easiest one is well explained on O'Reilly's RPi site:

* [Connecting a Push Switch, O'Reilly](http://razzpisampler.oreilly.com/ch07.html)
* Watch the [Connecting a Push Switch with Raspberry Pi video on YouTube](https://youtu.be/3TDJ4FmtGgk)

-------------------------------------------------------------------

[... 20 lines cut ...]
  |               |                      |               |
  |           SW  |                      |               |
  |               |                      |               |
  |           +   |----------------------| 3.3V          |
  |               |                      |               |
  |           GND |----------------------| GND           |
  |               |                      |               |
  '---------------'                      '---------------'
       KY-040                                Raspberry

```
Once you've got the hardware done, edit `/home/pi/RPi-Jukebox-RFID/settings/gpio_settings.ini` and change:
```
[VolumeControl]
enabled: True
Type: TwoButtonControl ;or RotaryEncoder
```
to
```
[VolumeControl]
enabled: True
Type: RotaryEncoder
```

After that restart the GPIO-Service: `sudo systemctl restart phoniebox-gpio-control.service`

Huseriato, 2020-10-23:
> * **Volume Down** GPIO6 (PIN35) and GND

GPIO6 is Pin 31.
PIN35 is GPIO19

So something is wrong here. Thank your for documentation. I prefer not to use PIN35, because it's reserved by HifiBerry Boards (Sound interface): https://www.hifiberry.com/docs/hardware/gpio-usage-of-hifiberry-boards/

s-martin, 2020-10-23:
@Huseriato good catch

s-martin, 2020-11-29:
Updated the wiki and users report it works, so close this here.

Please feel free to reopen the issue, if you still experience problems.

closed 2020-11-29 by s-martin