MiczFlor/RPi-Jukebox-RFID #1094: 🐛 | BUG SUMMARY

olli83, 2020-10-16:
## Bug
Status LED not working

### What I did
Upgraded to 2.1.1
Status LED connected to GPIO12, was working before.

### I expected this to happen

Expected the status LED to continue function.

### Further information that might help

Status LED is not listed in the new GPIO settings file. It appears an entry is missing.

s-martin, 2020-10-16:
There is documentation missing.

Could you please try to add

```
[StatusLED] 
enable: True 
Type: MPDStatusLED 
Pin: 12
```

to the GPIO settings file?

See also #900

s-martin, 2020-10-16:
See also https://www.github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components%2Fgpio_control%2Fexample_configs%2Fgpio_settings_rotary_and_led.ini

olli83, 2020-10-16:
Nope, not working. Also tried "enabled" instead of "enable", I am confused as you link above use enable several times, whereas in the previous version it was always "enabled".
Also, would it not need a pull_up: true or similar? Tried that, but did not change anything.

s-martin, 2020-10-16:
Did you restart the service?

s-martin, 2020-10-17:
Please see #1096, the paths for `gpio_settings.ini` were mixed up. This is now fixed in `develop` branch.

Please give some info, if these changes fix this issue.

s-martin, 2020-10-18:
I added some updates to the wiki
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Using-GPIO-hardware-buttons

topas-rec, 2020-10-19:
(Accidently saw this issue - haven't updated my box for longer)
I have made some documentation at https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Using-GPIO-hardware-status-led about the status led. Perhaps that helps or has to be updated to recent or upcoming changes.

s-martin, 2020-10-19:
> (Accidently saw this issue - haven't updated my box for longer)
> I have made some documentation at https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Using-GPIO-hardware-status-led about the status led. Perhaps that helps or has to be updated to recent or upcoming changes.

I’ve added that article to the navigation bar, so people will find it 😁

olli83, 2020-10-20:
Solved by #1096 and with the above comments. Thanks!

closed 2020-10-20 by olli83

topas-rec, 2020-11-23:
Thanks @s-martin, but that entry was already there and is now doubled. :smiley: 
[image]
I thinks yours is more intuitive to find and I'll remove my entry in the navigation bar.

s-martin, 2020-11-23:
:D sorry, didn't see that.

[Screenshot from 2020-11-23 17-02-01]<https://user-images.githubusercontent.com/20391449/99985019-12543500-2dae-11eb-9439-04eeebb8d7d5.png>

s-martin, 2020-11-23:
Sorry, didn't see that.