MiczFlor/RPi-Jukebox-RFID #1095: 🐛 | Rotary button does not control volume properly

olli83, 2020-10-17:
## Bug
The rotary button leads to random volume changes. When I turn it, volume goes up. Further turn in same direction leads to volume up or down randomly. Same the other way. No predictable behaviour.

### What I did
Installed the Ky-040 rotary dial. Double checked the wiring.

Noticed that https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Audio-RotaryKnobVolume shows CLK and DATA Pins connected to GPIO 5 and 6 the other way round than described on https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Hardware-Pinout-Overview

Tried both variants. Same result.

### I expected this to happen

Volume up and down in line with button rotation.

### Further information that might help

Latest Phoniebox 2.1.1 install

## Hardware

### RaspberryPi version

Yordan1976, 2020-10-17:
I have the same issue on a fresh install (2.1.1 - 70f989b - master). I use the Classic Pinout since i built the box in march 2019.

s-martin, 2020-10-17:
You should use gpio_settings.ini.

Please see #1096, the paths for `gpio_settings.ini` were mixed up. This is now fixed in `develop` branch.

Please give some info, if these changes fix this issue.

Yordan1976, 2020-10-17:
I copied the gpio_settings.ini-file to the new path and it seems to work. How do I restart the service to let changes take effect without rebooting?

s-martin, 2020-10-17:
`sudo systemctl restart phoniebox-gpio-control`

olli83, 2020-10-20:
Yes, thank you. #1096 solved it and it is now working fine.

closed 2020-10-20 by olli83