MiczFlor/RPi-Jukebox-RFID #1097: 🚀 RC522: Stop on RFID removal

mattelacchiato, 2020-10-17:
## RC522: Stop on RFID removal

### What functionality would you like to see in your phoniebox?

In contrast to USB card readers (#43), the RC522 Reader can continuously scan which card is currently present. It would be awesome to have a "tonie-like" behaviour: As long as the card is present, the folder is played and when removed, the playback is paused.

### How do you envision the feature to work from a users perspective?

This is the most intuitive way of handling Play/Pause, I can imagine!

s-martin, 2020-10-17:
Please check #1049, if this fixes your issue.

mattelacchiato, 2020-10-20:
That's hard to tell from a user perspective. Maybe it's a groundwork for my wished feature, but I'm not that familiar with the details...

Toqqi, 2020-10-21:
Hi, I wanted to have exactly the same "pause on removal" functionality as described.

I was able to achieve a satisfying behavior by modifying [/scripts/daemon_rfid_reader.py](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/scripts/daemon_rfid_reader.py) as posted in https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/786ecf4874db54c0fd6257a7aa8cf8fe611c0232.

closed 2021-02-21 by s-martin

damaev, 2021-04-23:
I ran into the issue of a high cpu-load of the reader-service on my pi zero. By decreasing its nice-level the impact is reduced significantly