MiczFlor/RPi-Jukebox-RFID #1101: 🚀 Call a command before something is played and when playing stops

Huseriato, 2020-10-19:
## Feature Description

### What functionality would you like to see in your phoniebox?

I would like to disable the crackling sound of some HifiBerry boards when powering on or off. In my example the HifiBerry MiniAMP. Also I want to stop the little noise, when nothing is played.

### How do you envision the feature to work from a users perspective?

There should be no unneeded cracklings. The device should be muted, when not playing anything. The HifiBerry MiniAMP supports a MUTE input on GPIO 16. I created a topic in german Raspberry PI forum explaining the problem (German): https://forum-raspberrypi.de/forum/thread/49514-hifiberry-miniamp-und-das-knacken-im-zusammenhang-mit-dem-onoff-shim/

### Further information that might help

Setting GPIO 16 to GND (with a 10kOhm resistor) mutes the HifiBerry MiniAMP. Because the MiniAMP has a 5V Pull up with 10kOhm ths GPIO 16 is considered as in input PIN on Boot time. 

So before playback you can call `raspi-gpio set 16 op dh` to unmute the MiniAMP. But I need an entry point, without blocking updates. So it would be really nice to execute something before playback, that is not reverted on update. Same for the situation after a playback. I don't know how to deal with system sound situations. May be its more like a trigger named "activate_sound.sh" and "deactivate_sound.sh" or something similar.

0x0lli, 2020-10-19:
hi, habe gerade Dein setup gesehen und bin am verzweifeln; was hast Du gemacht, um den rotary encoder zum Laufen zu kriegen?
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1093#issuecomment-712481154

Huseriato, 2020-10-30:
I tried something relative simple to have an event before and after any command. I've added before

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/scripts/playout_controls.sh#L191

This is my testcode:

```bash
if [ -f $PATHDATA/../events/$COMMAND.before.sh ];
then
  if [ "${DEBUG_playout_controls_sh}" == "TRUE" ]; then echo "   Executing ${COMMAND}.before.sh..." >> ${PATHDATA}/../logs/debug.log; fi

  $PATHDATA/../events/$COMMAND.before.sh

  if [ $? != "0" ]; 
  then
    if [ "${DEBUG_playout_controls_sh}" == "TRUE" ]; then echo "   ${COMMAND}.before.sh returned exit code ${?}, ${COMMAND} cancelled." >> ${PATHDATA}/../logs/debug.log; fi
    exit $?
  fi;
fi;
```

So the idea is, that everyone can place files in `~/RPi-Jukebox-RFID/events` with the format `<Command>.before.sh` or `<Command>.after.sh` - So it easy to extend the functionally of every command. In the before events the whole command can be cancelled, when sending an exitcode <> 0. I've a similar block like the one above at the end of `playout_controls.sh`.

Do you think this might be a good idea to implement it this way?