MiczFlor/RPi-Jukebox-RFID #1102: Problem with RFID reader service using a PN532

daved3luxe, 2020-10-19:
I use a PN532 via I2C, it is found and also reads cards via CLI.
However, the service cannot be started.

Installation:
Activating I2C interface...
Installing i2c-tools...
Checking if PN532 RFID reader is found through I2C...
20: -- -- -- -- 24 -- -- -- -- -- -- -- -- -- -- --
  PN532 was found.
Installing Python requirements for PN532...
Configure RFID reader in Phoniebox...
Restarting phoniebox-rfid-reader service...
Done.

Service:
sudo systemctl status phoniebox-rfid-reader
● phoniebox-rfid-reader.service - Phoniebox RFID-Reader Service
   Loaded: loaded (/etc/systemd/system/phoniebox-rfid-reader.service; enabled; v
   Active: failed (Result: exit-code) since Tue 2020-10-20 00:14:38 CEST; 24s ag
  Process: 8392 ExecStart=/home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.p
 Main PID: 8392 (code=exited, status=1/FAILURE)

Oct 20 00:14:38 Phoniebox systemd[1]: phoniebox-rfid-reader.service: Servic
Oct 20 00:14:38 Phoniebox systemd[1]: phoniebox-rfid-reader.service: Schedu
Oct 20 00:14:38 Phoniebox systemd[1]: Stopped Phoniebox RFID-Reader Service
Oct 20 00:14:38 Phoniebox systemd[1]: phoniebox-rfid-reader.service: Start
Oct 20 00:14:38 Phoniebox systemd[1]: phoniebox-rfid-reader.service: Failed
Oct 20 00:14:38 Phoniebox systemd[1]: Failed to start Phoniebox RFID-Reader

s-martin, 2020-10-20:
It's possible that some unrelated dependencies are missing, please try:
`sudo python3 -m pip install --upgrade --force-reinstall -q -r /home/pi/RPi-Jukebox-RFID/components/rfid-reader/RC522/requirements.txt`

Afterwards restart the service:
`sudo systemctl restart phoniebox-rfid-reader`

daved3luxe, 2020-10-20:
Thank you, that worked

closed 2020-10-20 by daved3luxe