MiczFlor/RPi-Jukebox-RFID #1103: 🐛 | BUG SUMMARY

0x0lli, 2020-10-19:
## Bug

web-interface not working

### What I did

install Raspberry Pi OS full edition, ran installer script, configured hifiberry miniamp, then tried to see the web interface

### What happened

page doesnt load. it is the first time i try to see the web interface on rpiOS-full, it did work on rpiOS-lite edition

### I expected this to happen

regular web interface

### Further information that might help

https://pastebin.com/BJK5Hgj6

## Software

### Base image and version

2020-08-20-raspios-buster-armhf-full

### Branch / Release

"Your branch is up to date with 'origin/master'."

### Installscript

scripts/installscripts/buster-install-default.sh
Version 2.1.1

## Hardware

pi0
hifiberry miniamp
RFID-RC522
0,96" OLED display, 1106 chipset
rotary encoder (never worked on any release since first installing v2.0, it is connected with four cables and did work in two previous setups using volumio and snapcast)

### RaspberryPi version

Raspberry Pi Zero W Rev 1.1

### RFID Reader

RFID-RC522

### Soundcard

hifiberry miniamp

### Other notable hardware

0.96" OLED display with 1106 chipset
one rotary encoder KY040   [(two) running rotary encoder for volume and track, aswell as one GPIO button for play/pause is my ultimate goal]

s-martin, 2020-10-20:
Could you additionally provide `tail -n 500 /var/log/syslog` and the output of `journalctl -u lighttpd`?

0x0lli, 2020-10-20:
Flashed over this SD-card setup two times already so no mor elogs to post, sorry

closed 2020-10-20 by 0x0lli