MiczFlor/RPi-Jukebox-RFID #1104: 🐛 | BUG SUMMARY

Huseriato, 2020-10-20:
## Bug

### What I did

I changed my default media directory in the installation script to `/media/usb`.

### What happened

The new directory is used in the Phoniebox configuration, but not adjusted in `/etc/mopidy/mopidy.conf`. I have to change it manually.

### I expected this to happen

The configuration file of Mopdiy should be automatically adjusted.

### Further information that might help

I use the following command in my documentation to change this: `sudo sed -i -e 's=/home/pi/RPi-Jukebox-RFID/shared/audiofolders=/media/usb=g' /etc/mopidy/mopidy.conf`

## Software

Phoniebox 2.1.1

### Installscript

scripts/installscripts/buster-install-default.sh

s-martin, 2020-10-20:
If you want, you can add a PR for the changes in the install script.

Huseriato, 2020-10-20:
Theoretically yes, but I cannot verify this for the 3 other install scripts. I also have no automated test for this. So I have to install from scratch for every test. This is a little annoying.

I think it has to be something like in line 910-933 of the `buster-install-default.sh`, which is relative easy to replicate for the media folder. I think the sample has to be replaced, too (https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/misc/sampleconfigs/mopidy-etc.sample) - I think this is easy, when having a full setup for development. But I don't have one, yet. So it may be I can change this for my own fork in the buster-install script, but I cannot guarantee it will work.

s-martin, 2020-10-20:
Thanks for the pointers, I'll look into it.

PR #1108 (2020-10-21): Fix custom mopidy media dir
Fixes #1104 
including unit tests for spotify installation
commit 205135a3 (2020-10-21): make sure, custom audio folder is also used in mopidy; fixes #1104
closed 2020-10-21 by s-martin