MiczFlor/RPi-Jukebox-RFID #1105: 🐛 | RC522 and PN532 require each other's dependencies

s-martin, 2020-10-20:
## Bug

### What I did

Normally a user just uses either RC522 or PN532 reader.

### What happened

Due to the imports in https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/scripts/Reader.py.experimental 
currently Python libs from both readers are required.

### I expected this to happen

Only Python libs for the specific reader are required.

### Further information that might help

Bugs are e.g. #1102.

Possible solution would be to separate the classes for RC523 and PN532

## Software

### Base image and version
N/A

### Branch / Release

At least since 2.1

### Installscript

N/A

## Hardware
N/A

arne123, 2021-02-16:
Python does allow conditional imports,
so one solution could be to separate all sub classes in separate files with their necessary imports, and just import these in the class chooser before instantiation.

Probably it will also work, if you put the imports in the __init__ functions of the subclass, as far as I have seen in this particular case the only reference is needed here. 
It seems also to be possible to import within the class namespace like this
class xxx:
    import yyy

    def zzzz:

s-martin, 2021-02-16:
If you have any proposals or PR, very happy to integrate or support....

closed 2021-02-23 by MiczFlor