MiczFlor/RPi-Jukebox-RFID #1106: 🚀 Skip WINS configuration in install script when installing Samba

Huseriato, 2020-10-20:
## Feature Description

### What functionality would you like to see in your phoniebox?

The install script is somewhere in the middle of the installation paused by the following screen:

```
Samba server and utilities

If your computer gets IP address information from a DHCP server on the network,
the DHCP server may also provide information about WINS servers
("NetBIOS name servers") present on the network.
This requires a change to your smb.conf file so that DHCP-provided WINS settings
will automatically be read from /var/lib/samba/dhcp.conf.

The dhcp-client package must be installed to take advantage of this feature.
Modify smb.conf to use WINS settings from DHCP?

<Yes>   <No>
```

I would like to skip this step. Because the installation will not work unattended. You have to return to the installation process, just to press `<Yes>` or `<No>`.

### How do you envision the feature to work from a users perspective?

Skip this dialog completely. I don't think, that any home user will have a WINS servers running. If you think this is a point, this question should be asked in the wizard at the beginning.

### Further information that might help

It's possible to send some information to the apt configuration before installing an package to automate this. I found something here: https://unix.stackexchange.com/questions/546470/skip-prompt-when-installing-samba

I think before executing this line (784 in `buster-install-default.sh`): 

`${apt_get} ${allow_downgrades} install samba samba-common-bin gcc lighttpd php7.3-common php7.3-cgi php7.3 at mpd mpc mpg123 git ffmpeg resolvconf spi-tools`

we can simply add the lines from stackexchange:

```
echo "samba-common samba-common/workgroup string WORKGROUP" | sudo debconf-set-selections
echo "samba-common samba-common/dhcp boolean false" | sudo debconf-set-selections
echo "samba-common samba-common/do_debconf boolean true" | sudo debconf-set-selections
```

May be the install script can ask for the desired workgroup. 

I hope I have no more bugs or feature requests today. I thank all authors for making the phoniebox working so nicely out of the box.

s-martin, 2020-10-20:
For the non-interactive installation mode, we already have this:

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/4185897a85ccf6ec2d93af2c5ff358346ff93dd8/scripts/installscripts/buster-install-default.sh#L1231

Huseriato, 2020-10-21:
I think that has to move on top of the surrounding.

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/4185897a85ccf6ec2d93af2c5ff358346ff93dd8/scripts/installscripts/buster-install-default.sh#L1217

The WINS configuration part should not be part of the interactive installation.

Huseriato, 2020-10-21:
So I checked this now in a fresh installation. Moving the line with

```bash
echo "samba-common samba-common/dhcp boolean false" | sudo debconf-set-selections
```

above line 1217 skips the WINS configuration. I made an Installation of my fork. 

Can anyone explain me how to make a pull request with one file/some lines changed? Github itself is always merging the whole fork into its parent. But I hat to change the git location at the beginning of my script to test this. Github is really missing a 1:1 communication feature.

PR #1115 (2020-10-23): Fix Samba WINS configuration
fixes  #1106
closed 2020-10-23 by s-martin