MiczFlor/RPi-Jukebox-RFID #1107: 🐛 | (Pi zero) on buster lite will not run GPIO-control using Phoniebox Script v2.1.1

0x0lli, 2020-10-20:
## Bug
pi zero on buster lite will not run GPIO-control using Phoniebox Script v2.1.1

### What I did

regular install of buster lite, configure hifiberry miniamp, RC-522, ,096" SH-1106 OLED

### What happened

when installing the GPIO control service after editing the .ini and .py files according to known issue-fixes 

sudo nano /home/pi/RPi-Jukebox-RFID/settings/gpio_settings.ini
sudo nano /home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py
sudo /home/pi/RPi-Jukebox-RFID/components/gpio_control/install.sh

it would abort the installation and give out error, I edited within the install.sh the requirement.txt into ./requirement.txt but the problem remained.
(I think it was error2 but am not sure as I have overwritten the SD again, sorry for that)

### I expected this to happen

same as in buster-full : 
working GPIO-control-service or
requirement.txt fetching the needed dependencies

### Further information that might help

journalctl -u phoniebox-gpio-control.service -f

Oct 20 18:43:14 phoniebox systemd[1]: phoniebox-gpio-control.service: Main process exited, code=exited, status=1/FAILURE
Oct 20 18:43:14 phoniebox systemd[1]: phoniebox-gpio-control.service: Failed with result 'exit-code'.
Oct 20 18:43:17 phoniebox systemd[1]: phoniebox-gpio-control.service: Service RestartSec=3s expired, scheduling restart.
Oct 20 18:43:17 phoniebox systemd[1]: phoniebox-gpio-control.service: Scheduled restart job, restart counter is at 99.
Oct 20 18:43:17 phoniebox systemd[1]: Stopped Phoniebox GPIO Control Service.
Oct 20 18:43:17 phoniebox systemd[1]: Started Phoniebox GPIO Control Service.
Oct 20 18:43:18 phoniebox PhonieboxGPIOControl[13424]: INFO:root:['VolumeControl', 'PrevNextControl']
Oct 20 18:43:18 phoniebox PhonieboxGPIOControl[13424]: INFO:root:adding GPIO-Device, VolumeControl
Oct 20 18:43:18 phoniebox PhonieboxGPIOControl[13424]: INFO:root:adding GPIO-Device, PrevNextControl
Oct 20 18:43:18 phoniebox PhonieboxGPIOControl[13424]: VolumeControl
Oct 20 18:43:18 phoniebox PhonieboxGPIOControl[13424]: PrevNextControl
Oct 20 18:43:18 phoniebox PhonieboxGPIOControl[13424]: Traceback (most recent call last):
Oct 20 18:43:18 phoniebox PhonieboxGPIOControl[13424]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 104, in <module>
Oct 20 18:43:18 phoniebox PhonieboxGPIOControl[13424]:     devices = get_all_devices(config)
Oct 20 18:43:18 phoniebox PhonieboxGPIOControl[13424]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 82, in get_all_devices
Oct 20 18:43:18 phoniebox PhonieboxGPIOControl[13424]:     device = generate_device(config[section], section)
Oct 20 18:43:18 phoniebox PhonieboxGPIOControl[13424]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 68, in generate_device
Oct 20 18:43:18 phoniebox PhonieboxGPIOControl[13424]:     getFunctionCall(config.get('functionCallUp')),
Oct 20 18:43:18 phoniebox PhonieboxGPIOControl[13424]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 23, in getFunctionCall
Oct 20 18:43:18 phoniebox PhonieboxGPIOControl[13424]:     return getattr(function_calls, function_name)
Oct 20 18:43:18 phoniebox PhonieboxGPIOControl[13424]: TypeError: getattr(): attribute name must be string

i.e. `find logfiles at https://paste.ubuntu.com/p/cRS7qM8ZmP/`
-->

## Software

### Base image and version

2020-08-20-raspios-buster-armhf-lite
updated and -graded

### Branch / Release

master, v2.1.1

### Installscript

scripts/installscripts/buster-install-default.sh
v2.1.1

## Hardware

### RaspberryPi version
pi zer ov1.1

### RFID Reader

RC-522

### Soundcard

hifiberry miniamp

### Other notable hardware

SH-1106, successfully installed via splitti:
cd; rm o4p_installer.sh; wget https://raw.githubusercontent.com/splitti/oled_phoniebox/master/scripts/install/o4p_installer.sh; chmod +x o4p_installer.sh; ./o4p_installer.sh

s-martin, 2020-10-20:
Please post the content of your `gpio_settings.ini`.

In addition the installer writes log files in the home directory, please post the log file as well.

0x0lli, 2020-10-20:
/home/pi/RPi-Jukebox-RFID/settings/gpio_settings.ini:

[DEFAULT]
enabled: True
[VolumeControl]
enabled: True
Type: RotaryEncoder
PinUp: 5
PinDown: 6
pull_up: True
timeBase: 0.1
functionCallDown: functionCallVolD
functionCallUp: functionCallVolU

[PrevNextControl]
enabled: True
Type: RotaryEncoder
Pin1: 12
Pin2: 23
functionCall1: functionCallPlayerPrev
functionCall2: functionCallPlayerNext
functionCallTwoButtons: None
pull_up: True
hold_time: 0.3
hold_repeat: False

#[PlayPause]
#enabled: True
#Type: Button
#Pin: 26
#pull_up: True
#hold_time: 0.3
#functionCall: functionCallPlayerPause

0x0lli, 2020-10-20:
pi@phoniebox:~ $ sudo /home/pi/RPi-Jukebox-RFID/components/gpio_control/install.sh

disable old services: phoniebox-gpio-buttons and phoniebox-rotary-encoder
Failed to stop phoniebox-rotary-encoder.service: Unit phoniebox-rotary-encoder.service not loaded.
Failed to disable unit: Unit file phoniebox-rotary-encoder.service does not exist.
Failed to stop phoniebox-gpio-buttons.service: Unit phoniebox-gpio-buttons.service not loaded.
Failed to disable unit: Unit file phoniebox-gpio-buttons.service does not exist.
Install all required python modules
Could not open requirements file: [Errno 2] No such file or directory: 'requirements.txt'

Installing GPIO_Control service
Press enter to continue
/etc/systemd/system/phoniebox-gpio-control.service exists.
systemctl daemon-reload
restarting service
Press enter to continue

Problem during installation occured
   Service not running, please check functionallity by running gpio_control.py
   in the directory ~/RPi-Jukebox-RFID/components/gpio_control:
      $ cd ~/RPi-Jukebox-RFID/components/gpio_control
      $ python gpio_control.py
   or check output of journaclctl by:
      $ journalctl -u phoniebox-gpio-control.service -f

0x0lli, 2020-10-20:
apparently, I made another mistake configuring:
/home/pi/RPi-Jukebox-RFID/settings/gpio_settings.ini
although I cannot reproduce this, I replaced the above file with:
/home/pi/RPi-Jukebox-RFID/components/gpio_control/example_configs/gpio_setting_rotary_vol_prevnext.ini
@s-martin thanks again for your patience, I am a happy uncle this evening. I think there is little left that I can mess up by now :) care for a beer? coffee is not my cup of tea :)

closed 2020-10-20 by 0x0lli