MiczFlor/RPi-Jukebox-RFID #1109: Pimoroni Pirate Audio

clemsonfire, 2020-10-22:
Guten Morgen!

Ich habe nun meine Phoniebox schweren Herzens auf USB Reader umgerüstet und mir Pimoroni Audio Hat 3W ausgestattet. Die Grundfunktionen funktionieren, doch leider nicht das Display und die Tasten des Audio HAT.
Ich habe gemäß ~/RPi-Jukebox-RFID/com~/RPi-Jukebox-RFID/components/audio/PirateAudioHAT/README.md installiert, Develop Version ist es auch.

Wo muss ich ansetzen, um das Display Problem zu lösen und die Tasten zum Funktionieren zu bewegen?
Das Display scheint leicht zu leuchten, doch zeigt es nix an, weder Cover-Art noch sonstwas.
Bitte um Eure geschätzte Hilfe!
Im voraus sagt danke
clemsonfire

s-martin, 2020-10-23:
Might be interfering with standard GPIO control.

Please try 
```
sudo systemctl stop phoniebox-gpio-control.service 
sudo systemctl disable phoniebox-gpio-control.service
```

Maybe you need to reboot the Pi.

clemsonfire, 2020-10-25:
Hi s-martin,
thank you for your answer!
i did your console proposal, but display is still empty - no cover art, no soft-keys butt seems to be backlighted.

any further ideas?

TIA! 
Clemsonfire

s-martin, 2020-10-25:
Do you use classic or Spotify edition of Phoniebox?

clemsonfire, 2020-10-26:
Hi, s-martin,
Thanks for your interest!
I'm using
	
2.2rc1 - caeee32 - develop Edition
Classic edition (barebones)

s-martin, 2020-10-26:
So I had a closer look and the current script just works with Mopidy, which we only use in the Spotify edition.

So as a quick solution you could install the Spotify edition. This one usually starts a little slower, but otherwise the functionality is the same. Sometimes there are difficulties making it run (look for issues here tagged with spotify).

If you want to make it run with classic edition first you need to update the pins of the buttons of the HAT in settings/gpio_settings.ini.
All buttons should be simple buttons and you need to update the pin and pull_down.

Afterwards please restart the service with
```
sudo systemctl start phoniebox-gpio-control.service 
sudo systemctl enable phoniebox-gpio-control.service
```

The display won't work with classic edition at the moment. We need to implement a display (probably derived from https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components/displays/HD44780-i2c).

I've ordered a Private Audio HAT yesterday and want to make it run with classic edition, but that will probably take a couple of weeks (depending how much my normal life interferes ;)).

So bottom line:
- Quick: use Spotify edition
- Buttons only: adapt config
- classic edition: implement yourself or wait a couple of weeks

;)

clemsonfire, 2020-11-03:
Good evening!
Sorry for my late response, had some other work to do :-(

Thank you for your massive effort to get Pimoroni running with phoniebox!

I've now reinstalled Phoniebox. Now I've chosen the spotify version.
After successful install I did the pimoroni audio installation without any errors. After this I did the display installation.
Here I had to install PIP which seemed to be missing.

After this command 
`sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                   `
I get this strange pattern. What does this mean? How shall I use this information? Am I right assuming that my Pimoroni Audio HAT display isn'recognized by the system?

best regards, clemsonfire

clemsonfire, 2020-11-19:
Hi!
Thank you for your incredible help and that you've even bought a Pimoroni Audio HAT! 

What I've done meanwhile and what's now working and what's beeing not working in a chaotic listing:

- Display of Pimoroni works - the display shows the name of the played file but no cover art. A file called cover.jpg ist situated in the same dir as the mp3 file. At githib there is a *.shh fscript for cover art; I've no idea where this should be placed.

- The buttons of Pimoroni work.

- In /var/log/syylog there are masses of lines... coming up every second. Not so good for the sd card, isn'it? shortend output:

` Nov 16 17:29:27 phoniebox mopidy[1211]: INFO     [MpdSession-435] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39626
Nov 16 17:29:27 phoniebox mopidy[1211]: INFO     [MpdSession-436] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39630
Nov 16 17:29:27 phoniebox mopidy[1211]: INFO     [MpdSession-437] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39634
Nov 16 17:29:28 phoniebox mopidy[1211]: INFO     [MpdSession-438] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39638
Nov 16 17:29:28 phoniebox mopidy[1211]: INFO     [MpdSession-439] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39642
Nov 16 17:29:28 phoniebox mopidy[1211]: INFO     [MpdSession-440] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39646
Nov 16 17:29:28 phoniebox mopidy[1211]: INFO     [MpdSession-441] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39652
Nov 16 17:29:28 phoniebox mopidy[1211]: INFO     [MpdSession-442] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39654
Nov 16 17:29:29 phoniebox mopidy[1211]: INFO     [MpdSession-443] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39658
Nov 16 17:29:29 phoniebox mopidy[1211]: INFO     [MpdSession-444] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39662
Nov 16 17:29:29 phoniebox mopidy[1211]: INFO     [MpdSession-445] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39666
Nov 16 17:29:29 phoniebox mopidy[1211]: INFO     [MpdSession-446] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39670
Nov 16 17:29:29 phoniebox mopidy[1211]: INFO     [MpdSession-447] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39674
Nov 16 17:29:29 phoniebox mopidy[1211]: INFO     [MpdSession-448] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39678
Nov 16 17:29:33 phoniebox systemd[1]: getty@tty1.service: Succeeded.
Nov 16 17:29:33 phoniebox systemd[1]: getty@tty1.service: Service has no hold-off time (RestartSec=0), scheduling restart.
Nov 16 17:29:33 phoniebox systemd[1]: getty@tty1.service: Scheduled restart job, restart counter is at 2.
Nov 16 17:29:33 phoniebox systemd[1]: Stopped Getty on tty1.
Nov 16 17:29:33 phoniebox systemd[1]: Started Getty on tty1.
Nov 16 17:29:34 phoniebox daemon_rfid_reader.py[1193]: 2020-11-16 17:29:34,285 - root - INFO - Trigger Play Cardid=0006642271
Nov 16 17:29:34 phoniebox mopidy[1211]: INFO     [MpdSession-449] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39682
Nov 16 17:29:35 phoniebox mopidy[1211]: INFO     [MpdSession-450] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39686
Nov 16 17:29:35 phoniebox mopidy[1211]: INFO     [MpdSession-451] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39690
Nov 16 17:29:35 phoniebox mopidy[1211]: INFO     [MpdSession-452] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39694
Nov 16 17:29:35 phoniebox mopidy[1211]: INFO     [MpdSession-453] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39698
Nov 16 17:29:35 phoniebox mopidy[1211]: INFO     [MpdSession-454] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39702
Nov 16 17:29:36 phoniebox mopidy[1211]: INFO     [MpdSession-455] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39706
Nov 16 17:29:36 phoniebox daemon_rfid_reader.py[1193]: volume: 90%   repeat: off   random: off   single: off   consume: off
Nov 16 17:29:36 phoniebox mopidy[1211]: INFO     [MpdSession-456] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39710
Nov 16 17:29:36 phoniebox mopidy[1211]: INFO     [MpdSession-457] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39714
Nov 16 17:29:36 phoniebox mopidy[1211]: INFO     [MpdSession-458] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39718
Nov 16 17:29:37 phoniebox mopidy[1211]: INFO     [MpdSession-459] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39722
Nov 16 17:29:37 phoniebox mopidy[1211]: INFO     [MpdSession-460] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39726
Nov 16 17:29:37 phoniebox mopidy[1211]: INFO     [MpdSession-461] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39730
Nov 16 17:29:37 phoniebox mopidy[1211]: INFO     [MpdSession-462] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39734
Nov 16 17:29:37 phoniebox mopidy[1211]: INFO     [MpdSession-463] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39738
Nov 16 17:29:37 phoniebox mopidy[1211]: INFO     [MpdSession-464] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39742
Nov 16 17:29:37 phoniebox mopidy[1211]: INFO     [MpdSession-465] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39746
[... cut ...]
Nov 19 23:54:37 phoniebox systemd[26104]: mpd.service: Failed with result 'exit-code'.
Nov 19 23:54:37 phoniebox systemd[26104]: Failed to start Music Player Daemon.
Nov 19 23:54:37 phoniebox systemd[26104]: Reached target Default.
Nov 19 23:54:37 phoniebox systemd[26104]: Startup finished in 9.015s.
```

Sorry for this chaotic listing of my troubles. Please tell me which informtion could be usefull for your development of the cool phoniebox system!
Kind regards from locked down Austria :-(

Clemsonfire

s-martin, 2020-11-21:
Do you intend to use Spotify edition or classic?

Mopidy requires the music directory in
`/etc/mopidy/mopidy.conf`
`${HOME_DIR}/.config/mopidy/mopidy.conf`

Phoniebox configures the music path in both, not sure, if that’s really necessary.

clemsonfire, 2020-11-22:
Good morning,

I'd prefer the classic edition due to quicker starting and lacking spotify account.
Now I've installed the spotify edition because you've told me to do so, if I remember correctly.

Kind regards, clemsonfire

s-martin, 2020-11-22:
Yeah, at the moment it works only with the Spotify edition.
I started working on a PR #1112 to implement this for classic edition, but right now normal life gets in the way very often ;)