MiczFlor/RPi-Jukebox-RFID #1110: 🐛 | 3 minute start-up time

Toqqi, 2020-10-22:
## Bug

phoniebox-startup-scripts.service takes around 3 minutes to start up, then runs fine. I'm looking for help to resolve whatever error is causing the delay.

### What I did

I installed the raspberry pi with above mentioned buster image and ran the installer script.

### What happened

I takes a long time until the startup sound is played (3 minutes).
From `systemd-analyze blame` I instantly noticed `2min 57.826s phoniebox-startup-scripts.service`.

Then `journalctl -u phoniebox-startup-scripts.service -b` revealed
```-- Logs begin at Thu 2020-10-22 11:26:40 BST, end at Thu 2020-10-22 11:39:02 BST. --
Oct 22 11:26:52 raspberrypi systemd[1]: Starting Phoniebox Startup...
Oct 22 11:26:52 raspberrypi bash[315]: AUDIOFOLDERSPATH="/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
Oct 22 11:26:52 raspberrypi bash[315]: PLAYLISTSFOLDERPATH="/home/pi/RPi-Jukebox-RFID/playlists"
Oct 22 11:26:52 raspberrypi bash[315]: SECONDSWIPE="PLAY"
Oct 22 11:26:52 raspberrypi bash[315]: SECONDSWIPEPAUSE="2"
Oct 22 11:26:52 raspberrypi bash[315]: SECONDSWIPEPAUSECONTROLS="ON"
Oct 22 11:26:52 raspberrypi bash[315]: AUDIOIFACENAME="Headphone"
Oct 22 11:26:52 raspberrypi bash[315]: VOLUMEMANAGER="mpd"
Oct 22 11:26:52 raspberrypi bash[315]: AUDIOVOLCHANGESTEP="3"
Oct 22 11:26:52 raspberrypi bash[315]: AUDIOVOLMAXLIMIT="100"
Oct 22 11:26:52 raspberrypi bash[315]: AUDIOVOLMINLIMIT="1"
Oct 22 11:26:52 raspberrypi bash[315]: AUDIOVOLSTARTUP="5"
Oct 22 11:26:52 raspberrypi bash[315]: VOLCHANGEIDLE="TRUE"
Oct 22 11:26:52 raspberrypi bash[315]: IDLETIMESHUTDOWN="0"
Oct 22 11:26:52 raspberrypi bash[315]: SHOWCOVER="ON"
Oct 22 11:26:52 raspberrypi bash[315]: READWLANIPYN="OFF"
Oct 22 11:26:52 raspberrypi bash[315]: EDITION="plusSpotify"
Oct 22 11:26:52 raspberrypi bash[315]: LANG="en-UK"
Oct 22 11:26:52 raspberrypi bash[315]: VERSION="2.1.1 - 70f989b - master"
Oct 22 11:26:52 raspberrypi bash[315]: CMDVOLUP=""
Oct 22 11:26:52 raspberrypi bash[315]: CMDVOLDOWN=""
Oct 22 11:26:52 raspberrypi bash[315]: CMDNEXT=""
Oct 22 11:26:52 raspberrypi bash[315]: CMDPREV=""
Oct 22 11:26:52 raspberrypi bash[315]: CMDREWIND=""
Oct 22 11:26:52 raspberrypi bash[315]: CMDSEEKFORW=""
Oct 22 11:26:52 raspberrypi bash[315]: CMDSEEKBACK=""
Oct 22 11:26:53 raspberrypi bash[315]: 5
Oct 22 11:26:53 raspberrypi sudo[336]:       pi : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID ; USER=root ; COMMAND=/
Oct 22 11:26:53 raspberrypi sudo[336]: pam_unix(sudo:session): session opened for user root by (uid=0)
Oct 22 11:26:53 raspberrypi sudo[336]: pam_unix(sudo:session): session closed for user root
Oct 22 11:26:54 raspberrypi sudo[357]:       pi : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID ; USER=root ; COMMAND=/
Oct 22 11:26:54 raspberrypi sudo[357]: pam_unix(sudo:session): session opened for user root by (uid=0)
Oct 22 11:26:54 raspberrypi sudo[357]: pam_unix(sudo:session): session closed for user root
Oct 22 11:26:54 raspberrypi sudo[374]:       pi : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID ; USER=root ; COMMAND=/
Oct 22 11:26:55 raspberrypi sudo[374]: pam_unix(sudo:session): session opened for user root by (uid=0)
Oct 22 11:26:55 raspberrypi sudo[374]: pam_unix(sudo:session): session closed for user root
Oct 22 11:29:58 raspberrypi bash[315]: OK MPD 0.19.0
Oct 22 11:29:58 raspberrypi bash[315]: OK
Oct 22 11:29:58 raspberrypi bash[315]: OK
Oct 22 11:29:58 raspberrypi bash[315]: High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
Oct 22 11:29:58 raspberrypi bash[315]:         version 1.25.10; written and copyright by Michael Hipp and others
Oct 22 11:29:58 raspberrypi bash[315]:         free software (LGPL) without any warranty but with best wishes
Oct 22 11:29:59 raspberrypi bash[315]: Directory: /home/pi/RPi-Jukebox-RFID/shared/
Oct 22 11:29:59 raspberrypi bash[315]: Playing MPEG stream 1 of 1: startupsound.mp3 ...
Oct 22 11:29:59 raspberrypi bash[315]: MPEG 1.0 L III cbr64 44100 j-s
Oct 22 11:29:59 raspberrypi bash[315]: Artist:  Phoniebox
Oct 22 11:30:03 raspberrypi bash[315]: [0:04] Decoding of startupsound.mp3 finished.
Oct 22 11:30:03 raspberrypi bash[315]: volume:  5%   repeat: off   random: off   single: off   consume: off
Oct 22 11:30:03 raspberrypi systemd[1]: Started Phoniebox Startup.
```

### I expected this to happen

From other posts, I am seeing that `phoniebox-startup-scripts.service` only takes around 4 seconds to start.

[... 508 lines cut ...]
```

### Branch / Release

master

### Installscript

scripts/installscripts/buster-install-default.sh

## Hardware

### RaspberryPi version

Raspberry Pi Zero W Rev 1.1

### RFID Reader

SBC-RFID-RC522

### Soundcard

HiFiBerry Miniamp

### Other notable hardware

Huseriato, 2020-10-22:
Because I had issues with my sd card: Can you test it with a faster card? Are tracks played instantly?

Toqqi, 2020-10-22:
I'm using a Transcend 300S 16 GB (Class 10 + UHS-I + UHS-Class 1, Read 95 MB/s, Write 10 MB/s).
I only have slower cards around.
Tracks are played instantly without issues.

Toqqi, 2020-10-23:
I'm trying to figure out where `mpd` is hanging now, since I commented out the wait for the mpd startup in 
`sudo systemd-analyze blame` now gives me
<pre>
         14.029s lighttpd.service
         12.876s smbd.service
          <b>8.528s phoniebox-startup-scripts.service</b>
          6.324s nmbd.service
          6.298s dev-mmcblk0p2.device
          5.569s user@1000.service
          4.782s dhcpcd.service
          2.372s ssh.service
          2.012s networking.service
          1.981s systemd-udev-trigger.service
          1.909s phoniebox-startup-sound.service
          1.796s systemd-timesyncd.service
          1.687s rng-tools.service
          1.568s rpi-eeprom-update.service
          1.550s avahi-daemon.service
          1.501s rsyslog.service
          1.455s systemd-update-utmp-runlevel.service
          1.419s systemd-journald.service
          1.401s atd.service
          1.373s systemd-fsck@dev-disk-by\x2dpartuuid-d6a683cc\x2d01.service
          1.371s mopidy.service
          1.345s systemd-logind.service
          1.264s user-runtime-dir@1000.service
          1.127s systemd-rfkill.service
          1.095s systemd-fsck-root.service
          1.062s systemd-user-sessions.service
           770ms wpa_supplicant.service
           767ms systemd-update-utmp.service
           709ms resolvconf.service
           672ms systemd-tmpfiles-setup.service
           607ms systemd-sysctl.service
           594ms systemd-remount-fs.service
           563ms fake-hwclock.service
           537ms systemd-modules-load.service
           534ms kmod-static-nodes.service
           420ms dev-mqueue.mount
           419ms alsa-restore.service
           391ms systemd-journal-flush.service
           367ms systemd-sysusers.service
           331ms sys-kernel-debug.mount
           323ms systemd-random-seed.service
           320ms run-rpc_pipefs.mount
           280ms sys-kernel-config.mount
           277ms systemd-udevd.service
           264ms console-setup.service
           224ms nfs-config.service
           178ms boot.mount
           153ms systemd-tmpfiles-setup-dev.service
            83ms ifupdown-pre.service
</pre>
So I can basically exclude an issue with the phoniebox startup script.

I have also removed all Spotify folders and playlists to no effect.

I'm wary about two stretches from the syslog:
<pre>
Oct 23 10:22:52 raspberrypi mpd: exception: Failed to create pid file "/run/mpd/pid": Permission denied
Oct 23 10:22:52 raspberrypi systemd[710]: mpd.service: Main process exited, code=exited, status=1/FAILURE
Oct 23 10:22:53 raspberrypi systemd[710]: mpd.service: Failed with result 'exit-code'.
Oct 23 10:22:53 raspberrypi systemd[710]: Failed to start Music Player Daemon.
Oct 23 10:22:53 raspberrypi systemd[710]: Reached target Default.
Oct 23 10:22:53 raspberrypi systemd[710]: Startup finished in 35.338s.
</pre>
and
<pre>
Oct 23 10:22:53 raspberrypi mopidy[538]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Oct 23 10:22:53 raspberrypi mopidy[538]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
[... 3 lines cut ...]
Oct 23 10:22:53 raspberrypi mopidy[538]: INFO     [MainThread] mopidy.config Loading config from command line options
Oct 23 10:22:53 raspberrypi mopidy[538]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Oct 23 10:23:00 raspberrypi mopidy[538]: INFO     [MainThread] mopidy.__main__ Enabled extensions: iris, file, local, http, m3u, stream, spotify, softwaremixer, mpd
Oct 23 10:23:00 raspberrypi mopidy[538]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Oct 23 10:23:06 raspberrypi mopidy[538]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Oct 23 10:23:06 raspberrypi mopidy[538]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Oct 23 10:23:06 raspberrypi mopidy[538]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Oct 23 10:23:06 raspberrypi mopidy[538]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBackend, LocalBackend
Oct 23 10:23:07 raspberrypi systemd[1]: phoniebox-idle-watchdog.service: Succeeded.
Oct 23 10:23:07 raspberrypi mopidy[538]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Oct 23 10:23:13 raspberrypi mopidy[538]: INFO     [SpotifyEventLoop] mopidy_spotify.backend Logged in to Spotify in online mode
<b>Oct 23 10:23:14 raspberrypi mopidy[538]: INFO     [SpotifyBackend-6] mopidy_spotify.web Logged into Spotify Web API as *commented out*
Oct 23 10:24:45 raspberrypi mopidy[538]: INFO     [SpotifyBackend-6] mopidy_spotify.playlists Refreshed 26 Spotify playlists</b>
Oct 23 10:24:45 raspberrypi mopidy[538]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Oct 23 10:24:46 raspberrypi mopidy[538]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: IrisFrontend, HttpFrontend, MpdFrontend
Oct 23 10:24:46 raspberrypi mopidy[538]: INFO     [IrisFrontend-11] mopidy_iris.core Starting Iris 3.52.4
Oct 23 10:24:46 raspberrypi mopidy[538]: INFO     [HttpFrontend-13] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Oct 23 10:24:46 raspberrypi mopidy[538]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Oct 23 10:24:46 raspberrypi mopidy[538]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
Oct 23 10:24:46 raspberrypi mopidy[538]: INFO     [MpdSession-15] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:47524
Oct 23 10:24:48 raspberrypi mopidy[538]: INFO     [MpdSession-16] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:47528
Oct 23 10:24:48 raspberrypi bash[326]: OK MPD 0.19.0
Oct 23 10:24:48 raspberrypi bash[326]: OK
Oct 23 10:24:48 raspberrypi bash[326]: OK
</pre>

gstrauss, 2020-10-25:
lighttpd should start up very quickly, even on embedded devices, though openssl may pause during initialization while waiting for a newly booted system to generate sufficient entropy.

I'd recommend making sure you are running the latest lighttpd release from Debian buster (lighttpd 1.4.53 + patches), and, if you can, the latest lighttpd from buster-backports (lighttpd 1.4.55)

s-martin, 2020-10-25:
Another idea:
Have you mounted any network drives?
I remember some users had similar issues, because they had mounted a network drive.

Toqqi, 2020-10-26:
I've been trying to cut down my start up time a bit and right now I'm at roughly 90 seconds. The most time is used to start up Mopidy now.
The biggest impact was disabling the mopidy-spotify playlist refresh on its startup, since I don't need those in my Phoniebox.
@s-martin I have no mounted network drives.
@gstrauss I tried updating lighttpd from buster-backports, but I got a segmentation fault error and went back to 1.4.53.

I wasn't successful in trying to cut down the used mopidy extensions. I just want to play Spotify tracks/albums/playlists with the Phoniebox, but disabling any of the extensions made the box unusable.

Toqqi, 2020-10-26:
Overclocking the RPi Zero for the initial 60 seconds, I'm now at `Startup finished in 4.027s (kernel) + 58.961s (userspace) = 1min 2.988s.`
I'll try to keep working on this nevertheless.

olli83, 2020-10-26:
I am also looking into getting my Phoniebox to start up faster. Only at the beginning of the process and with two very small kids I am a bit slow getting into it (new to playing with a Raspberry...)

@Toqqi 
> The biggest impact was disabling the mopidy-spotify playlist refresh on its startup, since I don't need those in my Phoniebox.

How is this done? Taking quite some time on mine, too...

Is it still possible to refresh playlists manually when needed? I would like to use a playlists where I can add songs that my kids like just from my Spotify app, but I would be ok to manually refresh the list on the Phoniebox if that gets it to start faster.

Toqqi, 2020-10-26:
@olli83 I will be able to share the small changes I made in the mopidy-spotify tomorrow.

Regarding the refreshing, I noticed that mopidy-spotify loads all, even just favourited playlists from my Spotify accounts on every startup (e.g. "Refreshed 26 playlists from Spotify").
The lists that are associated to RFID tags are refreshed independently from that through mpc, so I did not notice any negative impact on my way of using the Phoniebox.

Toqqi, 2020-10-27:
@olli83 I tested this again and actually you just have to set `allow_playlists = false` in the `[spotify]` configuration group of `/etc/mopidy/mopidy.conf`. There is no need to modify `mopidy-spotify` for this.

MiczFlor, 2020-10-27:
HI @Toqqi @olli83 
could you write your findings into the wiki somewhere? Things get lost in the fire when they are in threads only :)

Toqqi, 2020-10-27:
Hi @MiczFlor, I added the bit about skipping the initial Spotify playlist refresh by `mopidy-spotify` to the Troubleshooting / FAQ section of the wiki. https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#spotify-causes-slow-startup-time

r00tr4v3n, 2020-10-29:
Where did you comment out a wait for mpd? I am using the classic version without Spotify and my Raspberry boots 1min+
```
1min 21.585s phoniebox-startup-scripts.service
         10.248s dhcpcd.service
          2.605s lighttpd.service
          2.541s dev-mmcblk0p2.device
          1.767s nmbd.service
          1.731s smbd.service
           865ms phpsessionclean.service
           820ms user@1000.service
           736ms systemd-fsck@dev-disk-by\x2dpartuuid-ad1a268b\x2d01.service
           735ms avahi-daemon.service
           701ms systemd-udev-trigger.service
           661ms rsyslog.service
           640ms systemd-logind.service
           612ms dphys-swapfile.service
           569ms systemd-timesyncd.service
           556ms networking.service
           550ms ssh.service
           475ms wpa_supplicant.service
           432ms systemd-fsck-root.service
           381ms systemd-journald.service
           379ms rpi-eeprom-update.service
           363ms rng-tools.service
           281ms systemd-remount-fs.service
           267ms systemd-user-sessions.service
           243ms alsa-restore.service
           229ms systemd-udevd.service
           208ms atd.service
           201ms systemd-tmpfiles-setup.service
           185ms systemd-journal-flush.service
           169ms systemd-update-utmp.service
           167ms resolvconf.service
           166ms rc-local.service
           163ms fake-hwclock.service
           141ms run-rpc_pipefs.mount
           133ms systemd-sysctl.service
           131ms sys-kernel-debug.mount
           131ms systemd-update-utmp-runlevel.service
           126ms systemd-modules-load.service
           125ms user-runtime-dir@1000.service
           123ms systemd-sysusers.service
           118ms dev-mqueue.mount
           115ms kmod-static-nodes.service
           113ms systemd-tmpfiles-setup-dev.service
            99ms boot.mount
            97ms systemd-random-seed.service
            85ms systemd-rfkill.service
            79ms sys-kernel-config.mount
            68ms console-setup.service
            58ms mopidy.service
            43ms nfs-config.service
            37ms ifupdown-pre.service
```

Huseriato, 2020-10-30:
What is the command to list this startup time for the services? I found `journalctl -b` for listing. But this des not show the services. My total startup time is 12 seconds now. Assigning a fixed ip might speed this up a little more I think. I did not try it, because I don't want to risk loosing the connection to my installation now. ;)

I think the MDP wait is commented out in  `scripts/startup-script.sg`. Here the lines:

```bash
#########################################
# wait until mopidy/MPD server is running
STATUS=0
while [ "$STATUS" != "ACTIVE" ]; do STATUS=$(echo -e status\\nclose | nc -w 1 localhost 6600 | grep 'OK MPD'| sed 's/^.*$/ACTIVE/'); done
```

I commented this out, too. It does not have any negative effect for my setup. I also removed the start sound. This is 4 seconds long and seems so, that you have to wait for it until it's fully played. There are 2 seconds silent at the beginning of the start sound.

Toqqi, 2020-10-30:
The command to analyze the startup times is `sudo systemd-analyze blame`

Nice, I'll try to comment out the mpd startup and start sound also to check the impact on the times.

Huseriato, 2020-10-30:
Hmm, interesting that the started services are so different for our installations:

```
          2.539s apt-daily.service
          2.182s phoniebox-startup-scripts.service
          1.970s man-db.service
          1.106s apt-daily-upgrade.service
           942ms dev-mmcblk0p2.device
           919ms systemd-fsck@dev-sda1.service
           880ms rpi-eeprom-update.service
           688ms logrotate.service
           591ms lighttpd.service
           525ms dphys-swapfile.service
           453ms user@1000.service
           399ms raspi-config.service
           366ms systemd-logind.service
           363ms rsyslog.service
           359ms avahi-daemon.service
           353ms systemd-fsck@dev-disk-by\x2dpartuuid-738a4d67\x2d01.service
           346ms wpa_supplicant.service
           329ms triggerhappy.service
           323ms dhcpcd.service
           247ms ssh.service
           239ms phpsessionclean.service
           238ms keyboard-setup.service
           213ms networking.service
           207ms systemd-udev-trigger.service
           175ms systemd-timesyncd.service
           174ms systemd-fsck-root.service
           159ms systemd-remount-fs.service
           151ms cleanshutd.service
           150ms systemd-journald.service
            93ms atd.service
            92ms resolvconf.service
            92ms media-usb.mount
            87ms kmod-static-nodes.service
            84ms systemd-udevd.service
            84ms systemd-modules-load.service
            82ms run-rpc_pipefs.mount
            80ms mopidy.service
            80ms rng-tools.service
            76ms sys-kernel-debug.mount
            68ms fake-hwclock.service
            66ms systemd-tmpfiles-setup.service
            63ms systemd-tmpfiles-clean.service
            62ms exim4.service
            59ms dev-mqueue.mount
            55ms systemd-sysctl.service
            54ms alsa-restore.service
            53ms systemd-update-utmp.service
            52ms systemd-journal-flush.service
            44ms boot.mount
            43ms systemd-user-sessions.service
            40ms systemd-sysusers.service
            39ms rc-local.service
            39ms systemd-update-utmp-runlevel.service
            38ms systemd-tmpfiles-setup-dev.service
            34ms systemd-rfkill.service
            34ms systemd-random-seed.service
            32ms user-runtime-dir@1000.service
            30ms console-setup.service
            29ms sys-fs-fuse-connections.mount
            24ms sys-kernel-config.mount
            21ms nfs-config.service
            18ms ifupdown-pre.service
```

I also removed samba (don't know, why there is a service left) and exim4 using apt. I disabled "Wait for Network on Boot" in the raspi-config. That's all I've done so far. I hope this helps.

Toqqi, 2020-10-30:
I'm also on a RPi Zero W, I suppose that performance goes up considerably depending on the Pi that is used.

Huseriato, 2020-10-30:
Yes, that's true. I meant not the starting time itself, it's more about the service names.

@s-martin Do you know a command or preferred log file to analyse the start up of the phoniebox service better? syslog is not a good option, there is so much stuff going on.

Toqqi, 2020-10-30:
> 
> I think the MDP wait is commented out in `scripts/startup-script.sg`. Here the lines:
> 
> ```shell
> #########################################
> # wait until mopidy/MPD server is running
> STATUS=0
> while [ "$STATUS" != "ACTIVE" ]; do STATUS=$(echo -e status\\nclose | nc -w 1 localhost 6600 | grep 'OK MPD'| sed 's/^.*$/ACTIVE/'); done
> ```
> 
> I commented this out, too. It does not have any negative effect for my setup.

I commented out the wait for MPD, then startup finishes considerably faster, but the player still won't react until MPD and mopidy are running completely. So unfortunately I cannot apply this in my use case (only Spotify player).

s-martin, 2020-10-30:
@Huseriato Unfortunately I don't know a good tool. If there's something difficult to find i usually copy the log via WinSCP to my Windows machine and use Notepad++ to find stuff (and bulk remove noise).

Following things for startup time should be considered:

- Spotify edition boots much slower than classic, because Mopidy is used. Some users even have to add a precondition via raspi-config to wait for network access, otherwise Spotify would not connect (see various issues here). This could lead to even worse boot times.
- Pi Zero has only a limited performance, so expect longer boot times there
- I assume various services of RPi could be disabled to improve boot times, but that's something you have to test thoroughly. Removing Desktop stuff like LibreOffice probably makes sense. Please share your experiences in the wiki.
- Users reported in the past that mounted network drives could lead to much slower boot times. 

Hope this helps and I guess I should add that to the wiki, too. ;)

Huseriato, 2020-10-30:
@Toqqi you're right. Commenting out the MDP wait, leads to wait time after boot is finished until the cards are working. I was so focused on boosting this time, I forgot to test a card immediately.

I'm currently thinking about, if I really need Spotify. I think the box is more reliable, when using local media. I've not integrated a power bank, because I wanted to make the children's room more attractive to the kids. But the box is attachable to a powerbank or a car outlet. So the box is theoretically portable, but is considered to be in the children's room.

@s-martin Thanks for your input. So is it true, that the non spotify version plays the tracks using mpg123? Or how does this work here?

s-martin, 2020-10-30:
mpg123 is just used for startup sounds.

The classic edition uses MPD, spotify edition uses Mopidy.

r00tr4v3n, 2020-11-05:
A new installation improved my startup time from 90s to 30s. Weird.

arrisde, 2020-11-08:
I have the same issue (+Spotify edition, RPi0W, freshly installed 2.1.1). Future owner will have to stretch all his patience to wait ~3mins for the startup sound :(
@r00tr4v3n, which RPi are you on, and did you install from development or master branch? +Spotify or Classic edition?

s-martin, 2020-11-21:
@arrisde, unfortunately I think RPi Zero and Spotify edition may lead to long boot times, but if you have some improvements to share many Phoniebox users would be interested.

s-martin, 2020-12-05:
Please check #1174 for some performance improvements especially on Pi Zero

arne123, 2021-01-13:
I am trying to speed up the boot on the rpi zero as well.

The following steps I did with at least some success:

-     Starting Network stuff after audio stuff (I want music in first place network later)
-     upgrading mpd to the new version (0.23 here, the raspbian one is way older)
      In fact I recompile more then one time, after getting some recommendations in the mpd forum 
      There is still potential of further improvements  https://forum.musicpd.org/viewtopic.php?p=12805#p12805
-     mpd disabled auto-update
-     avoid parallel starting of other services in parallel to mpd 
      this is manly for effective for high cpu load tasks on single cores
-     shortened the startup sound (trimmed leading and trailing empty parts)

At the moment I have audio available 30s after the boot.
The bootchart looks looks like this:
https://gist.githubusercontent.com/arne123/5a614d01397680cc2eae12344026db6a/raw/476672d48c7e59979b4e4c2cb557479daaec2943/boot.svg

what I still want to do:

-     searching for more services which are not required and can be disabled or postponed after audio
-     skip systemd-fsck@dev, or execute it during shutdown (1.7s)
-     see if there are some possibilities to shorten the systemd start itself (~3s) 
      especially I see some potential in the generator phase, but I could not find good resource here yet
-     see if the kernel period can be shortened (3s) 
      here I see also some potential
-     see if dev-mmcblk0p2 can be shortened (7s) 
      actuall this would help a lot, but but maybe there is not so much potential here,
      mainly due to  lack of understanding and missing leverage
-     check if the startup script/sound can be modified in some ways 
      e.g. playing sound earlier (in parallel to mpd)(~3s)
-     SD-card: 
      initially I thought I bought a good sd card(80MB/s), but I doubt this somehow now.
      is anybody also using an rpi zero w and can compare the boot times for some services?
      Is there some potential in the sd card?

my goal is to boot til audio in at least 20s.

br
Arne

arne123, 2021-01-14:
So I improved some more seconds, basically be removing further plugins from mpd.
Also I am playing the startup sound in parallel to mpd start.

you can find the details here
https://gist.github.com/arne123/b35126771ffe19d73910c83d182ed0b9

Maybe there is still a little more potential in mpd, but not much. 
The start of mpd with running mpg123 in parallel just takes 1.6s sec. 
I will now focus on the startup script (why it takes 3 sec?) systemd startup and dev-mmcblk0p2 .

arne123, 2021-01-23:
I finally managed to boot to sound in 20 sec

- more services have been disabled or reorganized
- systemd generators which according to my understanding are not of any use at the moment have been disabled
- the sice of the /lib/udev/hwdb.bin has been reduce by removing all pci device
- not used kernel modules have been blacklisted 

https://gist.githubusercontent.com/arne123/656aeef30ef20edd381d8a63953a817f/raw/d1d9ce34c13113260ef06f26c7f4ca688060557b/boot.svg

A more detailed bootchart can be generated with systemd-bootchart
https://gist.githubusercontent.com/arne123/656aeef30ef20edd381d8a63953a817f/raw/d1d9ce34c13113260ef06f26c7f4ca688060557b/bootchart-20210123-1344.svg

Details:

disabled systemd generators
sudo chmod a-x /lib/systemd/system-generators/systemd-rc-local-generator 
sudo chmod a-x /lib/systemd/system-generators/systemd-cryptsetup-generator 
sudo chmod a-x /lib/systemd/system-generators/systemd-debug-generator 
sudo chmod a-x /lib/systemd/system-generators/systemd-bless-boot-generator 
sudo chmod a-x /lib/systemd/system-generators/systemd-hibernate-resume-generator
sudo chmod a-x /lib/systemd/system-generators/systemd-run-generator 
sudo chmod a-x /lib/systemd/system-generators/systemd-system-update-generator 
sudo chmod a-x /lib/systemd/system-generators/systemd-veritysetup-generator 

blacklisted modules:
/etc/modprobe.d/raspi-blacklist.conf
blacklist bcm2835_v4l2
blacklist bcm2835_isp
blacklist bcm2835_codec
blacklist v4l2_mem2mem
blacklist rfkill

arne123, 2021-01-29:
I got another 5s by compiling a custom kernel, sound in 15s now.

https://gist.githubusercontent.com/arne123/3f85530d23d812c0e2d00da67d1be5c7/raw/aaaa354b7749b54663419b76bc759cf14fdecc90/boot.svg

basically I tired to disable things which I don't need (trace_fs, usb_otg) 
and put everything which was load as module build in module (except wifi bcrmfmac, which failed somehow)

If you are interested, here is the kernel config as well:
https://gist.githubusercontent.com/arne123/3f85530d23d812c0e2d00da67d1be5c7/raw/aaaa354b7749b54663419b76bc759cf14fdecc90/.config

I mainly followed the instructions here for cross compiling:
https://www.raspberrypi.org/documentation/linux/kernel/building.md

splitti, 2021-03-01:
Hi,

mostly the startup time is caused because of mopidy refreshing the spotify playlists. I wrote some speed hacks here: [https://splittscheid.de/phoniebox-schneller-booten-fehler-beheben/](https://splittscheid.de/phoniebox-schneller-booten-fehler-beheben/)

Another option is to try a OS like dietPi. It boots up pretty fast. But i stay at Raspbian and i am searching for more Speed Hacks. The custom kernel sounds good @arne123, i will try that!

melo0187, 2021-03-15:
@arne123 can you provide some guidance on how to reproduce your steps?
What did you do to start the network stuff after all the audio things?
And how can you avoid starting other services in parallel to mpd?

I'm using a RPi Zero W with the RC522 RFID reader and the HifiBerry miniamp and it takes 60s until I hear the startup sound and can start using the box.
I already did follow the instructions on @splitti 's blog (except for setting a static ip, disabling ipv6, sdtweak/overclock and initial_turbo) and removed`mpc rescan` from startup-scripts.sh so I have no idea left what I could do to get even close to the 20s startup time you achieved.

Any help is greatly appreciated!

For the record my `systemd-analyze blame` output:
```
         26.001s mpd.service
         12.665s nmbd.service
         12.052s lighttpd.service
          8.179s dev-mmcblk0p2.device
          7.549s smbd.service
          7.188s phoniebox-startup-scripts.service
          3.783s systemd-timesyncd.service
          3.761s ssh.service
          3.724s networking.service
          3.198s systemd-udev-trigger.service
          2.592s systemd-logind.service
          2.479s cleanshutd.service
          2.136s systemd-rfkill.service
          2.022s avahi-daemon.service
          1.976s systemd-journald.service
          1.741s rpi-eeprom-update.service
          1.719s systemd-user-sessions.service
          1.533s wpa_supplicant.service
          1.431s alsa-restore.service
          1.348s systemd-remount-fs.service
          1.260s user@1000.service
          1.231s atd.service
          1.136s rsyslog.service
          1.050s systemd-update-utmp.service
          1.004s systemd-tmpfiles-setup.service
           808ms dhcpcd.service
           771ms systemd-journal-flush.service
           758ms systemd-sysctl.service
           719ms systemd-fsck-root.service
           656ms systemd-sysusers.service
           625ms fake-hwclock.service
           601ms dev-mqueue.mount
           550ms rc-local.service
           541ms systemd-random-seed.service
           541ms sys-kernel-debug.mount
           539ms systemd-tmpfiles-clean.service
           538ms resolvconf.service
           508ms systemd-modules-load.service
           495ms rng-tools.service
           485ms kmod-static-nodes.service
           436ms user-runtime-dir@1000.service
           386ms systemd-fsck@dev-disk-by\x2dpartuuid-8e16ee7a\x2d01.service
           376ms sys-kernel-config.mount
           368ms systemd-udevd.service
           360ms run-rpc_pipefs.mount
           280ms systemd-tmpfiles-setup-dev.service
           276ms console-setup.service
           248ms systemd-update-utmp-runlevel.service
           218ms nfs-config.service
           132ms boot.mount
           108ms ifupdown-pre.service
```

arne123, 2021-03-16:
I think you will have the biggest benefit recompiling mpd.
If you wish, I can provide you the binary I have compiled on my rpi zero.

Then you have to deal with systemd and the systemd configuration files. 
I made reorganized many services after sound.target and after network.target
unfortunately I don't know all service files I changed by hard, I need to look them up later.

Probably you don't need avahi-daemon.service, there are also some other services which I don't see on my setup.

melo0187, 2021-03-16:
Hey @arne123 thank you for your reply!
I would be happy to give your mpd binary a try. Am I assuming correctly, that I could just copy it over the original bin in `/usr/bin`?

Although I know my way around Linux I never found myself dealing with that aspect of the OS before, so I don't know much about systemd and how to tweak it. I just learned about `systemd-analyze blame` in this issue and haven't installed bootchart yet.
I am assuming that following @splitti​'s [FAQ](https://splittscheid.de/phoniebox-schneller-booten-fehler-beheben/#ftoc-mpd-service-eher-laden) for optimizing boot time changing `After=network.target sound.target` to read `After=sound.target` would be enough to start mpd as soon as audio is ready without waiting on network. Is this correct?
What could I do more in that regard?

Thanks again for your input, it is much appreciated.
My son is just hitting 17 months and so I figure reducing the boot time from 60 down to 40 if not 20 seconds would already be very helpful =)

arne123, 2021-03-16:
**systemd-analyze plot > boot.svg** will give you an svg where you can get an impression of the dependencies during boot. 
(you don't need bootchart unless you are intending to dive in very deep)

Try this mpd: [mpd.gz](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/6152347/mpd.gz),
hopefully it runs on your device and there are no further dependencies.
I would recommend to keep your original mpd, and put this somewhere else, you can modify the execution command in your systemd unit file:  (/lib/systemd/system/mpd.service)

[Service]
Type=notify
EnvironmentFile=/etc/default/mpd
**#ExecStart=/usr/bin/mpd --no-daemon $MPDCONF**
**ExecStart=/home/pi/RPi-Jukebox-RFID/mpd/mpd --no-daemon /etc/mpd.conf**

do you use samba to access your box?
If not you can disable it.

melo0187, 2021-03-16:
@arne123 I don't know how to thank you :joy: 

First of all good catch with the samba service. Since I use the webinterface to upload audio and scp to copy files like the mpd binary you just provided, I won't miss samba.

But most of all thank you for compiling and sharing your mpd binary with me! I have no idea what you did there or how, but it simply works and you just reduced my boot time from ~60 to ~35 seconds :exploding_head:

Thank you so much!

P.S.:
```
pi@phoniebox:~ $ sudo systemd-analyze blame
          8.824s dev-mmcblk0p2.device
          7.558s phoniebox-startup-scripts.service
          7.197s lighttpd.service
          6.309s dhcpcd.service
          4.851s nmbd.service
          3.754s networking.service
          3.024s systemd-udev-trigger.service
          2.710s systemd-rfkill.service
          2.580s mpd.service
          2.418s systemd-logind.service
          2.125s systemd-journald.service
          2.002s user@1000.service
          1.955s ssh.service
          1.816s wpa_supplicant.service
          1.698s systemd-fsck@dev-disk-by\x2dpartuuid-fdb3324f\x2d01.service
          1.686s systemd-fsck-root.service
          1.221s atd.service
          1.181s cleanshutd.service
          1.158s systemd-tmpfiles-setup.service
          1.141s rng-tools.service
          1.105s systemd-user-sessions.service
          1.063s resolvconf.service
           929ms systemd-remount-fs.service
           878ms systemd-sysctl.service
           755ms sys-kernel-debug.mount
           743ms systemd-update-utmp.service
           741ms run-rpc_pipefs.mount
           717ms systemd-modules-load.service
           658ms alsa-restore.service
           646ms fake-hwclock.service
           619ms kmod-static-nodes.service
           539ms systemd-journal-flush.service
           497ms systemd-sysusers.service
           471ms sys-kernel-config.mount
           465ms systemd-random-seed.service
           444ms user-runtime-dir@1000.service
           430ms systemd-update-utmp-runlevel.service
           411ms dev-mqueue.mount
           374ms systemd-udevd.service
           320ms nfs-config.service
           310ms console-setup.service
           304ms ifupdown-pre.service
           213ms systemd-tmpfiles-setup-dev.service
           143ms boot.mount
```

MiczFlor, 2021-03-17:
Hi @arne123 
I will give the recompiled `mpd` a try, too. 
Regarding your C++ work, I put together the ideas from the `discussions about the future of phoniebox` and would like to get your input regarding the little application you wrote in C++.
You can find my question and summary here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/discussions/1329#discussioncomment-479170
Thanks and all the best, micz

Lueffe, 2021-03-17:
@arne123 
Thx for the binary. It works fine. 
Can you post your meson Build Flags?

arne123, 2021-03-17:
Sure, I thought I already did, here is the latest:
https://gist.github.com/arne123/b14f9d67ddf3da6fbc91ed1320668f79

To all using that binary, please be aware that the mpd is build on the rpi zero w, with the instruction set armv6l of that specific processor . I believe that newer arm processors are mostly backwards compatible, but you might run into some issues running it on different environments.

Here is the discussion I had in the mpd forum:
https://forum.musicpd.org/viewtopic.php?f=7&t=8469

The conclusion is, that not mpd itself is causing the long start time, but the load of several shared libraries which are mostly not used, so I removed as much as possible of them.

Toqqi, 2021-03-19:
@arne123
Hi, I suppose your recompiled `mpd` only works for the non-Spotify version of Phoniebox?
I tried to copy over your executable and I think that mopidy uses its own version of mpd (mopidy-mpd extension). Can anyone confirm this?

I will do some research in order to apply your meson build flags to mopidy-mpd.
There still is a `OK MPD 0.19.0` in `journalctl -u phoniebox-startup-scripts.service -b` on my RPi.

Thanks!

passuff, 2022-03-02:
I have a fresh develop install with spotify. I was wondering about the Startup speed too as I did not yet add anything to the system:

[image]

So for sure I can disable some services but the first one is most interesting:
`pi@raspberrypi:~ $ journalctl -u phoniebox-startup-scripts.service -b
-- Logs begin at Wed 2022-03-02 16:26:37 GMT, end at Wed 2022-03-02 16:46:05 GMT
Mar 02 16:26:42 raspberrypi systemd[1]: Starting Phoniebox Startup...
Mar 02 16:26:42 raspberrypi bash[420]: Phoniebox is starting...
Mar 02 16:26:42 raspberrypi bash[420]: 2.3
Mar 02 16:26:42 raspberrypi bash[420]: AUDIOFOLDERSPATH="/home/pi/RPi-Jukebox-RF
Mar 02 16:26:42 raspberrypi bash[420]: PLAYLISTSFOLDERPATH="/home/pi/RPi-Jukebox
Mar 02 16:26:42 raspberrypi bash[420]: SWIPEORPLACE="SWIPENOTPLACE"
Mar 02 16:26:42 raspberrypi bash[420]: SECONDSWIPE="RESTART"
Mar 02 16:26:42 raspberrypi bash[420]: SECONDSWIPEPAUSE="2"
Mar 02 16:26:42 raspberrypi bash[420]: SECONDSWIPEPAUSECONTROLS="ON"
Mar 02 16:26:42 raspberrypi bash[420]: AUDIOIFACENAME="Master"
Mar 02 16:26:42 raspberrypi bash[420]: AUDIOIFACEACTIVE="0"
Mar 02 16:26:42 raspberrypi bash[420]: VOLUMEMANAGER="mpd"
Mar 02 16:26:42 raspberrypi bash[420]: AUDIOVOLCHANGESTEP="3"
Mar 02 16:26:42 raspberrypi bash[420]: AUDIOVOLMAXLIMIT="100"
Mar 02 16:26:42 raspberrypi bash[420]: AUDIOVOLMINLIMIT="1"
Mar 02 16:26:42 raspberrypi bash[420]: AUDIOVOLSTARTUP="100"
Mar 02 16:26:42 raspberrypi bash[420]: AUDIOVOLBOOT="100"
Mar 02 16:26:42 raspberrypi bash[420]: VOLCHANGEIDLE="TRUE"
Mar 02 16:26:42 raspberrypi bash[420]: IDLETIMESHUTDOWN="0"
Mar 02 16:26:42 raspberrypi bash[420]: POWEROFFCMD="sudo poweroff"
Mar 02 16:26:42 raspberrypi bash[420]: SHOWCOVER="ON"
Mar 02 16:26:42 raspberrypi bash[420]: READWLANIPYN="OFF"
Mar 02 16:26:42 raspberrypi bash[420]: EDITION="plusSpotify"
Mar 02 16:26:42 raspberrypi bash[420]: LANG="en-UK"
Mar 02 16:26:42 raspberrypi bash[420]: VERSION="2.3 - 5084ace1 - develop"
Mar 02 16:26:42 raspberrypi bash[420]: CHAPTEREXTENSIONS="mp4,m4a,m4b,m4r"
Mar 02 16:26:42 raspberrypi bash[420]: CHAPTERMINDURATION="600"
Mar 02 16:26:42 raspberrypi bash[420]: CMDVOLUP=""
Mar 02 16:26:42 raspberrypi bash[420]: CMDVOLDOWN=""
Mar 02 16:26:42 raspberrypi bash[420]: CMDNEXT=""
Mar 02 16:26:42 raspberrypi bash[420]: CMDPREV=""
Mar 02 16:26:42 raspberrypi bash[420]: CMDREWIND=""
Mar 02 16:26:42 raspberrypi bash[420]: CMDSEEKFORW=""
Mar 02 16:26:42 raspberrypi bash[420]: CMDSEEKBACK=""
Mar 02 16:26:42 raspberrypi bash[420]: 100 is the mpd startup volume
Mar 02 16:26:42 raspberrypi sudo[449]:       pi : TTY=unknown ; PWD=/home/pi/RPi
Mar 02 16:26:42 raspberrypi sudo[449]: pam_unix(sudo:session): session opened fo
Mar 02 16:26:42 raspberrypi sudo[449]: pam_unix(sudo:session): session closed fo
Mar 02 16:26:42 raspberrypi sudo[463]:       pi : TTY=unknown ; PWD=/home/pi/RPi
Mar 02 16:26:42 raspberrypi sudo[463]: pam_unix(sudo:session): session opened fo
Mar 02 16:26:42 raspberrypi sudo[463]: pam_unix(sudo:session): session closed fo
Mar 02 16:26:42 raspberrypi sudo[466]:       pi : TTY=unknown ; PWD=/home/pi/RPi
Mar 02 16:26:42 raspberrypi sudo[466]: pam_unix(sudo:session): session opened fo
Mar 02 16:26:42 raspberrypi sudo[466]: pam_unix(sudo:session): session closed fo
Mar 02 16:26:45 raspberrypi bash[420]: /home/pi/RPi-Jukebox-RFID/scripts/startup
Mar 02 16:27:09 raspberrypi bash[420]: OK MPD 0.19.0
Mar 02 16:27:09 raspberrypi bash[420]: OK
Mar 02 16:27:09 raspberrypi bash[420]: OK
Mar 02 16:27:10 raspberrypi bash[420]: OK MPD 0.19.0
Mar 02 16:27:10 raspberrypi bash[420]: OK
Mar 02 16:27:10 raspberrypi bash[420]: OK
Mar 02 16:27:10 raspberrypi bash[420]: 32768 is the mpg123 startup volume
Mar 02 16:27:10 raspberrypi bash[420]: High Performance MPEG 1.0/2.0/2.5 Audio P
Mar 02 16:27:10 raspberrypi bash[420]:         version 1.25.10; written and copy
Mar 02 16:27:10 raspberrypi bash[420]:         free software (LGPL) without any
Mar 02 16:27:10 raspberrypi bash[420]: Directory: /home/pi/RPi-Jukebox-RFID/shar
Mar 02 16:27:10 raspberrypi bash[420]: Playing MPEG stream 1 of 1: startupsound.
Mar 02 16:27:11 raspberrypi bash[420]: MPEG 1.0 L III cbr64 44100 j-s
Mar 02 16:27:11 raspberrypi bash[420]: Artist:  Phoniebox
Mar 02 16:35:12 raspberrypi bash[420]: [0:04] Decoding of startupsound.mp3 finis
Mar 02 16:35:12 raspberrypi bash[420]: volume:100%   repeat: off   random: off
Mar 02 16:35:12 raspberrypi systemd[1]: Started Phoniebox Startup.`

So most time is consumed while executing startup scripts, right?

Any recommendation how to speed this up? I'm using pi zero 2 and was hoping to get 20s start up time...

MiczFlor, 2022-03-05:
Have a look at this. The issue is often `mpd` which takes time to start. And that is inside the startup script. In the following link, there is a version of `mpd` that has only the necessary dependencies:
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1110#issuecomment-800639044

wastlmuc, 2022-08-30:
Just wanted to add 

> Sure, I thought I already did, here is the latest: https://gist.github.com/arne123/b14f9d67ddf3da6fbc91ed1320668f79
>
> To all using that binary, please be aware that the mpd is build on the rpi zero [...]

In case someone wants to rebuild mpd using above approach and following [mpd doc ](https://mpd.readthedocs.io/en/stable/user.html#compiling-from-source), for 0.23.9 you need to add
`option('pipewire', type: 'feature', value: 'disabled', description: 'PipeWire support')
option('snapcast', type: 'boolean', value: false, description: 'Snapcast output plugin')
option('openmpt', type: 'feature', value: 'disabled', description: 'OpenMPT decoder plugin')`
to medon_options.txt, as provided above.

Additionally for web streaming you need to explicitly set
`option('curl', type: 'feature', value: 'enabled', description: 'HTTP client using CURL')
option('mad', type: 'feature', value: 'enabled', description: 'MP3 decoder using libmad')`

mad is necessary for playing the streams, as mpg123 cannot deal with web urls.

Make sure you are building the proper config.
`meson configure output/release `

Hope this helps.

`[meson_options.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/9454800/meson_options.txt)`

AlvinSchiller, 2024-04-12:
The Mopidy refresh problematic is solved in the newest version introduced with #2315

closed 2024-04-14 by AlvinSchiller