MiczFlor/RPi-Jukebox-RFID #1111: 🐛 | I2C & Shutdown collision

daved3luxe, 2020-10-22:
## Bug

I have connected a PN532 to GPIO 2 & 3 via I2C.
If I connect GPIO 3 to the shutdown button and activate it in gpio_settings.ini, the RFID reader no longer works.

Can someone help?

### What I did

I edited the gpio_settings.ini

### What happened

RFID Reader does not work anymore

## Software

### Base image and version

2.1.1 - 70f989b - master
on  buster

### Branch / Release

master

### Installscript

scripts/installscripts/buster-install-default.sh

## Hardware

### RaspberryPi version

3 B+

### RFID Reader

PN532

### Other notable hardware

/home/pi/.config/phoniebox/gpio_settings.ini

s-martin, 2020-10-22:
Do I understand correctly that the reader and the shutdown button both use GPIO 3?
That won’t work.

But you can use the shutdown button on an available GPIO pin, check here for free pins: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Using-GPIO-hardware-buttons#pin-numbering-on-the-rpi

Then you need to update gpio_settings.ini and restart the service `sudo systemctl restart phoniebox-gpio-control`

See here also for used pin configurations (but you can change them):
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Hardware-Pinout-Overview

daved3luxe, 2020-10-22:
Thats correct because I read here: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Hardware-Pinout-Overview
:fixed GPIO if wake up is required can be used in conjunction with I2C devices

Because if I use another PIN for Shutdown I can´t wake the RPi up

daved3luxe, 2020-10-29:
Hi again,

here is my Solution:

https://imgur.com/oFYLv1n
https://imgur.com/iVSS4io

Shutdown-Pin is GPIO13

closed 2020-10-29 by daved3luxe