MiczFlor/RPi-Jukebox-RFID #1114: 🐛 | Wrong configuration path in gpio_control.py

Conendrum, 2020-10-23:
Hi everyone,

First of all, thanks a lot, this is a great project and my kids are already loving it!

I wanted to report a bug I found while enabling GPIO. 

I tried installing push buttons to control Play/Pause and VolumeControl following instructions here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Using-GPIO-hardware-buttons
and here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/master/components/gpio_control

But when pressing the button, the play/pause wasn't working even though correctly configured in: 
`~/RPi-Jukebox-RFID/settings/gpio_settings.ini`

Then I realized that the script "gpio_control.py" is actually looking for the config file somewhere else:
```
pi@raspberrypi:~/RPi-Jukebox-RFID $ cat components/gpio_control/gpio_control.py | grep "gpio_settings.ini"
    config_path = os.path.expanduser('~/.config/phoniebox/gpio_settings.ini')
```

I could fixed the issue by copying the gpio_settings.ini into the .config/phoniebox directory and then gpio controls were working. I looked around and there are indeed two config files (third one being an example":

```
pi@raspberrypi:~/RPi-Jukebox-RFID $ sudo find / -name "gpio_settings.ini"
/home/pi/.config/phoniebox/gpio_settings.ini
/home/pi/RPi-Jukebox-RFID/settings/gpio_settings.ini
/home/pi/RPi-Jukebox-RFID/components/gpio_control/example_configs/gpio_settings.ini
```

We could fix the bug by correcting the path in gpio_control.py (but I don't know if there are other dependencies in the code) or by updating the documentation (cf two links above).

Thanks and cheers
Romain

PS: I'm using the 2.1.1 Master version

s-martin, 2020-10-23:
Hi,
sorry for getting confused by not up-to-date docs.

This is already adressed in #1096 and fixed in develop branch.

We will release 2.2 soon, which includes this fix. In the meantime you could manually add the changes made in develop branch to your local installation.

I close this issue as duplicate.

closed 2020-10-23 by s-martin

Conendrum, 2020-10-23:
Thanks for the info, I had missed #1096  while looking for existing issues about this bug.