MiczFlor/RPi-Jukebox-RFID #1117: Can not load: Problem with Spotify/Modipy after Update to 2.1.1

SirY4, 2020-10-23:
**Can not load songs/playlists: Problem with Modipy after Update to 2.1.1**

After I manually updated to 2.1.1 I can only play local songs with the phoniebox. When I open IRIS Webplayer I get "_Modipy: Failed to add some tracks_" as a pop up notification.
**HOWEVER** there is a reproducable work around: after I do a "_scan database_" (in the phoniebox webinterface/folders and files) it is back to full function. After reboot (soft or hard) it is broken again.

I am greatful for any support. Thank you!

s-martin, 2020-10-23:
Please post the output of `journalctl -u mopidy`.

SirY4, 2020-10-23:
`-- Logs begin at Fri 2020-10-23 13:14:22 BST, end at Fri 2020-10-23 14:10:45 BST. --
Oct 23 13:14:26 raspberrypi systemd[1]: Starting Mopidy music server...
Oct 23 13:14:26 raspberrypi systemd[1]: Started Mopidy music server.
Oct 23 13:14:33 raspberrypi mopidy[460]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Oct 23 13:14:33 raspberrypi mopidy[460]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Oct 23 13:14:33 raspberrypi mopidy[460]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Oct 23 13:14:33 raspberrypi mopidy[460]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Oct 23 13:14:33 raspberrypi mopidy[460]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Oct 23 13:14:33 raspberrypi mopidy[460]: INFO     [MainThread] mopidy.config Loading config from command line options
Oct 23 13:14:33 raspberrypi mopidy[460]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Oct 23 13:14:34 raspberrypi mopidy[460]: INFO     [MainThread] mopidy.__main__ Enabled extensions: m3u, softwaremixer, http, local, iris, spotify, file, mpd, stream
Oct 23 13:14:34 raspberrypi mopidy[460]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Oct 23 13:14:36 raspberrypi mopidy[460]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Oct 23 13:14:36 raspberrypi mopidy[460]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Oct 23 13:14:36 raspberrypi mopidy[460]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Oct 23 13:14:36 raspberrypi mopidy[460]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBackend, LocalBackend
Oct 23 13:14:36 raspberrypi mopidy[460]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Oct 23 13:14:37 raspberrypi mopidy[460]: ERROR    [SpotifyEventLoop] spotify.session Spotify login error: <ErrorType.UNABLE_TO_CONTACT_SERVER: 8>
Oct 23 13:14:40 raspberrypi mopidy[460]: INFO     [SpotifyBackend-6] mopidy_spotify.web Logged into Spotify Web API as 1i9cn0j5xgq68n5nyagby9rwt
Oct 23 13:15:39 raspberrypi mopidy[460]: INFO     [SpotifyBackend-6] mopidy_spotify.playlists Refreshed 77 Spotify playlists
Oct 23 13:15:39 raspberrypi mopidy[460]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Oct 23 13:15:39 raspberrypi mopidy[460]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: IrisFrontend, HttpFrontend, MpdFrontend
Oct 23 13:15:39 raspberrypi mopidy[460]: INFO     [IrisFrontend-11] mopidy_iris.core Starting Iris 3.52.4
Oct 23 13:15:39 raspberrypi mopidy[460]: INFO     [HttpFrontend-13] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Oct 23 13:15:39 raspberrypi mopidy[460]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Oct 23 13:15:39 raspberrypi mopidy[460]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
Oct 23 13:15:39 raspberrypi mopidy[460]: INFO     [MpdSession-15] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51556
Oct 23 13:15:39 raspberrypi mopidy[460]: INFO     [MpdSession-16] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51560
Oct 23 13:15:44 raspberrypi mopidy[460]: INFO     [MpdSession-17] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51568
Oct 23 13:16:39 raspberrypi mopidy[460]: INFO     [MpdSession-18] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51572
Oct 23 13:16:39 raspberrypi mopidy[460]: INFO     [MpdSession-19] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51576
Oct 23 13:17:28 raspberrypi mopidy[460]: INFO     [SpotifyBackend-6] mopidy_spotify.lookup Failed to lookup 'spotify:track:29Ofm06sP9wJTc2BLkztyY': Session must be logged in
Oct 23 13:17:29 raspberrypi mopidy[460]: INFO     [SpotifyBackend-6] mopidy_spotify.lookup Failed to lookup 'spotify:track:3RUytZYw4jWOKGUDLABemc': Session must be logged in
Oct 23 13:17:29 raspberrypi mopidy[460]: INFO     [SpotifyBackend-6] mopidy_spotify.lookup Failed to lookup 'spotify:track:7jY2wjlJ5IfHIlMihHIvRu': Session must be logged in
Oct 23 13:17:29 raspberrypi mopidy[460]: INFO     [SpotifyBackend-6] mopidy_spotify.lookup Failed to lookup 'spotify:track:5vzFnlJE4MSKmV24u2F89j': Session must be logged in
Oct 23 13:17:29 raspberrypi mopidy[460]: INFO     [SpotifyBackend-6] mopidy_spotify.lookup Failed to lookup 'spotify:track:543EF2bXYULbVkEPivKwGz': Session must be logged in
Oct 23 13:17:29 raspberrypi mopidy[460]: INFO     [SpotifyBackend-6] mopidy_spotify.lookup Failed to lookup 'spotify:track:1cH1Z3uninmgDTd8Z0VP3P': Session must be logged in
Oct 23 13:17:29 raspberrypi mopidy[460]: INFO     [SpotifyBackend-6] mopidy_spotify.lookup Failed to lookup 'spotify:track:4Kdcx5sWtd6U98iTNeGssx': Session must be logged in
Oct 23 13:17:29 raspberrypi mopidy[460]: INFO     [SpotifyBackend-6] mopidy_spotify.lookup Failed to lookup 'spotify:track:2lQFkkfnjQOyepSZLnj75t': Session must be logged in
Oct 23 13:17:29 raspberrypi mopidy[460]: INFO     [SpotifyBackend-6] mopidy_spotify.lookup Failed to lookup 'spotify:track:7x84LZKE5mSdxZI6MBCRbw': Session must be logged in
[... cut ...]
Oct 23 13:20:56 raspberrypi mopidy[460]: INFO     [SpotifyBackend-6] mopidy_spotify.lookup Failed to lookup 'spotify:track:7jY2wjlJ5IfHIlMihHIvRu': Session must be logged in
Oct 23 13:21:01 raspberrypi mopidy[460]: INFO     [MpdSession-28] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51612
Oct 23 13:21:01 raspberrypi mopidy[460]: INFO     [MpdSession-29] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51616
Oct 23 13:21:01 raspberrypi mopidy[460]: INFO     [MpdSession-30] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51620`
Is this the output you are looking for?

s-martin, 2020-10-23:
Yes, that’s the output, but it looks like that you’re not logged in to Spotify.

s-martin, 2020-10-23:
That’s probably same as #893.

Please check there for workarounds. I think the main problem is that internet connection is not yet available, when mopidy starts, so the login won’t work.
Scanning the library also restarts mopidy and login works then.

If you are fine with the prerequisite that your Phoniebox works only when an internet connection is available, there are some workarounds described in that issue.

Duplicate if #893

s-martin, 2020-10-23:
Workaround should be this: https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/893#issuecomment-613952753

s-martin, 2020-10-23:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#spotify-wont-play-but-after-scan-of-library-it-works

closed 2020-10-25 by SirY4