MiczFlor/RPi-Jukebox-RFID #1120: 🐛 | netcat binary not installed

varac, 2020-10-25:
## Bug

### What I did

* installed `raspios_lite_arm64-2020-08-24`
* installed phoniebox by using the installer script from the master branch

### What happened

I see this error in the syslog:

    Oct 25 13:12:02 musicbox bash[402]: /home/pi/RPi-Jukebox-RFID/scripts/startup-scripts.sh: line 32: nc: command not found

### I expected this to happen

The installer script should install the `netcat` packge.

## Software

### Base image and version

* `raspios_lite_arm64-2020-08-24` (from https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2020-08-24/)

### Branch / Release

* master branch

### Installscript

* `scripts/installscripts/buster-install-default.sh`

## Hardware

### RaspberryPi version

* `Raspberry Pi 3 Model B Plus Rev 1.3`

### RFID Reader

* rc522

s-martin, 2020-10-25:
Fixed in develop.

Thanks for finding this.

closed 2020-10-25 by s-martin

varac, 2020-10-25:
@s-martin Thx for fixing this so fast.
Btw, I realized that the arm64 image won't work since mopidy will fail to install. Maybe worth mentioning in the install docs (if it's not already there and I oversaw).

s-martin, 2020-10-25:
> @s-martin Thx for fixing this so fast.
> Btw, I realized that the arm64 image won't work since mopidy will fail to install. Maybe worth mentioning in the install docs (if it's not already there and I oversaw).

What was the error message? If it's worth it you could open a separate issue.

commit b9b21468 (2020-10-25): Install netcat - fixes #1120