MiczFlor/RPi-Jukebox-RFID #1126: 🐛 | BUG SUMMARY - GPIO

aeskimo, 2020-10-30:
Warum läuft die Installation nicht?

sudo /home/pi/RPi-Jukebox-RFID/components/gpio_control/install.sh
cp: der Aufruf von stat f?r '/example_configs/gpio_settings.ini' ist nicht m?glich: Datei oder Verzeichnis nicht gefunden
disable old services: phoniebox-gpio-buttons and phoniebox-rotary-encoder
Failed to stop phoniebox-rotary-encoder.service: Unit phoniebox-rotary-encoder.service not loaded.
Failed to disable unit: Unit file phoniebox-rotary-encoder.service does not exist.
Failed to stop phoniebox-gpio-buttons.service: Unit phoniebox-gpio-buttons.service not loaded.
Failed to disable unit: Unit file phoniebox-gpio-buttons.service does not exist.
Install all required python modules
Could not open requirements file: [Errno 2] Datei oder Verzeichnis nicht gefunden: 'requirements.txt'

Installing GPIO_Control service
Press enter to continue 
/etc/systemd/system/phoniebox-gpio-control.service exists.
systemctl daemon-reload
restarting service
Press enter to continue 
Phoniebox GPIO Service started correctly .....
For further configuration of GPIO-devices consult the wiki:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Using-GPIO-hardware-buttons

Danke für eure Rückmeldung

s-martin, 2020-10-30:
Why are you calling install.sh directly and don’t use the regular installer script for Phoniebox?

What’s the output of `cat /home/pi/RPi-Jukebox-RFID/settings/version_number`?

s-martin, 2021-02-28:
Closing this now. If you still have the issue feel free to reopen it.

closed 2021-02-28 by s-martin