MiczFlor/RPi-Jukebox-RFID #1127: 🐛 | CSV Export of CARD IDs does not include Spotify links

Streuerei, 2020-11-01:
## Bug

### What I did

- Go to /cardRegisterNew.php and click on 'Create .csv file of available RFID links'
- Download the file and open it: [PhonieboxRFID-2020-11-01_6-51-59.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/5470281/PhonieboxRFID-2020-11-01_6-51-59.txt)
- The file contains:
-- Card ID
-- Name of the folder

The Spotifiy link / Stream URL  is missing. You can find the file attached

As I try to import the file to another Phoniebox, the import is shown as succesfull, but not folders / Card IDs are created

### What happened

An CSV file whith Card ID and Name of the Folder is created, but without the Spotifiy link / Stream URL 
An Import to another Phoniebox did not create any new folder / card id.

### I expected this to happen

A CSV file, including all required information (Card ID, Folder/Album Name, Stream URL) is created
All IDs in the CSV are imported to another Phoniebox and listed under Available audio

## Software

### Base image and version

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Branch / Release

Streuerei, 2020-11-01:
Some part of the post got lost.
Branch: Master
Pie: Raspberry Pi 4 Model B Rev 1.1
RFID: HXGCoLtd Keyboard
Soundgard: onboard

s-martin, 2020-11-01:
Looks like Spotify URLs were never implemented in `htdocs/rfidExportCsv.php`.

But I agree, in spotify edition this should be available.

Nuexo, 2021-02-14:
Hi, 
I'm also interested in this Feature.
I would try to fix ist by my own. Ist there a hint where I should start? E.g. whrere the Export function is. (I'm expert with Microcontroller C only and not very experienced in Linux)

s-martin, 2021-02-14:
The script here should be a start https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/htdocs/rfidExportCsv.php

jonaskor, 2024-12-17:
This issue still persists. It would be so cool to be able to export the album list including the spotify links. If there is a reinstall necessary for some reason, this would be the ideal backup (I rarely use mp3 files). Any chance to get this enabled?