MiczFlor/RPi-Jukebox-RFID #1128: configurable PCM  interface 🚀 | FEATURE SUMMARY

schirrm, 2020-11-05:
## Feature Description
first of all thanks a lot for this awesome project! really love it. was thinking about setting up a own alternative for the toniebox as i don't like the cloud enforcement. before starting I took some moments looking for existing aproaches and am happy to safe my time on the implementation. really well done.

as I set up a test environment I experienced a challenge:
for my CM108 usb audio controller I needed to do some adaption: I think it might make sense to review whether this is relevant to all.
I couldn't change the volume because the skript tried to access the card, not the pcm device of it. I added in my installation:

playout_controls.sh:PCMIFACENAME="PCM"
playout_controls.sh:echo "$PCMIFACENAME"
playout_controls.sh:                amixer $ADDPARAM sget \'$PCMIFACENAME\' | grep -Po -m 1 '(?<=\[)[^]]*(?=%])' > $VOLFILE

[basically changed the audioifacename for all occurances to PCMIFACENAME]

as there are quite some installation in the wild, not sure whether this feature helps more people, but looks like a reasonable change to me ;)

### What functionality would you like to see in your phoniebox?
have an variable in a central config for setting the audioifacename and pcmiface in general.

i.e. `I would love to cook my breakfast eggs with the help of our phoniebox`

### How do you envision the feature to work from a users perspective?

i.e. `In the mornings when I stumble to the bathroom, I want the phoniebox to react on the RFID chip injected under the skin of my right hip. I envision it to walk to the kitchen, get the eggs and a pot out, and start the cooking of the eggs on the stove.`

### Further information that might help

i.e. `the stove is operated with cooking gas, so the implemenation has to make sure the stove is handled with the greatest possible care`

s-martin, 2020-11-17:
Could you explain a little more why changing the variable NAME (but not the variable CONTENT) would fix your issue?

I don’t know, if I get it right

schirrm, 2020-11-17:
Sure
Because the playback still uses the audioiface. But thé volume control the pcmiface. 
Typically i uses the amixer pcm iface for playback and volume. But as the playback didnt work on the pcm interface for me and thé volume control didnt make it on thé audioiface i introduced for my box a 2nd variable. This currently is only used for the pcm functions of the amixer

s-martin, 2024-04-19:
If someone wants to implement this, I suggest to implement it for V3