MiczFlor/RPi-Jukebox-RFID #1133: 🐛 | Wrong track number displayed (using HD44780 LCD)

Conendrum, 2020-11-08:
Hello everyone,

I've mounted my Jukebox including a LCD display HD44780 (4x20) using the wiki information. So far everything works great! (FYI I'm using 2.1.1 master branch).

However I've noticed one inconstancy in the display of the track number. 
In the file "RPi-Jukebox-RFID/components/displays/HD44780-i2c/i2c_lcd.py", the function "print_track_time_and_number()" display the track number that way:

```
def print_track_time_and_number():
    song_of_playlist = track_number + "/" + playlist_length
    return (fill_with_spaces(track_time, n_cols))[:(n_cols - len(song_of_playlist))] + song_of_playlist
```

"playlist_length" corresponds indeed to the length of your playlist in your "shared/audiofolder" directory, so that number if fine.
However "track_number" corresponds to the track metadata of the mp3 file currently being played:

```
try:                                                                              #
                track_number = current_song_infos['track']                                #
            except KeyError:                                                                  #
                track_number = "1"
```

Which means that your "track_number" will never linearly increment until reaching "playlist_length". Track metadata of mp3 file usually varies from 1 to around 13-15. So even though you have 100 songs in your playlist, it will always display "[1-15]/100".

My suggestion would be:
- to stop using the "track" metadata of the mp3 file
- to define instead a counter that would start at 1 and increment everytime a new song is played

I haven't had the time to look for a solution in the code (just noticed the issue tonight). If I manage to come up with a proposal, I'll post it here.
I'm also open to any other suggestion to fix that.

thanks and cheers

s-martin, 2020-11-08:
Thanks for the bug report.

PRs are always welcome :)

Conendrum, 2020-11-20:
Hello,

Here my proposal to fix this bug (I run some tests and it worked reliably, not sure to have tested all possible combination though). It's probably not the cleanest / best approach, but it works. 

I only had to modify the i2c_lcd.py file, attached the diff file

```
86,89d85
< last_playlist = 0
< track_counter = 1
< list_tracks = [" "]
< first_track = True
310,313c306
<         elif state == "stop":                                                           #
<             track_counter = 0 
<             list_tracks = [" "]
<             first_track = True                                                      
---
>         elif state == "stop":                                                  #
324c317
<                 track_number = str(track_counter)                                             #
---
>                 track_number = current_song_infos['track']                                #
418,440d410
<         ########################### UPDATE TRACK COUNTER #####################################
<         if first_track:
<             list_tracks.append(title)
<             first_track = False
< 
<         if last_playlist == playlist:
<             # Music change within a playlist
<             if (last_title != title) and (title not in list_tracks):
<                 list_tracks.append(title)
<                 track_counter += 1
<             elif (last_title != title) and (title in list_tracks):
<                 list_tracks = list_tracks[:-1]
<                 track_counter -= 1
<                 if track_counter < 1:
<                     track_counter = 1
<         else: 
<             # New Playlist, reset values
<             track_counter = 1
<             list_tracks = [" "]
<             first_track = True
<         #####################################################################################
< 
< 
443,444c413
<         last_title = title
<         last_playlist = playlist                                                               #
---
>         last_title = title                                                                     #

```

[diff.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/5575764/diff.txt)

closed 2020-11-22 by MiczFlor

DSteg, 2020-11-30:
I fixed it for my i2c_lcd.py with the following lines:

line 315:

` try:                                                                              #
                #track_number = current_song_infos['track']                                    #
                track_number = status['song']                                                 #
                track_number_int = int(track_number)                                          #
                track_number = str(track_number_int + 1)                                      #
`

I simply use the song number in the playlist instead of the ID3 track number and adding 1, because song number starts by 0.